felis/USB_Host_Shield_2.0 #680: Using the BTD::btService Array to Support Multiple Remote Connections

Jaboop8, 2021-12-09:
I am using a BT-400 dongle which should support multiple device connections. So I am looking for any examples on how to setup multiple remote connections on a single BTD. I can see when creating a second remote Instance that a second memory address is gets added to the BTD::btService Array associated with the Bluetooth Dongle but when I try to connect the a second PS4 controller to the dongle it never connects. Does anyone have a working example of two PS4 controllers (or Other controller type) connecting to a single Bluetooth Dongle. 

The first PS4 connection works fine but am unsure if I am missing any additional setup that might be needed to get multiple connections working other than just creating the second remote instance.

Thanks in Advance for any Advice!

neromir, 2021-12-11:
Don't have anything helpful to add, but I'm also trying to do this, but with two Switch Joycons. I can get them to work independently, but haven't been able to get them to work at the same time.

xxxajk, 2021-12-13:
While admittedly I'm not a bluetooth expert by any means, I'm going to make an educated guess here.

- The ability to do this is going to use twice the amount of ram, because it has to track two.
- The ability to do this wasn't written because there hasn't been a need.
- The ability to do this is difficult to fit in a low-end MCU with very limited resources, thus not within the scope of the framework.

So here's a quite possible suggestion.

Break the problem down into smaller parts to make each part easier to do. 

That means 3 MCUs. One for each BT, and a central one to coordinate everything. This is exactly what I tend to do in more complex projects, The performance gains, and reduction in complexity pays off in ways you might not expect. This is because you specialize sub-systems that do only one thing, and do them well. The industry has realized this in recent times, by providing a main CPU, and adding extra separate MCU which they call "application processors". The same has been done for several decades on PC's, for example the GPU in your PC.

neromir, 2021-12-13:
After a lot of fiddling, I got it to work. I've included a link to a gist of a modified version of the SwitchProBT.cpp example from the repo.

Some notes:

- I reformatted it for easier reading for me, so hopefully that's not off-putting. 
- The code has also been slightly modified because I am using CLion and PlatformIO, rather than the Arduino IDE. This mostly just means that there are some method forward declarations and inclusion of `Arduino.h` in there that you shouldn't need in the Arduino IDE.
- I am doing this on an Arduino Mega 2560
- I got the idea for how to go about this from the SHADOW controller code from the R2 Builders club on astromech.net. It also uses 2 PS3 Navigation controllers to control the droid using an older version of this library and the same shield.

The process I followed to make it work was as below:

1. Comment out references to `joyconR` (also commented out its `joyconROnInit()` method)
2. Put the SwitchProBT object for `joyconL` into pairing mode by passing `PAIR` as the second parameter.
3. Recompile, push code to Arduino, pair the left Joycon. Note MAC address, store it in `joyconLMac`
4. Uncomment references to `joyconR`, and comment references to `joyconL` as per 1. Repeat steps 1-3 for `joyconR`. DO NOT push buttons on left joycon during this procedure, to avoid prompting it to attempt to connect.
5. Uncomment references to `joyconL` and remove `PAIR` parameter from initialization of both.
6. Recompile, push code to Arduino. Press a button on left joycon, prompting it to connect, note that it works and buttons are recognized. Press a button on right joycon, prompting it to connect, note that it works and buttons are recognized. Pressing buttons on each generates output for each; both joycons are now connected and working.

https://gist.github.com/neromir/26a91d43a63fa609f52750d9a4f36dc0

petanque, 2021-12-17:
Just look at the differences between PS3BT and PS3Multi examples, the same will work for PS4BT.

Lauszus, 2021-12-19:
I'll close this issue, as it sounds like it's solved to me.

There is nothing stopping you from making multiple instances, as also shown in the `PS3Multi` example, as also pointed out by @petanque.

closed 2021-12-19 by Lauszus