felis/USB_Host_Shield_2.0 #683: Reading buffer size that is not length 8 in hidboot.h HIDBoot<BOOT_PROTOCOL>::Poll()

gustavodiaz7722, 2021-12-20:
I have been experimenting with the USBHIDMultimediaKbd and USBHIDBootKbd example projects included with the library.  I have noticed that the USBHIDMultimediaKbd example uses a different HID poll function than the USBHIDBootKbd example.  USBHIDMultimediaKbd uses the poll function defined in hidcomposite.h, while USBHIDBootKbd uses a poll function defined in hidboot.h. 

My issue lies with the poll function located in hidboot.h. This function does not appear to accept buffer lengths that are not equal to 8 in length and simply returns zero values for the buffer. In contrast, the poll function in hidcomposite.h returns the the values stored in the buffer regardless of what length it is. 

My question then is this, can I somehow modify the poll function in hidboot.h to return the buffer regardless if it is not of length 8?

Lauszus, 2021-12-20:
Simply fork this repo, modify the desired file and send a PR :)

You can read more about on how to do it here: https://docs.github.com/en/get-started/quickstart/fork-a-repo

gustavodiaz7722, 2021-12-21:
Ok great thanks, Ill see what I can do to get it to work. 

 Do you have any idea where I might start looking in the code to implement the proposed change?  From what I have seen so far it appears that I will need to make a change to the Poll function or the inTransfer function located in hidboot.h. Any suggestions would be appreciated.

closed 2026-10-29 by gustavodiaz7722