felis/USB_Host_Shield_2.0 #684: Latency Servicing Multiple Remotes

Jaboop8, 2021-12-20:
Thanks for the point in the right direction on issue #680  the PS3Multi Example was a big help. 

Now I've run into what might be a deeper issue. I successfully connect both remotes and get reads off of both of them. However, the first remote connection seems to be serviced way more frequently than the second. I am calling Usb.Task() at a frequency of about 250Hz giving me a latency of about 8ms on the first connection. On the second remote I see intermittent servicing, low latency for a second or so and then a couple dropped button presses or high latency on the scale of about 1 second.

I'm looking into Usb.Task() as well as going to mess around with the frequency of this polling but was curious if anyone had any initial thoughts on where this priority to servicing the first BT connection more frequently might be happening or what else might be causing the intermittent high latency on the second BT connection.

Thanks in Advance!

Jaboop8, 2021-12-20:
Some Simple test I ran that might give others clues into where optimization might be needed.

After connecting a third controller. The first connection slowed noticeably and occasionally missed button presses. The third one would miss button presses entirely but would get the L2 and L3 reads. The third controller also did not get any touch data but the first and second did.

I tried increasing the polling of the USB.Task() to be at a frequency of 400Hz and this had some improvement in the reading of the second connection but still missed button presses.

xxxajk, 2021-12-23:
If UHS3 gets a bluetooth port (I lack the time) then you won't have to worry about the servicing times, since those will happen in the background.

Lauszus, 2022-01-03:
Most Bluetooth dongle has a poll interval of 1 ms, thus you need to call `Usb.Task` at 1000 Hz.

Jaboop8, 2022-01-11:
@Lauszus Turning up the polling to about 1500Hz seemed to get rid of most of the latency. I did find that if using the PS4 parser a PS5 controller will work for the majority of functionality but for whatever reason if the PS5 controller is not the first connection in the BTD::btService Array the PS5 controller will drop button presses. Might just be a weird configuration causing some sort of parsing issue but thought it was interesting. If the PS5 is the first connection and the PS4 is second things work much better. I'll follow up with more details once I can get two PS4 controllers to do more tests with.

Lauszus, 2022-01-11:
@Jaboop8 good to hear that you are making progress! :)