felis/USB_Host_Shield_2.0 #688: After using Midi:SendData() the library stops receiving messages on the second virtual interface

chrisp250, 2021-12-30:
Hi,
I'm using the library with a mini USB shield 2.0 and a Teensy 3.2. Also SPI4TEENSY3 is disabled.

I'm trying to communicate with a Novation Launchpad Mini MK3. This devices has two virtual interfaces for midi.
This is a summary of the code:
Midi:SendSysEx()
loop:
  Midi:RecvData()

This works fine and RecvData() received data on both ports.
However, if I instead do this:
Midi:SendData()
loop:
  Midi:RecvData()

I can no longer receive any events coming on the second virtual interface. The first virtual interface is fine and I can receive the notes coming from the pads.
I tested the messages on a PC with the same device and it works fine, so I'm confident this is not a device issue.
Also I need be able to send "Note On" messages to the device, so just replacing SendData with SendSysEx is not a usable workaround in my case.

Thank you in advance
Christian

Lauszus, 2022-01-03:
@YuuichiAkagawa can you take a look at this?

YuuichiAkagawa, 2022-01-04:
The two MIDI ports use the same endpoint and are not distinguished by the MIDI driver.
- CN=0 : LPMiniMK3 DAW
- CN=1 : LPMiniMK3 MIDI

If you can't get the value, it's likely that the device isn't sending.
In what mode are you operating? Is it in DAW mode? Or is it Programmer mode?

chrisp250, 2022-01-04:
Hi Yuuichi,
Thank you for your reply.

> The two MIDI ports use the same endpoint and are not distinguished by the MIDI driver.
> 
> * CN=0 : LPMiniMK3 DAW
> * CN=1 : LPMiniMK3 MIDI

Yes, that's correct.
 
> If you can't get the value, it's likely that the device isn't sending. In what mode are you operating? Is it in DAW mode? Or is it Programmer mode?

I'm confident the launchpad still sends messages because I tested it sending the same exact messages using a PC/python combination and it works fine. I have tested on both DAW and Live mode and the behaviour is the same. As soon as I do a _SendData()_ I stop getting messages on **CN=1 : LPMiniMK3 MIDI**.

I put together a sample/simplified code hoping it will help identifying the issue:

`
/*
 *******************************************************************************
 * USB-MIDI Launchpad Mini mk3 test
 *
 *******************************************************************************
 */

#define SPI4TEENSY3 0
#include <Arduino.h>
#include <usbh_midi.h>
#include <usbhub.h>
#include <MIDI.h>

#define _MIDI_SERIAL_PORT Serial1

USB Usb;
USBH_MIDI  Midi1(&Usb);
USBH_MIDI  Midi2(&Usb);

void MIDI_poll();

// Send test messages
void onInit()
{
  //Switch DAW mode on message
  uint8_t daw_mode[9] = {0xf0,0x00,0x20, 0x29, 0x02, 0x0d, 0x10, 0x01, 0xf7};
  //Device Inquire message
  uint8_t device_inq[6] = {0xF0, 0x7E, 0x7F, 0x06, 0x01, 0xF7};
  
  delay(100);
  
  Serial.printf("Sending device inquiry...\n");

  //Send switch to DAW mode message
  //Midi1.SendSysEx(daw_mode, sizeof(daw_mode),0);
  
  //Send device inquiry message:
  
  //If sent with this function interface on cable 1 received messages normally
  Midi1.SendSysEx(device_inq, sizeof(device_inq),0);

  //If sent with this function instead interface on cable 1 stops receiveing messsages
  //This is the case not only for SysEx messages but for noteon messages as well.
  //Midi1.SendData(device_inq);
  
}

void setup()
{

  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.printf("Start...");
  
  if (Usb.Init() == -1) {
    while (1); //halt
  }//if (Usb.Init() == -1...
  delay( 200 );
  Usb.Task();
  Serial.printf("init..\n");
  Midi1.attachOnInit(onInit);
}

void loop()
{
  Usb.Task();

  if ( Midi1 ) {
    MIDI_poll();
  }
  //delay(1ms) if you want
  //delayMicroseconds(1000);
}

// Poll USB MIDI Controler and print received data
void MIDI_poll()
{
  uint8_t outBuf[ 20 ];
  uint8_t size;

  //Serial.printf("Polling\n");
  do {
    if ( (size = Midi1.RecvData(outBuf,true)) > 0 ) {
      Serial.printf("Raw received: (s)%d, %x, %x, %x, %x\n",size, outBuf[0],outBuf[1],outBuf[2],outBuf[3]);
    }
  } while (size > 0);
}
` 

Thank you again for looking at this.
Christian

YuuichiAkagawa, 2022-01-05:
Thank you very much for the code.
I found a bug in usbh_midi.cpp. If the CIN is non-zero, it will not work properly.

Change the following lines and try it.
https://github.com/felis/USB_Host_Shield_2.0/blob/d1f82b4fb47c48c5e279bbeb1d70f1a9e8442da3/usbh_midi.cpp#L362
```
        return getMsgSizeFromCin((cin & 0x0f));
```

chrisp250, 2022-01-05:
Thank you for looking into this. This fixed the problem, I'm now getting the replies on cable 1.

I had seen the issue with the incorrect message size being reported for cable 1 but I didn't expect this would be related.

Much appreciated!

Lauszus, 2022-01-07:
I've merged: https://github.com/felis/USB_Host_Shield_2.0/pull/689, so closing this issue.

@chrisp250 please use the latest version from the master branch :)

closed 2022-01-07 by Lauszus