felis/USB_Host_Shield_2.0 #692: usb host controller shield (with modified SS pin to avoid conflict) + library wont coexist with seeed wd5500 shield

geetee24, 2022-01-29:
usb host controller shield (with modified SS pin to avoid conflict) + library wont coexist with seeed wd5500 shield

Each lib works solo but together the 2 libs dont work at all.  No usb access and no ethernet access.  IT IS NOT a SS issue nor any other port conflict.   Appears to be a SPI issue

xxxajk, 2022-01-31:
1. Seeed Studio's shield  isn't an official shield.
2. Seeed Studio doesn't use the universal 6-pin SPI port connector. So you will have to modify the board to get what you want.
3. Seeed Studio assumes an UNO is going to be used, which by today's standards, is quite obsolete.

xxxajk, 2022-01-31:
Correction. it is using the 6pin header, it's probably not sharing SPI properly.