felis/USB_Host_Shield_2.0 #694: esp32 node mcu 32s rebooting continously with USB_MIDI_Converter

StureKn, 2022-02-08:
I have nothing connected to the node mcu 32s board and it works fine with all sketches I've tried but when I load the USB_MIDI_Converter it continously reboots.
Any ideas on how to solve it?

YuuichiAkagawa, 2022-02-13:
I don't have NodeMCU-32S so I tested it with ESP32-DevKitC.
The version of Arduino core worked fine with either 1.0.6 or 2.0.2. 
I have no idea what caused the reboot.

> I made a test using USB_desc and disovered that it is the USB.init that makes it reboot.

This is not a MIDI problem. Please change the title.

xxxajk, 2022-02-20:
It's probably missing a yield(); in a loop some where...

StureKn, 2022-03-05:
Yes, you are right. It is not a MIDI problem and I will change the title.
I have now used the **board_qc.ino** 
I'm using the USB Host Shield mini and the NodeMCU-32S and I get the following output.

-------------------------------------------
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e38bc: 00545414 00001004 00004010
Core 1 register dump:
PC      : 0x400e38c0  PS      : 0x00060930  A0      : 0x800d266d  A1      : 0x3ffb1f10  
A2      : 0x3ffbdc64  A3      : 0x0000000c  A4      : 0x00000001  A5      : 0x3ff42000  
A6      : 0x3ffbdc64  A7      : 0x00000002  A8      : 0x800d28c2  A9      : 0x3ffb1ef0  
A10     : 0x3ffb844c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000014  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400d2b4c  LEND    : 0x400d2b59  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e38c0:0x3ffb1f10 0x400d266a:0x3ffb1f30 0x400d11a5:0x3ffb1f50 0x400d12e1:0x3ffb1f70 0x400d3ff2:0x3ffb1fb0 0x400861c1:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e38bc: 00545414 00001004 00004010
Core 1 register dump:
PC      : 0x400e38c0  PS      : 0x00060930  A0      : 0x800d266d  A1      : 0x3ffb1f10  
A2      : 0x3ffbdc64  A3      : 0x0000000c  A4      : 0x00000001  A5      : 0x3ff42000  
A6      : 0x3ffbdc64  A7      : 0x00000002  A8      : 0x800d28c2  A9      : 0x3ffb1ef0  
A10     : 0x3ffb844c  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000004  SAR     : 0x00000014  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x400d2b4c  LEND    : 0x400d2b59  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e38c0:0x3ffb1f10 0x400d266a:0x3ffb1f30 0x400d11a5:0x3ffb1f50 0x400d12e1:0x3ffb1f70 0x400d3ff2:0x3ffb1fb0 0x400861c1:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57
----------------------------------------------------

I also bought the ESP32-DevKitC but it gives the same results.

Can it be that I connect the UHS mini to the wrong pins? Which pin should be used?
I really would like to get this working so please, let me know if you have any ideas on what I'm doing wrong.

I have tested the board_qc with Arduino Uno and a USB Host Shield and it works fine.

YuuichiAkagawa, 2022-03-06:
@StureKn See #580.

StureKn, 2022-03-15:
It works now.
I updated the USB Host Shield library 2.0 to version 1.6.1

closed 2022-03-15 by Lauszus