felis/USB_Host_Shield_2.0 #695: PS3 Controller paired, but not working if condition PS3.PS3Connected in file PS3BT.ino is not removed

mugiro, 2022-02-10:
Hello,
I'm trying to connect a SONY PS3 (model CECHZC2U - [Amazon Link](https://www.amazon.com/CECHZC2U-PLAYSTATION-SIXAXIS-WIRELESS-CONTROLLER-3/dp/B00FXTBJN0) ) via BT4.0 dongle with chipset CSR8510 A10 with Uno+host shield and connection with the example code PS3BT.ino looks ok and it is working:
```
PS3 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length: 07
Type: 05
Address: 81
Attributes: 03
MaxPktSize: 0010
Poll Intrv: 01
Endpoint descriptor:
Length: 07
Type: 05
Address: 02
Attributes: 02
MaxPktSize: 0040
Poll Intrv: 01
Endpoint descriptor:
Length: 07
Type: 05
Address: 82
Attributes: 02
MaxPktSize: 0040
Poll Intrv: 01
Bluetooth Dongle Initialized
Connection established
...
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 00:1A:7D:DA:71:13
HCI version: 06
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 25 08
Incoming Connection Request
Remote Name: Wireless Controller
PS4/PS5 controller is connecting
Ignore HCI Event: 12
...
Ignore HCI Event: 12
Connection established
Connected to Device: 41:42:EF:C9:02:90
```

However, once the connection is established the PS3 controller is not responding. Nothing in the Serial monitor. 
We do not know if the message is "PS4/PS5 controller is connecting" instead of some sort of PS3 is the issue. 

If the condition in PS3BT.ino is removed:
`(PS3.PS3Connected || PS3.PS3NavigationConnected) {`

then we are able to receive information from the PS3 controller as follows:

```
Connected to Device: 41:42:EF:C9:02:90
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
Ignore HCI Event: 20
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
L2CAP Connection Request - PSM: 00 11 SCID: 00 40 Identifier: 01
HID Control Incoming Connection Request
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 0
LeftHatX: 0 LeftHatY: 1 RightHatX: 2 RightHatY: 160
HID Control Successfully Configured
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
L2CAP Connection Request - PSM: 00 13 SCID: 00 41 Identifier: 03
HID Interrupt Incoming Connection Request
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 1 RightHatX: 2 RightHatY: 160
HID Interrupt Successfully Configured
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 0 LeftHatY: 0 RightHatX: 0 RightHatY: 1
LeftHatX: 75 LeftHatY: 128 RightHatX: 128 RightHatY: 128
LeftHatX: 75 LeftHatY: 128 RightHatX: 128 RightHatY: 128
LeftHatX: 75 LeftHatY: 128 RightHatX: 128 RightHatY: 128
```

We forked the code to show the changes in branch:

[Branch PS3connectedCond](https://github.com/mugiro/USB_Host_Shield_2.0/tree/PS3connectedCond)

Sorry if I'm missing something herein but we want to know what is happening and why the conditional PS3.PS3Connected never changes its value.

EpicInfernos, 2022-05-15:
I have the same issue. Do I download and use this branch for the time being? Is there a difference between this and the master branch?

EpicInfernos, 2022-05-15:
It seems that the board and host shield recognizes the controller not as a PS3 controller but a generic wireless controller. Does anyone know how to fix that?

HisashiKato, 2022-05-15:
BTD.cpp in the original USB HOST Library distinguishes between DUALSHOKC3 and DUALSHOCK4/5 by remote_name (Bluetooth Device Name).

The Bluetooth Device Name for Sony's DUALSHOCK3 is "PLAYSTATION(R)3".
However, the Bluetooth Device Name of the controller in the log is "Wireless Controller" which is the same as DUALSHOCK4/5.
In its log, the library BTD.cpp was determined that the controller is a PS4/5 controller, not a PS3 controller.

And in PS3BT.cpp, "PS3Connected = true" when the first letter of remote_name is "P".
https://github.com/felis/USB_Host_Shield_2.0/blob/master/PS3BT.cpp#L471-L476

Thus PS3Connected will not be true.

(Translated from Japanese to English using a translation site)

EpicInfernos, 2022-05-15:
Is that a problem with the controller or is the something to do with the dongle or the library or missing drivers? I bought the controller from a certified reseller and I checked the VID and PID of my controller and found that it was the same as the one defined in the library. I can say that it is a genuine ps3 controller.

HisashiKato, 2022-05-15:
I can't say for sure, but I suspect it is a controller issue.
I cannot rule out the possibility that Sony has changed the device name.
But the VID and PID of a device can be easily spoofed (like the many spoofed controllers and converters on the market)
Either way, I don't know.