felis/USB_Host_Shield_2.0 #700: Teensy 3.2 + USB Host Shield

escape-lib, 2022-03-17:
Hi all,

I'm really at a loss here. I purchased the full sized USB shield (as I couldn't get the mini in Australia) to use with my Teensy 3.2. On my Arduino UNO it functions perfectly fine, however with my Teensy, it fails the board_qc sketch.

`Circuits At Home 2011

USB Host Shield Quality Control Routine

Reading REVISION register... Die revision invalid. Value returned: 09

Unrecoverable error - test halted!!

0x55 pattern is transmitted via SPI

Press RESET to restart test`

My pin connections are as follows (following this SPI hookup guide: https://www.pjrc.com/teensy/td_libs_SPI.html):

INT -> D9
SS -> D10
MOSI -> D11
MISO -> D12
SCK -> D13
3.3V -> 3.3V + RESET
GND -> GND
VIN -> VIN

VBUS is jumped to 5V & my VIN coming from the Teensy is 5V.

Thanks.

escape-lib, 2022-03-17:
Not sure if it's relevant, but the `Value returned: 09` only returns 09 when there's no USB device plugged into the shield. Once a USB device is plugged in it returns 00.

xxxajk, 2022-03-17:
it isn't. I don't see why you would want to use this library on a teensy 4.1, especially since it already provides EHCI USB, which is high-speed USB.

escape-lib, 2022-03-17:
> it isn't. I don't see why you would want to use this library on a teensy 4.1, especially since it already provides EHCI USB, which is high-speed USB.

I’m using a Teensy 3.2 (as said in the title and the post) not a 4.1 - the 3.2 does not provide native USB host capabilities.

escape-lib, 2022-03-17:
> it isn't. See the state machine return values.

I'm unsure of what you mean by the state machine return values - I couldn't find reference to that in the documentation.

xxxajk, 2022-03-17:
Unfortunately, UHS2 scatters the codes.
https://github.com/felis/USB_Host_Shield_2.0/blob/a321ad36ab4dd9ea000ffa809532a937ffca06d0/UsbCore.h#L90
https://github.com/felis/USB_Host_Shield_2.0/blob/master/max3421e.h#L214
PIDERR is 0x09