felis/USB_Host_Shield_2.0 #703: Is the INT pin made use of

versamodule, 2022-04-06:
Using an ESP32 along with a Mini USB Host Shield.
Connected a USB keyboard and it does report what I press on it. 
I connected my scope to all the pins and there is no activity on the MAX3421 INT pin. I also see that there is constant communication on the SPI bus with no breaks.   It's my understanding that the INT pin signals the micro that something needs to be dealt with. 
So why the constant spi comms then ? 

With the IRQ pin connected or not makes no difference in operation.

xxxajk, 2022-04-06:
because uhs2 is polled.

versamodule, 2022-04-06:
Doesn't it seem a waste to keep polling when there is a IRQ available ?

xxxajk, 2022-04-06:
In UHS30 this uses the IRQ line, but ESPxxxx is not yet supported. 
The MCU should always check every 1ms in the optimum circumstances.
UHS30 uses the IRQ for SOF, in order to provide the housekeeping that you are describing.
Some systems avoid this by using DMA, and more intelligent silicon. This isn't an option for the MAX chip.

versamodule, 2022-04-06:
ok. thank you. 
FWIW.  My first attempt before moving onto the ESP32 was the MEGA 2560, and it too does not make use of the IRQ line. 
My confusion comes from 2 places. The code comments, where the IRQ pin is defined, and people showing hookup diagrams using the IRQ line.

closed 2022-04-06 by versamodule