felis/USB_Host_Shield_2.0 #710: Why does USB Host Shield SS pin voltage keep high?

Low-maooldcat, 2022-04-29:
Hello:
I used Arduino UNO with USB Host Shield, and they can work, but I found USB Host Shield SS pin is high.
In MAX3421E data sheet, if SS pin is low, then SPI data can transmit. 
As far as I understand, SS pin should be low to transmit data, but SS pin is now high and can be used normally.

Thanks.

YuuichiAkagawa, 2022-04-29:
My Arduino UNO's SS is low. I don't know why yours is high.  
Note : It's only LOW during communication, not always LOW.
[image]

Low-maooldcat, 2022-04-30:
> 我的 Arduino UNO 的 SS 很低。不知道為什麼你的高。 注意：它只是在通信期間為低電平，並不總是為低電平。 [image]

Hi, can you share your test code?

YuuichiAkagawa, 2022-04-30:
https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USBH_MIDI/USBH_MIDI_dump/USBH_MIDI_dump.ino
And MIDI keyboard is connected.

or

https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/USBHIDBootMouse/USBHIDBootMouse.ino
And USB mouse is connected.