felis/USB_Host_Shield_2.0 #720: How communicate between the arduino mega host and the stm32f407 microcontroller via usb?

Navigatoroleg, 2022-05-28:
Good afternoon. I ask you to help me figure out how to communicate between the arduino mega host and the stm32f407 microcontroller via usb. I need to send a hex from arduino mega host to the stm32 microcontroller and receive a response from it via usb 2.0. what libraries do I need to use?

Navigatoroleg, 2022-05-30:
01
--
​
Device descriptor:​
Descriptor Length:​ ​ ​ 12
Descriptor type:​ ​ ​ 01
USB version:​ ​ ​ ​ ​ ​ 0200
Device class:​ ​ ​ ​ ​ ​ 02
Device Subclass:​ ​ ​ 02
Device Protocol:​ ​ ​ 00
Max.packet size:​ ​ ​ 40
Vendor ​ ID:​ ​ ​ ​ ​ ​ 0483
Product ID:​ ​ ​ ​ ​ ​ 5740
Revision ID:​ ​ ​ ​ ​ ​ 0200
Mfg.string index:​ ​ ​ 01
Prod.string index:​ ​ ​ 02
Serial number index:​ ​ ​ 03
Number of conf.:​ ​ ​ 01
​
Configuration descriptor:
Total length:​ ​ ​ ​ ​ ​ 0043
Num.intf:​ ​ ​ ​ ​ ​ 02
Conf.value:​ ​ ​ ​ ​ ​ 01
Conf.string:​ ​ ​ ​ ​ ​ 00
Attr.:​ ​ ​ ​ ​ ​ ​ ​ ​ C0
Max.pwr:​ ​ ​ ​ ​ ​ 32
​
Interface descriptor:
Intf.number:​ ​ ​ ​ ​ ​ 00
Alt.:​ ​ ​ ​ ​ ​ ​ ​ ​ 00
Endpoints:​ ​ ​ ​ ​ ​ 01
Intf. Class:​ ​ ​ ​ ​ ​ 02
Intf. Subclass:​ ​ ​ ​ ​ ​ 02
Intf. Protocol:​ ​ ​ ​ ​ ​ 01
Intf.string:​ ​ ​ ​ ​ ​ 00
Unknown descriptor:
Length:​ ​ ​ ​ ​ ​ 05
Type:​ ​ ​ ​ ​ ​ 24
Contents:​ ​ ​ 0010010524
Unknown descriptor:
Length:​ ​ ​ ​ ​ ​ 05
Type:​ ​ ​ ​ ​ ​ 24
Contents:​ ​ ​ 0100010424
Unknown descriptor:
Length:​ ​ ​ ​ ​ ​ 04
Type:​ ​ ​ ​ ​ ​ 24
Contents:​ ​ ​ 02020524
Unknown descriptor:
Length:​ ​ ​ ​ ​ ​ 05
Type:​ ​ ​ ​ ​ ​ 24
Contents:​ ​ ​ 0600010705
​
Endpoint descriptor:
Endpoint address:​ ​ ​ 82
Attr.:​ ​ ​ ​ ​ ​ ​ ​ ​ 03
Max.pkt size:​ ​ ​ ​ ​ ​ 0008
Polling interval:​ ​ ​ 10
​
Interface descriptor:
Intf.number:​ ​ ​ ​ ​ ​ 01
Alt.:​ ​ ​ ​ ​ ​ ​ ​ ​ 00
Endpoints:​ ​ ​ ​ ​ ​ 02
Intf. Class:​ ​ ​ ​ ​ ​ 0A
Intf. Subclass:​ ​ ​ ​ ​ ​ 00
Intf. Protocol:​ ​ ​ ​ ​ ​ 00
Intf.string:​ ​ ​ ​ ​ ​ 00
​
Endpoint descriptor:
Endpoint address:​ ​ ​ 01
Attr.:​ ​ ​ ​ ​ ​ ​ ​ ​ 02
Max.pkt size:​ ​ ​ ​ ​ ​ 0040
Polling interval:​ ​ ​ 00
​
Endpoint descriptor:
Endpoint address:​ ​ ​ 81
Attr.:​ ​ ​ ​ ​ ​ ​ ​ ​ 02
Max.pkt size:​ ​ ​ ​ ​ ​ 0040
Polling interval:​ ​ ​ 00
​
Addr:1(0.0.1)
​
--

touchgadget, 2022-05-30:
```
Device class:       02
Device Subclass:    02
Device Protocol:    00
```
This indicates the device is a CDC ACM class device. The driver may have to be changed because the protocol is 0 instead of 1 (CDC_PROTOCOL_ITU_T_V_250).

Navigatoroleg, 2022-05-31:
> ```
> Device class:       02
> Device Subclass:    02
> Device Protocol:    00
> ```
> 
> Это указывает на то, что устройство является устройством класса CDC ACM. Драйвер, возможно, придется изменить, поскольку протокол равен 0 вместо 1 (CDC_PROTOCOL_ITU_T_V_250).
> 
> К сожалению, драйвер cdcacm.c смотрит на параметры интерфейса, а не на устройство, поэтому он может работать как есть.
> 
> ```
> Intf. Class:​ ​ ​ ​ ​ ​ 02
> Intf. Subclass:​ ​ ​ ​ ​ ​ 02
> Intf. Protocol:​ ​ ​ ​ ​ ​ 01
> ```

Great, but I don't quite understand where to make your defines, I have to use cdcacm.h cdcacm.cpp . how do I register my descriptors in this library?

Navigatoroleg, 2022-05-31:
Great, but I don't quite understand where to make your defines, I have to use cdcacm.h cdcacm.cpp . how do I register my descriptors in this library?

Navigatoroleg, 2022-05-31:
> ```
> Device class:       02
> Device Subclass:    02
> Device Protocol:    00
> ```
> 
> This indicates the device is a CDC ACM class device. The driver may have to be changed because the protocol is 0 instead of 1 (CDC_PROTOCOL_ITU_T_V_250).
> 
> Oops, the cdcacm.c driver looks at interface parameters, not device so it might work as is.
> 
> ```
> Intf. Class:​ ​ ​ ​ ​ ​ 02
> Intf. Subclass:​ ​ ​ ​ ​ ​ 02
> Intf. Protocol:​ ​ ​ ​ ​ ​ 01
> ```

if( Acm.isReady()) {
       uint8_t rcode;

       /* reading the keyboard */
       if(Serial.available()) {
         uint8_t data= Serial.read();
         /* sending to the phone */
         rcode = Acm.SndData(1, &data);
         if (rcode)
            ErrorMessage<uint8_t>(PSTR("SndData"), rcode);
       }//if(Serial.available()...

As I understand it, this function can only send one byte. And how to send an array of bytes for example 0001AAF111AA and accept about this size in response

Navigatoroleg, 2022-05-31:
The question is removed I have already done everything myself

Navigatoroleg, 2022-05-31:
But still there is a problem, answers every other time happens twice and I shit several times. If I send bytes as in the picture, then each byte is assigned a client address. How do I send an array with one address and an array of bytes.
[image]

Navigatoroleg, 2022-05-31:
Help me how to redo the function so that it sends an address and an array once. In this case, it sends addresses to each byte and therefore does not work correctly.

Navigatoroleg, 2022-05-31:
Thank you very much and a low bow, everything works fine!

Navigatoroleg, 2022-05-31:
> SndData takes nbytes and data pointer so you can send an array of bytes. I do not see any picture. The link points back to this issue.
> 
> ```
> uint8_t mydata[10];
> memset(mydata, 0, 10);
> SndData(10, mydata);
> ```
Thank you very much and a low bow, everything works fine

Navigatoroleg, 2022-06-01:
> > ```
> > Device class:       02
> > Device Subclass:    02
> > Device Protocol:    00
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > This indicates the device is a CDC ACM class device. The driver may have to be changed because the protocol is 0 instead of 1

uint8_t mydata[10];
memset(mydata, 0, 10);
SndData(sizeof(mydata), mydata);

Lauszus, 2022-06-09:
@Navigatoroleg please close issues when they have been resolved :)

closed 2022-06-09 by Lauszus