felis/USB_Host_Shield_2.0 #723: Using host shield with NRF52-DK

hosseinmsvz, 2022-06-07:
Hello,
I am trying to connect the USB host shield to NRF52. I completely removed two level-shifters from the shield and closed all jumpers to prevent any problems.
Since the DK cannot supply the 3v3 voltage for the shield, I used an external power source to produce the supply voltage.
I wrote the initialization and test code based on your library to test the circuit, but I see strange behavior on the MISO pin.
As you can see on the below picture, which shows SPI signals right after the restart, the MISO pin sends 0x19 every time the SS pin is low, no matter what value is on the MOSI pin. 
[image]
I couldn't find anything on the documents to know what created this problem, so I would be grateful if you could help me solve the problem.
 I also tested the board with an Arduino Mega before changing it to 3.3v, and it worked perfectly there.

xxxajk, 2022-06-16:
Does the library use transactions? if not, that's possibly the problem, also, the additional hardware should be using a separate CS pin.

hosseinmsvz, 2022-06-18:
Thanks, the problem was because of an error in the register read function. Also, I was using a switching regulator to supply two boards, which caused the oscillator to fail.

closed 2022-06-18 by hosseinmsvz