felis/USB_Host_Shield_2.0 #725: Cannot recieve large Sysex from NUX MG-30

karozas, 2022-07-04:
Hello from Greece.I am trying to build a MIDI contoller for NUX MG-30 (guitar multieffect unit). I  send this sysex message 
F0 43 58 70 0A 00  F7 in order the NUX MG-30 to respond with a 218 bytes long sysex that contains the data of the current patch . The NUX MG-30  is recognised by the usb host , I can send PC or CC messages to the NUX MG-30 or other sysex messages and the device responds normally but does not when I send the above sysex.This is what I get when I run MIDI_USB_desc sketch :
Start

01
--
String Descriptors:
Manufacturer:		NUX
Product:		NUX MG-30
Serial:			220526115751

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		00
Device Subclass:	00
Device Protocol:	00
Max.packet size:	40
Vendor  ID:		1FC9
Product ID:		8260
Revision ID:		0101
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		0158
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			C0
Max.pwr:		00
Unknown descriptor:
Length:		08
Type:		0B
Contents:	000301002000

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		01
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_CONTROL>>>
Unknown descriptor:
Length:		09
Type:		24
Contents:	010002047F0000
Unknown descriptor:
Length:		08
Type:		24
Contents:	0A1001070000
Unknown descriptor:
Length:		11
Type:		24
Contents:	020101020010040000000000000000
Unknown descriptor:
Length:		1A
Type:		24
Contents:	0602010F0000000F0000000F0000000F0000000F00000000
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03030101000210000000
Unknown descriptor:
Length:		11
Type:		24
Contents:	020401010010040000000000000000
Unknown descriptor:
Length:		1A
Type:		24
Contents:	0605040F0000000F0000000F0000000F0000000F00000000
Unknown descriptor:
Length:		0C
Type:		24
Contents:	03060103000510000000

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		01
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>
Unknown descriptor:
Length:		10
Type:		24
Contents:	0103000101000000040000000000
Unknown descriptor:
Length:		06
Type:		24
Contents:	02010418

Endpoint descriptor:
Endpoint address:	83
Attr.:			0D(Isochronous)
Max.pkt size:		0300
Polling interval:	04
Unknown descriptor:
Length:		08
Type:		25
Contents:	010000000000

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		00
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>

Interface descriptor:
Intf.number:		02
Alt.:			01
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		02
Intf. Protocol:		20
Intf.string:		00

<<<AUDIO_STREAMING>>>
Unknown descriptor:
Length:		10
Type:		24
Contents:	0104000101000000040000000000
Unknown descriptor:
Length:		06
Type:		24
Contents:	02010418

Endpoint descriptor:
Endpoint address:	01
Attr.:			05(Isochronous)
Max.pkt size:		0310
Polling interval:	04
Unknown descriptor:
Length:		08
Type:		25
Contents:	010000000000

Endpoint descriptor:
Endpoint address:	81
Attr.:			11(Isochronous)
Max.pkt size:		0003
Polling interval:	04

Interface descriptor:
Intf.number:		04
Alt.:			00
Endpoints:		02
Intf. Class:		01
Intf. Subclass:		03
Intf. Protocol:		00
Intf.string:		00

<<<MIDIStreaming>>>

MS Interface descriptor
07240100014100
DescriptorSubtype: 01
USB_DESCRIPTOR_CS_INTERFACE :	MS_HEADER
MIDIStreaming SubClass Specification Release number: 0100
wTotalLength:		0041

MS Interface descriptor
062402010100
DescriptorSubtype: 02
USB_DESCRIPTOR_CS_INTERFACE :	MIDI_IN_JACK(Embedded)
JackType:	01
JackID:		01

MS Interface descriptor
062402020200
DescriptorSubtype: 02
USB_DESCRIPTOR_CS_INTERFACE :	MIDI_IN_JACK(External)
JackType:	02
JackID:		02

MS Interface descriptor
092403010301020100
DescriptorSubtype: 03
USB_DESCRIPTOR_CS_INTERFACE :	MIDI_OUT_JACK(Embedded)
JackType:	01
JackID:		03
NrInputPins:	01
020100

MS Interface descriptor
092403020401010100
DescriptorSubtype: 03
USB_DESCRIPTOR_CS_INTERFACE :	MIDI_OUT_JACK(External)
JackType:	02
JackID:		04
NrInputPins:	01
010100

Endpoint descriptor:
Endpoint address:	02
Attr.:			02(Bulk)
Max.pkt size:		0200
Polling interval:	00

MS Endpoint descriptor
0525010101
USB_DESCRIPTOR_CS_ENDPOINT :	MS_GENERAL
bNumEmbMIDIJack: 01
01

Endpoint descriptor:
Endpoint address:	82
Attr.:			02(Bulk)
Max.pkt size:		0200
Polling interval:	00

MS Endpoint descriptor
0525010103
USB_DESCRIPTOR_CS_ENDPOINT :	MS_GENERAL
bNumEmbMIDIJack: 01
03

Addr:1(0.0.1)

Thank you for your time!

YuuichiAkagawa, 2022-07-04:
See https://github.com/YuuichiAkagawa/USBH_MIDI/issues/45

karozas, 2022-07-04:
Thanks for the response.I will check it.

karozas, 2022-07-04:
`#include <usbh_midi.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

#define _MIDI_SERIAL_PORT Serial

uint8_t reqpatch_NUMBER[] {
   0xf0, 0x43, 0x58, 0x70, 0x15, 0x00,  0xf7
};

uint8_t reqpatch_DATA[] {
   0xF0, 0x43, 0x58, 0x70, 0x0A, 0x00,  0xF7
  };

  USB Usb;
  USBH_MIDI  Midi(&Usb);
  uint8_t phase = 0;

  void setup()
  {
    _MIDI_SERIAL_PORT.begin(115200);

    //Workaround for non UHS2.0 Shield
    pinMode(7, OUTPUT);
    digitalWrite(7, HIGH);

    if (Usb.Init() == -1) {
      while (1); //halt
    }//if (Usb.Init() == -1...
    delay( 200 );
  }

  void loop()
  {
    Usb.Task();
    if ( Midi ) {
      MIDI_poll();
      if ( phase == 0 ) {
        _MIDI_SERIAL_PORT.println("\nRequest: reqpatch_NUMBER");
        Midi.SendSysEx(reqpatch_NUMBER, sizeof(reqpatch_NUMBER));
        phase++;
        delay(10);
      } else if ( phase == 1 ) {
        _MIDI_SERIAL_PORT.println("\nRequest: reqpatch_DATA");
        Midi.SendSysEx(reqpatch_DATA, sizeof(reqpatch_DATA));
        phase++;
        delay(10);
      }
    }
  }

  // Poll USB MIDI Controler
  void MIDI_poll()
  {
    uint8_t size;
    uint8_t outBuf[4];
    uint8_t sysexBuf[3];
    char buf[20];

    do {
      if ( (size = Midi.RecvRawData(outBuf)) > 0 ) {
        uint8_t rc = Midi.extractSysExData(outBuf, sysexBuf);
        if ( rc != 0 ) {
          //SysEx
          _MIDI_SERIAL_PORT.print("SysEx:");
          for (int i = 0; i < rc; i++) {
            sprintf(buf, " %02X", sysexBuf[i]);
            _MIDI_SERIAL_PORT.print(buf);
          }

        } else {
          _MIDI_SERIAL_PORT.print("_____:");
          for (int i = 1; i <= size; i++) {
            sprintf(buf, " %02X", outBuf[i]);
            _MIDI_SERIAL_PORT.print(buf);
          }
        }
        _MIDI_SERIAL_PORT.println();
      }
    } while (size > 0);
  }`

karozas, 2022-07-04:
I 've loaded this example sketch : 

`#include <usbh_midi.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

#define _MIDI_SERIAL_PORT Serial

uint8_t reqpatch_NUMBER[] {
   0xf0, 0x43, 0x58, 0x70, 0x15, 0x00,  0xf7
};

uint8_t reqpatch_DATA[] {
   0xF0, 0x43, 0x58, 0x70, 0x0A, 0x00,  0xF7
  };

  USB Usb;
  USBH_MIDI  Midi(&Usb);
  uint8_t phase = 0;

  void setup()
  {
    _MIDI_SERIAL_PORT.begin(115200);

    //Workaround for non UHS2.0 Shield
    pinMode(7, OUTPUT);
    digitalWrite(7, HIGH);

    if (Usb.Init() == -1) {
      while (1); //halt
    }//if (Usb.Init() == -1...
    delay( 200 );
  }

  void loop()
  {
    Usb.Task();
    if ( Midi ) {
      MIDI_poll();
      if ( phase == 0 ) {
        _MIDI_SERIAL_PORT.println("\nRequest: reqpatch_NUMBER");
        Midi.SendSysEx(reqpatch_NUMBER, sizeof(reqpatch_NUMBER));
        phase++;
        delay(10);
      } else if ( phase == 1 ) {
        _MIDI_SERIAL_PORT.println("\nRequest: reqpatch_DATA");
        Midi.SendSysEx(reqpatch_DATA, sizeof(reqpatch_DATA));
        phase++;
        delay(10);
      }
    }
  }

  // Poll USB MIDI Controler
  void MIDI_poll()
  {
    uint8_t size;
    uint8_t outBuf[4];
    uint8_t sysexBuf[3];
    char buf[20];

    do {
      if ( (size = Midi.RecvRawData(outBuf)) > 0 ) {
        uint8_t rc = Midi.extractSysExData(outBuf, sysexBuf);
        if ( rc != 0 ) {
          //SysEx
          _MIDI_SERIAL_PORT.print("SysEx:");
          for (int i = 0; i < rc; i++) {
            sprintf(buf, " %02X", sysexBuf[i]);
            _MIDI_SERIAL_PORT.print(buf);
          }

        } else {
          _MIDI_SERIAL_PORT.print("_____:");
          for (int i = 1; i <= size; i++) {
            sprintf(buf, " %02X", outBuf[i]);
            _MIDI_SERIAL_PORT.print(buf);
          }
        }
        _MIDI_SERIAL_PORT.println();
      }
    } while (size > 0);
  }`

and I get this in serial monitor:

Request: reqpatch_NUMBER
SysEx: F0 43 58
SysEx: 70 15 02
SysEx: 01 03 64
SysEx: 00 00 00
SysEx: 00 00 F7

Request: reqpatch_DATA

karozas, 2022-07-08:
sorry I closed this case by mistake.Anyway I 've loaded the above  example sketch and I got this in serial monitor:

Request: reqpatch_NUMBER
SysEx: F0 43 58
SysEx: 70 15 02
SysEx: 01 03 64
SysEx: 00 00 00
SysEx: 00 00 F7

Request: reqpatch_DATA

the response in reqpatch_DATA should be something like this :

 F0 43 58 70 0A 02 00 01 02 41 01 0A 02 00 06 01 00 04
 07 00 0A 14 01 02 01 00 00 32 02 00 02 00 28 35 01 0A
 00 00 06 00 00 62 64 00 00 00 00 00 07 00 1C 41 00 48
 33 00 76 29 01 0C 4C 00 18 36 00 56 38 00 74 32 00 58
 34 00 60 2F 00 60 2F 00 5A 02 00 28 32 00 00 00 00 06
 45 00 2B 56 00 00 00 00 00 08 00 26 31 03 2C 45 01 19
 55 01 0C 34 00 06 0D 01 10 0D 00 00 06 00 00 00 00 64
 25 01 48 00 00 04 32 00 64 02 00 00 64 00 64 00 00 00
 62 00 00 05 00 00 01 00 04 03 00 12 04 00 14 06 00 0E
 08 00 16 43 01 28 2D 01 02 6D 01 44 69 01 4A 6E 01 68
 43 01 58 65 01 42 6E 00 00 01 00 10 02 00 02 00 00 00
 64 00 00 64 00 00 00 01 20 00 03 70 0B 03 70 0B 03 70
 0B F7
but I get nothing .I ve used ARUINO MEGA  and UNO with usb host shield and the results are the same.When I use serial MIDI and Raspberry pi as a usb host the device I built works fine.Thank you again and sorry for closing the issue by mistake.

YuuichiAkagawa, 2022-07-08:
Run with debugging enabled.
https://github.com/felis/USB_Host_Shield_2.0#enable-debugging

karozas, 2022-07-08:
I run it with with debugging enabled and this is what a get

MIDI Init
USBH_MIDI version 010000
Addr:01
VID:1FC9 PID:8260 #Conf:01

STEP1: MIDI Start
Interface descriptor:
 Inf#:		00
 Alt:		00
 EPs:		00
 IntCl:		01
 IntSubcl:	01
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		01
 Alt:		00
 EPs:		00
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		01
 Alt:		01
 EPs:		01
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		02
 Alt:		00
 EPs:		00
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		02
 Alt:		01
 EPs:		02
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		04
 Alt:		00
 EPs:		02
 IntCl:		01
 IntSubcl:	03
 Protocol:	00
+MIDI found

>Extracting endpoint
Endpoint descriptor:
 Length:	09
 Type:		05
 Address:	02
 Attributes:	02
 MaxPktSize:	0200
 Poll Intrv:	00
+valid EP found.
>Extracting endpoint
Endpoint descriptor:
 Length:	09
 Type:		05
 Address:	82
 Attributes:	02
 MaxPktSize:	0200
 Poll Intrv:	00
+valid EP found.
STEP1: MIDI,NumEP:03
Conf:01
EPin :82
EPout:02
Init done.

Request: reqpatch_NUMBER
SendSysEx:
	 Length:	0007
 wptr:	0C
SysEx: F0 43 58
SysEx: 70 15 02
SysEx: 02 03 64
SysEx: 00 00 00
SysEx: 00 00 F7

Request: reqpatch_DATA
SendSysEx:
	 Length:	0007
 wptr:	0C

YuuichiAkagawa, 2022-07-08:
Sorry, please enable EXTRADEBUG .
https://github.com/felis/USB_Host_Shield_2.0/blob/39ab2a6944e4cba9a08858b82d358be13bbd3cb0/usbh_midi.cpp#L29

karozas, 2022-07-08:
I enabled      EXTRADEBUG   and this the result:
⸮⸮B⸮⸮⸮R⸮
MIDI Init
USBH_MIDI version 010000
Addr:01
VID:1FC9 PID:8260 #Conf:01

STEP1: MIDI Start
Interface descriptor:
 Inf#:		00
 Alt:		00
 EPs:		00
 IntCl:		01
 IntSubcl:	01
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		01
 Alt:		00
 EPs:		00
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		01
 Alt:		01
 EPs:		01
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		02
 Alt:		00
 EPs:		00
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		02
 Alt:		01
 EPs:		02
 IntCl:		01
 IntSubcl:	02
 Protocol:	20
-MIDI not found

Interface descriptor:
 Inf#:		04
 Alt:		00
 EPs:		02
 IntCl:		01
 IntSubcl:	03
 Protocol:	00
+MIDI found

>Extracting endpoint
Endpoint descriptor:
 Length:	09
 Type:		05
 Address:	02
 Attributes:	02
 MaxPktSize:	0200
 Poll Intrv:	00
+valid EP found.
>Extracting endpoint
Endpoint descriptor:
 Length:	09
 Type:		05
 Address:	82
 Attributes:	02
 MaxPktSize:	0200
 Poll Intrv:	00
+valid EP found.
STEP1: MIDI,NumEP:03
Conf:01
EPin :82
EPout:02
Init done.
inTransfer():04

Request: reqpatch_NUMBER
SendSysEx:
	 Length:	0007
 Total pktSize:	0003
 wptr:	0C
SysEx: F0 43 58
SysEx: 70 15 02
SysEx: 02 03 64
SysEx: 00 00 00
SysEx: 00 00 F7
inTransfer():04

Request: reqpatch_DATA
SendSysEx:
	 Length:	0007
 Total pktSize:	0003
 wptr:	0C
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
inTransfer():0F
..................
inTransfer():0F       loops for ever

YuuichiAkagawa, 2022-07-09:
Page 36 of the [Programming Guide](https://pdfserv.stg-maximintegrated.com/en/an/AN3785.pdf)
0x0F  : hrBABBLE : Device talked too long

We suspect that the MG-30 is transmitting more than 64 bytes of data at a time.
Unfortunately, it is not supported by USB Host Shield.
You must select a microcontroller that supports USB 2.0 Hi-Speed. For example, Teensy 3.6/4.1.

karozas, 2022-07-09:
I see!Thank you very much  for your help.I will try a Teensy 3.6/4.1 when I have time.

closed 2022-07-09 by karozas