felis/USB_Host_Shield_2.0 #728: PS4 controller pairs through BT but won't connect afterwards.

vathomass, 2022-07-08:
Hi,

Firstly, many thanks for the great library. I opened this issue because I am facing the following connectivity problem when I try to connect a PS4 controller with an Arduino Uno through Bluetooth. In short, I am able to pair the two using `PS4BT PS4(&Btd, PAIR);` but I am unable to connect when using `PS4BT PS4(&Btd);`.

I am using the latest available library (version 1.6.2) and as for the hardware:
1. PS4 controller.
2. Arduino Uno R3 board.
3. USB host shield from [elecrow](https://www.elecrow.com/usb-host-shield-for-arduino-p-686.html?wiki).
4. Bluetooth dongle Esperanza EA159 USB Bluetooth 2.0 Adapter.

Steps I have tried:
1. PS4 controller works as expected when connected through USB cable to the host shield (tested using the `PS4USB.ino`).
2. PS4 controller works as expected when paired through Bluetooth (tested using the `PS4BT.ino` with the pairing option).
3. PS4 controller fails to connect through Bluetooth (tested using the `PS4BT.ino` without the pairing option).

Enabling the `ENABLE_UHS_DEBUGGING` in `setting.h` and the `EXTRADEBUG` in `BTD.cpp` produces the following output:
```
PS4 Bluetooth Library Started
BTD ConfigureDevice
BTD Init
Addr: 01
Endpoint descriptor:
Length:		07
Type:		05
Address:	81
Attributes:	03
MaxPktSize:	0040
Poll Intrv:	01
Endpoint descriptor:
Length:		07
Type:		05
Address:	82
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Endpoint descriptor:
Length:		07
Type:		05
Address:	02
Attributes:	02
MaxPktSize:	0040
Poll Intrv:	00
Bluetooth Dongle Initialized
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
Connection established
HCI Reset complete
Write class of device
Local Bluetooth Address: 33:03:30:09:96:89
HCI version: 06
Wait For Incoming Connection Request
Gamepad is connecting
Class of device: 00 25 08
Incoming Connection Request
Connection Failed: 10
```
After a little digging (and zero understanding of the Bluetooth stack), it seems that something goes wrong after the Bluetooth dongle changes state to `HCI_REMOTE_NAME_STATE`, asking for the controller name. For some reason, the controller responds to the name request (the handled event is `EV_REMOTE_NAME_COMPLETE`) with error `0x0C` (Command Disallowed according to the internet).

Any idea to what may be the issue or things that I can try to resolve it.

Best Regards,
Thomas

microbmen, 2023-07-31:
Same issue.  When it tries to connect after pairing I get a "PS" , the push of the PS button, and then the controller turns back off due to lack of connect.

microbmen, 2023-08-01:
Nevermind figured it out.  Kind of silly.  When it connect it registers a press of the PS button, and in the example code the PS Button performs a disconnect.

jimiwester, 2024-08-21:
> Hi,
> ...
> Connection Failed: 10
> ```
> 
> After a little digging (and zero understanding of the Bluetooth stack), it seems that something goes wrong after the Bluetooth dongle changes state to `HCI_REMOTE_NAME_STATE`, asking for the controller name. For some reason, the controller responds to the name request (the handled event is `EV_REMOTE_NAME_COMPLETE`) with error `0x0C` (Command Disallowed according to the internet).
> 
> Any idea to what may be the issue or things that I can try to resolve it.
> 
> Best Regards, Thomas

Found any solution to "Connection Failed: 10"?

vathomass, 2024-08-26:
> Found any solution to "Connection Failed: 10"?

Hi, not really. I pair the controller every time I want to use it.

Best