felis/USB_Host_Shield_2.0 #733: Failed board_qc

Hurtigmus, 2022-07-22:
I am trying to the use a USB Host shield with an Arduino Uno, but when I run the board_qc it gets stuck at 

Checking USB device communication.

Waiting for device...

I have a bluetooth dongle that should be compatible connected to the board.

How can I solve this problem?

OwenKruse, 2023-02-16:
I have the same issue on the Leonardo and Uno with two different boards.

touchgadget, 2023-02-17:
Many UHS clone boards require some soldering. https://www.youtube.com/watch?v=1dxwU87GQVM

There may be other board problems so far without solutions. https://github.com/felis/USB_Host_Shield_2.0/issues/755 https://github.com/felis/USB_Host_Shield_2.0/issues/765

xxxajk, 2023-02-19:
That's what you get for a crappy clone. Don't buy knockoffs.