felis/USB_Host_Shield_2.0 #735: NACK while reading configuration descriptor

timo-schluessler, 2022-08-05:
`USB::getConfDescr` returned `4` = `hrNAK` while trying to read configuration descriptor of METAPACE QR-Code scanner. It turned out that the device returned the NAK because `USB::ctrlReq` requested too many bytes from the device: `nbytes` instead of `total`. Probably this is a bug introduced [here](https://github.com/felis/USB_Host_Shield_2.0/commit/2e52a1dfa66680d6e1a770c9b9840ffc3150b27d#diff-61af881f3df93abb980b039ff74b1ce8f1063830418511fb0b5df167c2c3503cR152). Uncommenting the marked line (and outcommenting the previous one) fixed the issue.

xxxajk, 2022-08-08:
I haven't used or done any dev on 2.0 since using 3.0... depending on the MCU you are using, can you see if UHS3 does the correct thing?

timo-schluessler, 2022-08-09:
I was using a M5Core2. It seems UHS3 doesn't support it yet? Unfortunately I don't have much time left over to test this. But if someone could supply a UHS3 example working with M5Core2 (ESP32) I would give it another try.

xxxajk, 2022-08-10:
The problem with esp32/8266 currently (and a few others too, such as Spresense) is properly supporting pin interrupts, and hooking into the IRQ at a low level.

The three boards *DO* have the RTOS running in the BG, which could help with the servicing of the USB connections. Problem is that the service routines still must be able to service ISR at the same time that it is servicing the USB connections. I'm not exactly certain how well an RTOS would perform. Depends on what's on-chip... e.g. spresense has 3 cores...
Other platforms "cheat" to do this... AVR allows reentrant IRQ, and that's abused that way. 
ARM and MIPS steals an NVIC IRQ and fires off a hardware interrupt as a software interrupt... messy, but works.