felis/USB_Host_Shield_2.0 #742: Stepper motor with a Thrustmaster USB Joystick

Andrjazz, 2022-09-30:
I am trying to take the reading from a Thrustmaster USB Joystick and mapping it to two stepper motors(one of the for the x motion, the other for the y motion), so that the position of the Joystick corresponds to some angular velocity at which the motors turn.

I am currently using a Usb Host Shield 2.0 on an Arduino Mega, and I am using the UBS_Host_Shield_2.0 library. I am trying to modify the USBHIDJoystick example in this library to accomplish my goals. I have added a section where I included the AccelStepper library in order to drive the motors, and this code is posted below.

I am trying to get the motor to turn continuously at the constant speed the constant joystick position corresponds to. I have noticed both through the serial monitor and through the performance of the motors themselves that if I hold the joystick in some constant position, only the first reading in that position gets printed to the monitor, and the motor is actuated for a single pulse, even if I am continuing to hold the joystick position. These readings are then followed by readings which are identical to the readings I get from an un-actuated joystick. This response makes me believe that somewhere in the UBS_Host_Shield_2.0 library, there is a section of code where the previous usb signal is compared to the current one, and if it is not different, it starts the entire loop again to inspect the usb signal reading, without forwarding and mapping the signal to a value to drive the motors.

hidjoystickrptparser.cpp

void JoystickEvents::OnGamePadChanged(const GamePadEventData *evt) {
  
        stepper.setEnablePin(ENA);
        stepper.enableOutputs();
        stepper.setMaxSpeed(SPEED);
        stepper.setAcceleration(1500);
              
        Serial.print("X1: ");
        Serial.print(evt->X, DEC);
        if (evt->X >= 10){
          stepper.move(+setSPEED) ;
          stepper.run();
        }
        else if (evt->X == 0){
          stepper.stop();
          stepper.runToPosition();
          stepper.setCurrentPosition(0);
        }
        
        Serial.print("\tY1: ");
        Serial.print(evt->Y, DEC);
        Serial.print("\tX2: ");
        Serial.print(evt->Z1, DEC);
        Serial.print("\tY2: ");
        Serial.print(evt->Z2, DEC);
        Serial.print("\tRz: ");
        Serial.print(evt->Rz, DEC);
        Serial.println("");
        }

how to make it constantly to process the signal?

Help Please

touchgadget, 2022-09-30:
There is an example for a Thrustmaster T.16000 joystick. https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/t16km/t16km.ino

Change the code that ignores duplicate readings. https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/t16km/t16km.ino#L55

Andrjazz, 2022-10-03:
> There is an example for a Thrustmaster T.16000 joystick. https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/t16km/t16km.ino
> 
> Change the code that ignores duplicate readings. https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/HID/t16km/t16km.ino#L55

thanks for the answer

I changed the code like in your example (Thrustmaster T.16000 joystick.) it work, values ​​are constantly updated. But 
when i change the position of the joystick, the values ​​don't change
When I add in sketch  uint8_t oldHat;  values change when I change position joystick and stop when value dont change.
How it fixed?

what to do with it  uint8_t oldHat ?

`#include <usbhid.h>
#include <hiduniversal.h>
#include <usbhub.h>

#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

// Thrustmaster T.16000M HID report
struct GamePadEventData {
  
        uint8_t X, Y, Z1, Z2, Rz; 
          
};

class JoystickEvents {
public:
        virtual void OnGamePadChanged(const GamePadEventData *evt);
};

USB Usb;
USBHub Hub(&Usb);
HIDUniversal Hid(&Usb);
JoystickEvents JoyEvents;

#define RPT_GAMEPAD_LEN  5

class JoystickReportParser : public HIDReportParser{
        JoystickEvents    *joyEvents;

        uint8_t oldPad[RPT_GAMEPAD_LEN];
        uint8_t oldHat;

public:
        JoystickReportParser(JoystickEvents *evt);

        virtual void Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf);
};

JoystickReportParser Joy(&JoyEvents);

JoystickReportParser::JoystickReportParser(JoystickEvents *evt) :
joyEvents(evt)
{}

void JoystickReportParser::Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf) {
  
       bool match = (sizeof(oldPad) == len) && (memcmp(oldPad, buf, len) == 0);
  
    if (!match && joyEvents) {
     joyEvents->OnGamePadChanged((const GamePadEventData*)buf);
     memcpy(oldPad, buf, len);
     
    }
    
}

void JoystickEvents::OnGamePadChanged(const GamePadEventData *evt) {
  
        Serial.print("X1: ");
        PrintHex<uint8_t > (evt->X, 0x80);
        Serial.print("\tY1: ");
        PrintHex<uint8_t > (evt->Y, 0x80);
        Serial.print("\tX2: ");
        PrintHex<uint8_t > (evt->Z1, 0x80);
        Serial.print("\tY2: ");
        PrintHex<uint8_t > (evt->Z2, 0x80);
        Serial.print("\tRz: ");
        PrintHex<uint8_t > (evt->Rz, 0x80);
        Serial.println("");
}

void setup()
{
               
  Serial.begin( 19200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
      Serial.println("OSC did not start.");

  delay( 200 );

  if (!Hid.SetReportParser(0, &Joy))
      ErrorMessage<uint8_t>(PSTR("SetReportParser"), 1  );
}

void loop()
{
    Usb.Task();
}

`

If you can please help?

touchgadget, 2022-10-03:
Try the following which does not discard duplicate HID reports. Your code has many problems. For example, the t16k returns 16 bit values for X,Y but your code uses 8 bit variables.
```
/* Simplified Thrustmaster T.16000M FCS Joystick Report Parser */

#include <usbhid.h>
#include <hiduniversal.h>
#include <usbhub.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>

// Thrustmaster T.16000M HID report
struct GamePadEventData
{
  uint16_t  buttons;
  uint8_t   hat;
  uint16_t  x;
  uint16_t  y;
  uint8_t   twist;
  uint8_t   slider;
}__attribute__((packed));

class JoystickEvents
{
  public:
    virtual void OnGamePadChanged(const GamePadEventData *evt);
};

#define RPT_GAMEPAD_LEN	sizeof(GamePadEventData)

class JoystickReportParser : public HIDReportParser
{
  JoystickEvents		*joyEvents;

  public:
  JoystickReportParser(JoystickEvents *evt);

  virtual void Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf);
};

JoystickReportParser::JoystickReportParser(JoystickEvents *evt) :
  joyEvents(evt)
{}

void JoystickReportParser::Parse(USBHID *hid, bool is_rpt_id, uint8_t len, uint8_t *buf)
{
  joyEvents->OnGamePadChanged((const GamePadEventData*)buf);
}

void JoystickEvents::OnGamePadChanged(const GamePadEventData *evt)
{
  Serial.print("X: ");
  PrintHex<uint16_t>(evt->x, 0x80);
  Serial.print(" Y: ");
  PrintHex<uint16_t>(evt->y, 0x80);
  Serial.print(" Hat Switch: ");
  PrintHex<uint8_t>(evt->hat, 0x80);
  Serial.print(" Twist: ");
  PrintHex<uint8_t>(evt->twist, 0x80);
  Serial.print(" Slider: ");
  PrintHex<uint8_t>(evt->slider, 0x80);
  Serial.print(" Buttons: ");
  PrintHex<uint16_t>(evt->buttons, 0x80);
  Serial.println();
}

USB                                             Usb;
USBHub                                          Hub(&Usb);
HIDUniversal                                    Hid(&Usb);
JoystickEvents                                  JoyEvents;
JoystickReportParser                            Joy(&JoyEvents);

void setup()
{
  Serial.begin( 115200 );
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  Serial.println("Start");

  if (Usb.Init() == -1)
    Serial.println("OSC did not start.");

  delay( 200 );

  if (!Hid.SetReportParser(0, &Joy))
    ErrorMessage<uint8_t>(PSTR("SetReportParser"), 1  );
}

void loop()
{
  Usb.Task();
}
```

Andrjazz, 2022-10-04:
> Try the following which does not discard duplicate HID reports. Your code has many problems. For example, the t16k returns 16 bit values for X,Y but your code uses 8 bit variables.
>
> ```
> /* [...]

Lauszus, 2022-10-20:
Seems like this issue is solved, so I will close this issue.

closed 2022-10-20 by Lauszus