felis/USB_Host_Shield_2.0 #743: How I can get ps4 USB feature report?

ullMarksin, 2022-10-09:
How I can get ps4 USB feature report?  For example I want to retrieve ps4 controller internally saved mac address via get feature report with id 0x12, but I can't figure out how, because I can't find HID_GET_REPORT or similar method. I tried receive report via:

pUsb->ctrlReq(bAddress, epInfo[0].epAddr, bmREQ_HID_IN, HID_REQUEST_GET_REPORT, 0x12, 0x03, 0x00, sizeof(buf), sizeof(buf), buf, NULL);

But it didn't work for me.

Lauszus, 2022-10-20:
Where did you get the information for the report ID etc? Please share the link.

ullMarksin, 2022-10-20:
https://github.com/ViGEm/ViGEmBus/blob/master/sys/Ds4Pdo.cpp#L545

ullMarksin, 2022-10-20:
What I am trying to do is get saved in ps4 controller mac address, like in this code , shown how to get ps4 auth tokens: 
https://github.com/mathijsvandenberg/g29emu/blob/94b6bb941189a034b94ad3571fd51e5392d19bf7/PS4AuthDevice.cpp#L75