felis/USB_Host_Shield_2.0 #745: How to use / which example for sending Keyboard Key to PC?

christophepersoz, 2022-10-19:
Hi all,

Thanks everyone whom is ready to take time to read and answer me. I would like to have my ESP-32 board + MAX3421E module connected on it acting as a HID Keyboard when connected onto a PC. Then I would like that ESP software send a key pressed code to the PC.
I assume that is possible with this library from what I read, but can anybody tell me which example I have to follow to reach this goal?

Thanks a lot for your help, and sorry for this low level question 😉

touchgadget, 2022-10-19:
An ESP32 + MAX3421E cannot act as a USB keyboard. An ESP32S2 or ESP32S3 can do this and do not need the MAX3421E.

https://github.com/espressif/arduino-esp32/blob/master/libraries/USB/examples/Keyboard/KeyboardSerial/KeyboardSerial.ino

touchgadget, 2022-10-19:
I do not know of any way but I have not looked into this. You could do more web searching for "esp32 bit bang usb" and/or search forums on esp32.com and reddit.com/r/esp32.

Lauszus, 2022-10-20:
I'll close this issue, as it's unrelated to this library.

This library is for USB host functionality.

What you need is to emulate a USB device.

I would recommend you take a look at Teensy. They make it super easy to emulate a USB device: https://www.pjrc.com/teensy/td_keyboard.html

closed 2022-10-20 by Lauszus

christophepersoz, 2022-10-20:
Indeed, it is not related. I do know Teensy boards but they are all sold out since a year now, except 4.x versions and oversized to emulated a USB device. I used few of them by the past to emulate MIDI device and indeed, it's super easy to do.
It's so inconvenient that ESP are not able to handle this. Thanks!