felis/USB_Host_Shield_2.0 #746: USB host shield only recognizes Zoom G1on when powered by USB

Fernandoqsiqueira, 2022-10-24:
Hello

 I'm building a midi controller using Arduino Uno and USB host shield for control a Zoom g1xOn, but the midi command only is sent sucessfully when the arduino is powered by 5V usb. When I use a 9v power supply, the host shield doesn`t recognize the Zoom g1xOn. I need use the 9v power supply because using the 5v usb power supply i get 400 mA for Zoom, when its need 500ma

this is a code, in case to be a code problem

````
#include <MIDI.h>
#include <Usb.h>
#include <usbhub.h>
#include <usbh_midi.h>

MIDI_CREATE_DEFAULT_INSTANCE();
USB Usb;
USBH_MIDI Midi(&Usb);
const byte pushButtons[] = {2,3,4,6}; //Número de Switches
const byte leds[] = {16,17,18,19}; //Número de of leds
const byte outPin5 = 5;
byte buttonPushCounter[] = {0,0,0,0};   // contador para o número de vezes que o botão foi pressionado
static bool prevButtonState[4] = {};  // estado anterior do botao
bool buttonState;
bool tunerOn;

const byte pinBank = 7;
const byte pinLedBank = 15;
bool secondBank;
const byte fav = 8; // valor do preset favorito, no caso da g1On, seria o A8

void setup()   

{
  for(byte m=0; m<4 ;m++)
  {
    pinMode(pushButtons[m], INPUT_PULLUP);
    pinMode(leds[m], OUTPUT);
  }
  pinMode(13, OUTPUT);
  digitalWrite(13, HIGH);
  pinMode(15, OUTPUT);
  pinMode(pinBank, INPUT_PULLUP);
  if (Usb.Init() == -1) //indispensable para MS70
   {
    digitalWrite(outPin5, HIGH);
    while(1);  
    
    
     }             // Halt
  Serial.begin(31250);
  MIDI.begin(MIDI_CHANNEL_OMNI);
  intro();
  tunerOn = false;
  secondBank = false;
}
/**********************************************************************/
void resetAllLeds()
{
  for(byte k=0; k<4; k++)
  {
    digitalWrite(leds[k], LOW);
   }
  digitalWrite(13, LOW);
}
/**********************************************************************/
void resetCounter()
{
  for(byte j=0; j<4; j++)
  {
    buttonPushCounter[j]= 0;
  }
}  
/**********************************************************************/
void SendMIDI(byte number)
{
  Usb.Task();
  //Se o controlador reconhece o dispositivo perfeitamente, então o comando midi é enviado
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {
    SendMIDIintro(number); //
    
    //MIDI.sendControlChange(12,number, 0);
   } 
   
   //este estado do host acontece quando o arduino/host não reconheceu a pedaleira, serve como indicador
   //this host state happans when the arduino/host doesn't reconized the zoom g1xon, is an indicator
   if( Usb.getUsbTaskState() == USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE ){
          resetCounter();
          resetAllLeds();
          
            for (byte p = 0; p < 4; p++)
  {
          digitalWrite(leds[p], HIGH);
          delay(50);
          digitalWrite(leds[p], LOW);
          delay(50);
  }   
    
   }
}
/**********************************************************************/
void loop() //ok
{

  while(!secondBank){
  
  for (byte i = 0; i < 4; i++)
  {
    
    buttonState = !digitalRead(pushButtons[i]); // 
    if (buttonState != prevButtonState[i])
    {
      
      prevButtonState[i] = buttonState;
      if (buttonState)
      {
        
        if (buttonPushCounter[i] == 0)
        {
          
          SendMIDI(i);
          resetCounter();
          resetAllLeds();
          buttonPushCounter[i]++;
          ligarLed(leds[i]);
        }
        else
        {
          
          SendMIDI(fav);
          resetCounter();
          resetAllLeds();
          digitalWrite(13, HIGH);
        }
      }
      delay(420);  // anti rebond
    }
    }
    if(!digitalRead(pinBank)){
      secondBank = !secondBank;
      resetCounter();
      digitalWrite(pinLedBank, HIGH);
      delay(420);
      
    }
    
  }

  while(secondBank){
  
  for (byte i = 0; i < 4; i++)
  {
    
    buttonState = !digitalRead(pushButtons[i]); // 
    if (buttonState != prevButtonState[i])
    {
      
      prevButtonState[i] = buttonState;
      if (buttonState)
      {
        
        if (buttonPushCounter[i] == 0)
        {
          
          SendMIDI(i+4);
          resetCounter();
          resetAllLeds();
          buttonPushCounter[i]++;
           ligarLed(leds[i]);
        }
        else
        {
          
          SendMIDI(fav);
          resetCounter();
          resetAllLeds();
          digitalWrite(13, HIGH);
        }
      }
      delay(420);  // anti rebond
    }
    }
    if(digitalRead(pinBank)){
      secondBank = !secondBank;
      resetCounter();
      digitalWrite(pinLedBank, LOW);
      delay(420);
      
    }
    
  }

}

//Forçar a comunicão via usb
//Aguardar pedal inicializar
// Tempo de iniciar baseado na Zoom G1_On
void intro() {

// tempo de inicializaçao da G1On
delay(6000); 

  for(byte pos=0;pos< 4; pos++){
    SendMIDIintro(pos); 
     digitalWrite(leds[pos], HIGH);
     delay(350);
     digitalWrite(leds[pos], LOW);
  }

  SendMIDIintro(fav); 
  delay(300);
  
  for (int b=0;b<8; b++){
    for(byte k=0;k<4;k++){
     digitalWrite(leds[k], HIGH);
  }
  delay(75);
  
  for(byte l=0;l<4;l++){
     digitalWrite(leds[l], LOW);
  }
  delay(75); 
  }
 
}

void SendMIDIintro(byte number)
{
  Usb.Task();
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {
    byte Message[2];                 // Construct the midi message (2 bytes)
    Message[0]=0xC0;                 // 0xC0 is for Program Change
    Message[1]= number;            // Number is the program/patch
    Midi.SendData(Message);          // Send the message
    
    //MIDI.sendControlChange(12,number, 0);
   } 

}

void ligarLed(byte pinPort){
  digitalWrite(pinPort, HIGH);
  delay(40);
  digitalWrite(pinPort, LOW);
  delay(40);
  digitalWrite(pinPort, HIGH);
}

touchgadget, 2022-10-25:
Plug a power adapter into the Zoom device so it does not draw current from the USB host shield. The specs says the Zoom device needs 9V DC @ 500mA. Or use 4AA batteries.

Fernandoqsiqueira, 2022-10-25:
> Plug a power adapter into the Zoom device so it does not draw current from the USB host shield. The specs says the Zoom device needs 9V DC @ 500mA. Or use 4AA batteries.

I already try using a 9vDC 1A but the host don't recognize the Zoom, I found a 9vDC 600mA and tried now supply the zoom , but the Zoom receive the midi message during some time only, even if  Usb.getUsbTaskState() == USB_STATE_RUNNING. the zoom only works normally if It is supplied only by usb host shiled

YuuichiAkagawa, 2022-10-25:
I tested your code with G1on.  
The 9V power supply used for this test is a Roland PSB-1U (2A).

1. USB bus powerd
It seems to work.
[image]

2. Supply 9V only to G1on.
It does not work. We need to power the Arduino.
[image]

3. Supply 9V to G1on and 5V to Arduino.
It seems to work.
[image]

Which case are you trying?

Fernandoqsiqueira, 2022-10-25:
> I tested your code with G1on. The 9V power supply used for this test is a Roland PSB-1U (2A).
>
> 1. USB bus powerd
>    It seems to work.
>    [image]
> 2. Supply 9V only to G1on.
>    It does not work. We [...]

I'm trying the 3th case. the problem occurs when i change de state of pin bank (in this case the pin 15) from high to low and vice versa. when it happens, the zoom stop the receiving of midi data. I can't understand why  It happens, because works normally when the zoom is only supply by arduino only ( 1st case)

YuuichiAkagawa, 2022-10-25:
In my environment there does not appear to be any difference in operation based on the power supply type.

I think that Usb.Task() should be executed in loop() instead of SendMIDI() or SendMIDIintro().

```
void loop() //ok
{

  while(!secondBank){
  Usb.Task();
```

```
  while(secondBank){
  Usb.Task();
  for (byte i = 0; i < 4; i++)
```

Fernandoqsiqueira, 2022-10-25:
> No meu ambiente, não parece haver nenhuma diferença na operação com base no tipo de fonte de alimentação.
> 
> Eu acho que Usb.Task() deve ser executado em loop() em vez de SendMIDI() ou SendMIDIintro().
> 
> ```
> void loop() //ok
> {
> 
>   while(!secondBank){
>   Usb.Task();
> ```
> 
> ```
>   while(secondBank){
>   Usb.Task();
>   for (byte i = 0; i < 4; i++)
> ```

Can you explain exactly how works the function Usb.Task() ? I has implemented in this way because I'm not sure how It works

YuuichiAkagawa, 2022-10-25:
https://github.com/felis/USB_Host_Shield_2.0/blob/39ab2a6944e4cba9a08858b82d358be13bbd3cb0/Usb.cpp#L459-L460
First, the connection will not be completed unless Usb.Task() is executed until USB_STATE_RUNNING is reached.  
Reconnection when the state changes does not recover unless Usb.Task() is executed until USB_STATE_RUNNING.

Every example should run Usb.Task() at the beginning of loop().

Fernandoqsiqueira, 2022-10-26:
> 

> https://github.com/felis/USB_Host_Shield_2.0/blob/39ab2a6944e4cba9a08858b82d358be13bbd3cb0/Usb.cpp#L459-L460
>
>
> First, the connection will not be completed unless Usb.Task() is executed until [...]

Thanks for explaining!

After some hours and many tests, I found a soluction to problem and was in hardware project: 
when I declared a pin to use as bank indicator and a pin to led bank indicator, if use a spst footswitch to change the pin bank state and change de state of led indicator from High to Low. it doesn't work more. I used a 2pdt and 2 leds, in this case a pin led every is HIGH, only switch the led selected. Now I can use the controller normally and the zoom with power supply

Thank you very much!