felis/USB_Host_Shield_2.0 #747: List of compatible Bluetooth dongles out of date

AJ292, 2022-10-25:
The list of compatible Bluetooth dongles in the documentation (https://github.com/felis/USB_Host_Shield_2.0/wiki/Bluetooth-dongles) is fairly out of date. Nearly all the links listed no longer work, while searching for the models listed instead yields few results as most just aren't available to purchase anymore.

I've done some testing with some random, currently available Bluetooth dongles floating around and while I've had success getting a single controller paired and running with them, trying to get two controllers going simultaneously (specifically two PS3 Nav controllers) isn't working at all. Apologies if I'm combining two issues here, but I'm not sure if there's a problem with the compatibility of the dongles I'm using or if it's something in the code itself.

xxxajk, 2022-10-25:
The problem is because...
1: Newer bluetooth devices require closed source firmware to be uploaded (some how, nobody knows how) to the dongle, and...

2: There isn't enough room in flash or ram to upload said firmware, and...

3: There would be legal issues if we could upload the firmware, because we can't include it.

USB is a heavy lift for limited resource MCU's ...basically anything less than 32k RAM. It already doesn't allow much to be done on any AVR, even the mega2560 struggles without adding expensive RAM expansion. Much of the code in UHS2 cheats on the specs, and technically shouldn't work, and only does because you select only one thing to work with.

This is exactly why I wrote UHS3, which doesn't work for most things on an AVR at all because it doesn't have enough RAM to follow the USB spec properly and in full, and most people don't understand, or don't want to understand, or have different misguided understandings on how to interact with USB devices.
That all said...
The only solution to number 2 as far as flash/ram excludes all AVR, so people would STILL complain about it, could be done on UHS3, however see number 3... Most people aren't capable of goofing with it for simple projects. Use an ESP32, or something with opensource/allowed binary blobs BT included in the design.

Any dongle that does NOT require a firmware upload to enable it to work, will work. If you find some, you are welcome to add them to the list.

AJ292, 2022-10-25:
Ok, this makes a LOT of sense. It did feel like a lack of RAM was preventing me from connecting the second controller to the Arduino.

Unfortunately, at this stage, I can't realistically switch to an ESP32 (working on a pre-established project that utilises the Mega and the dual PS3 Nav setup) so I'm in a bit of a bind. I've managed to track down the TRENDnet TBW-106UB that's listed in the specs (which is randomly shipping from the Netherlands) but I have no idea if it will work or not. 

Out of curiosity, how much RAM would need to be added to the Mega, theoretically, to get this working? The QuadRAM expansion (https://www.rugged-circuits.com/new-products/quadram) adds 512 kilobytes to the board and would only require minor rewriting to the code I'm using to get it integrated into the system (just need to shift a couple of pins).

xxxajk, 2022-10-25:
Quadram + XMEM2,  which I wrote will not just work better, but provides preemptive multitasking as an added bonus.

https://github.com/xxxajk/xmem2

AJ292, 2022-10-25:
Amazing. Will grab the expansion and give it a whirl.

xxxajk, 2022-11-10:
Did some calculations, for two copies,  parts cost (rounded) is $60 + shipping of those parts, plus $20 for the boards. So you are looking at about $100 for two of them, if I hand make them.

AJ292, 2022-11-10:
Very much appreciate the offer, but I won't take you up on it at this time. The Bluetooth dongle I'm currently using does allow the connection of a single PS4 controller instead of two PS3 nav controllers, so I've decided to dive in and try and write my own code to use it instead. 

Again, many thanks for the offer and the clarity on the initial problem as well.

microbmen, 2023-08-16:
in my testing this guy works the best
https://www.amazon.com/dp/B007Q45EF4?psc=1&ref=ppx_yo2ov_dt_b_product_details

Kinivo USB Bluetooth Adapter for PC BTD400 (Bluetooth 4.0 Dongle Receiver, Low Energy)