felis/USB_Host_Shield_2.0 #752: OSC did not start

demorsick, 2022-11-19:
# problem:
after i connected my usb host shield with my arduino uno and i ran the sketch for PS4USB i see in the serial monitor
`OSC did not start`
and the controller doesn't even connect because it can't find the shield
--------------------------
### board info :
BN: Arduino Uno
VID: 2341
PID: 0043
SN: 75135333636351F07170
----------------------
### os info:
NAME="Fedora Linux"
VERSION="36 (Workstation Edition)"
ID=fedora
VERSION_ID=36
VERSION_CODENAME=""
PLATFORM_ID="platform:f36"
PRETTY_NAME="Fedora Linux 36 (Workstation Edition)"
ANSI_COLOR="0;38;2;60;110;180"
LOGO=fedora-logo-icon
CPE_NAME="cpe:/o:fedoraproject:fedora:36"
HOME_URL="https://fedoraproject.org/"
DOCUMENTATION_URL="https://docs.fedoraproject.org/en-US/fedora/f36/system-administrators-guide/"
SUPPORT_URL="https://ask.fedoraproject.org/"
BUG_REPORT_URL="https://bugzilla.redhat.com/"
REDHAT_BUGZILLA_PRODUCT="Fedora"
REDHAT_BUGZILLA_PRODUCT_VERSION=36
REDHAT_SUPPORT_PRODUCT="Fedora"
REDHAT_SUPPORT_PRODUCT_VERSION=36
PRIVACY_POLICY_URL="https://fedoraproject.org/wiki/Legal:PrivacyPolicy"
VARIANT="Workstation Edition"
VARIANT_ID=workstation
---------------------------------
[image]

xxxajk, 2022-11-30:
Please check that the off-brand (and possibly fake) shield has the correct chip on it.
see #744 
if the chip is correct, then return it for exchange or refund.

demorsick, 2022-11-30:
> off-brand (and possibly fake) shield 
i've brought 2 usb host shield from amazon,and they both had this problem,what should i buy?

touchgadget, 2022-11-30:
Your board may need 3 dots of solder. This is a common problem with UHS clone boards. See the video for the fix.

https://www.youtube.com/watch?v=1dxwU87GQVM

lhx2624, 2023-06-29:
I used aeduino uno and usb host shield to connect, the code I used was PS3 BT in the library function, but after using it, upload the serial port display surface to show me OSC did not start, how can I solve this