felis/USB_Host_Shield_2.0 #753: 1 of the 2 barcode readers don't get recognized despite them communicating the same way (Protocol analyzer report attached))

grayHRF, 2022-11-28:
I have 2 barcode readers with me: 
1. Henex HC-666
2. Henex HC-22

The HC-22 gets recognized by running USBDesc and USBHIDdesc but the HC-666 doesn't. The default example just stops after printing "Start".

I am using the USB Host Shield Mini with an Arduino Pro Mini. 

I used a USB protocol analyzer software and took the readings of the following actions for both scanners:
1. Scanner plug in
2. Scanning of a QR code with ID: 10215954
3. Removed the scanner from the computer's USB port. 

I am attaching both files here.

Can anyone please help me figure out what's different about the HC-66?  
I am unable to looking at the analyzer report. I used a diff tool and the only difference I could see was in the pipe handles. 

[HC22_scan.txt](https://github.com/felis/USB_Host_Shield_2.0/files/10104539/HC22_scan.txt)
[HC666_scan.txt](https://github.com/felis/USB_Host_Shield_2.0/files/10104540/HC666_scan.txt)

touchgadget, 2022-11-30:
If it is possible, power the barcode scanner separately. It may require more current than the Arduino can supply. Powering the Arduino board via its barrel connector may also help.

grayHRF, 2022-11-30:
> If it is possible, power the barcode scanner separately. It may require more current than the Arduino can supply. Powering the Arduino board via its barrel connector may also help.

Hey, I am using a separate 5V supply for the VBUS. And it works as intended for the other barcode scanner, mouse and keyboard.  

I dug a little further into this issue and I found the non working scanner (HC-666) does not get detected by the shield. 
I modified the default USBDesc code as follows:

```
 Usb.Task();

 Serial.println(Usb.getUsbTaskState());  //newly added line

  if ( Usb.getUsbTaskState() == USB_STATE_RUNNING )
  {
    Usb.ForEachUsbDevice(&PrintAllDescriptors);
    Usb.ForEachUsbDevice(&PrintAllAddresses);
  }

```

And from this I can see for other peripherals, the USB state changes but when I plug in HC-666, it stays at `USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE`
And sometimes it does change to
`USB_ATTACHED_SUBSTATE_SETTLE` 
for a very short time on plugging in before going back to wait (0x12) again

Any clue on how to resolve this?