felis/USB_Host_Shield_2.0 #755: Newbie help with USB HostShield and HID CNC Pendant

ColPaulR, 2022-12-11:
I'm trying to interface a USB CNC Pendant to my GRBL controller board's spare UART port. I can't get usb_task_state to get beyondUSB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE. I compiled the example USBHID_desc.ino and I get the "Start" message on USB monitor. I'm using an UNO with an external supply and a board from [AliExpress](https://www.aliexpress.us/item/3256804649395559.html). I've solder-jumped the 3.3V and 5V from the UNO to the shield and solder-jumped 5v to USB VBUS. 3.3V measures 3.9V on the shield and 5V is right at 5.0. The USB CNC Pendant powers up when I plug it in, but I get nothing beyond the above. Please help!

touchgadget, 2022-12-12:
Perhaps it is not a USB HID device. Run the example USB_desc and post the output.

ColPaulR, 2022-12-15:
I'm still learning, but what little I know I would not expect any change between an HID or non-HID device as the program never gets out of state USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE (starts in USB_DETACHED_SUBSTATE_INITIALIZE). Both programs expect MAX3421 to indicate that a device has been plugged in to move out of USB_DETACHED_SUBSTATE_INITIALIZE. I'm curious if the MAX3421 is not asserting an interrupt or if the MAX3421 is even responding over SPI. Unless someone has some ideas, I guess I'm going to have to put some print statements at the SPI returns from the MAX3421 to ensure that the host shield is even running.  I did confirm that if I remove the shield and run the program that I get the "Start\n OSC did not start." message, so I'm assuming that the SPI talking to a working MAX3421.

touchgadget, 2022-12-15:
If the board has never worked, try running board_qc. "OSC did not start." usually the CPU cannot communicate with the MAX3421.

3.9V seems high. 3.3V should very close to 3.3V such as +/- 5%.

Lots of debugging built in. See https://github.com/felis/USB_Host_Shield_2.0#enable-debugging

ColPaulR, 2022-12-15:
Agreed v3.9 is high.  Not sure what to do with that. 

I only get the “osc did not start” if I pull the shield and try running. I just wanted to confirm that the SPI was talking with the shield.  With the shield installed, it starts but doesn’t get past waiting for device to be plugged in. I’ve even tried other usb devices (thumb drive) to ensure that it isn’t some quirk in my pendant. 

Is there a way to not use the interrupts but rather have the software poll the registers?  I assume SPI is working since I get the start message. 

I’ve set  ENABLE_UHS_DEBUGGING but I don’t see anything different on the serial port.

touchgadget, 2022-12-15:
>  I assume SPI is working since I get the start message.
Start is printed before initializing the UHS so it does not mean much.

The last time I looked, Usb.Task() in loop is polling the MAX3421. Interrupts are not used.

Regarding 3.9V: Is it possible 3.3V is shorted to 5V? This may damage one or both boards. Cheap UHS boards usually require 3 solder bridges to work.

https://www.youtube.com/watch?v=1dxwU87GQVM

In the video at 48 seconds, shows the same fix I use for clone UHS boards.

touchgadget, 2022-12-15:
Given there are multiple problems with the board, I suggest returning it, if possible.