felis/USB_Host_Shield_2.0 #757: My mouse don't move

skhrlx, 2022-12-18:
1. It's not a problem with my arduino Leonardo + USB Host Shield 2.0
2. I'already soldered the USB Host Shield 2.0
3. The USB Host Shield 2.0 + Arduino Leonardo is working correctly with my keyboard

My mouse turn on his LEDS, but just don't move, I belive it's a problem with the library, my mouse: Pichau Hive S

The code I've been using:
```
#include <Mouse.h> 
#include <Wire.h> 
#include <SPI.h> 
#include <usbhub.h> 
USB     Usb; 
USBHub     Hub(&Usb); 
 
#include <hidboot.h> 
 
HIDBoot<USB_HID_PROTOCOL_MOUSE>    HidMouse(&Usb); 
 
String myString;  
int j = 0; 
int c = 0; 
int e = 0; 
int lmb = 0; 
int rmb = 0; 
int mmb = 0; 
int dx; 
int dy; 
int arr[2]; 
int arrv[8]; 
 
class MouseRptParser : public MouseReportParser 
 
{ 
 
  protected: 
 
    void OnMouseMove  (MOUSEINFO *mi); 
 
    void OnLeftButtonUp (MOUSEINFO *mi); 
 
    void OnLeftButtonDown (MOUSEINFO *mi); 
 
    void OnRightButtonUp  (MOUSEINFO *mi); 
 
    void OnRightButtonDown  (MOUSEINFO *mi); 
 
    void OnMiddleButtonUp (MOUSEINFO *mi); 
 
    void OnMiddleButtonDown (MOUSEINFO *mi); 
 
}; 
 
 
void MouseRptParser::OnMouseMove(MOUSEINFO *mi) 
 
{ 
 
  dx = mi->dX; 
 
  dy = mi->dY; 
 
}; 
 
  
 
  
 
void MouseRptParser::OnLeftButtonUp (MOUSEINFO *mi) 
 
{ 
 
  lmb = 0; 
 
}; 
 
  
 
  
 
void MouseRptParser::OnLeftButtonDown (MOUSEINFO *mi) 
 
{ 
 
  lmb = 1; 
 
}; 
 
  
 
  
 
void MouseRptParser::OnRightButtonUp  (MOUSEINFO *mi) 
 
{ 
 
  rmb = 0; 
 
}; 
 
  
 
  
 
void MouseRptParser::OnRightButtonDown  (MOUSEINFO *mi) 
 
{ 
 
  rmb = 1; 
 
}; 
 
  
 
  
 
void MouseRptParser::OnMiddleButtonUp (MOUSEINFO *mi) 
 
{ 
 
  mmb = 0; 
 
}; 
 
  
 
  
 
void MouseRptParser::OnMiddleButtonDown (MOUSEINFO *mi) 
 
{ 
 
  mmb = 1; 
 
}; 
  
MouseRptParser  Prs; 
 
void setup() {
 
  delay(5000); 
  Mouse.begin(); 
  Serial.begin(115200); 
  Serial.setTimeout(1);
   
  Usb.Init(); 
  HidMouse.SetReportParser(0, &Prs); 
  } 
 
void loop() {
  dx = 0; 
 
  dy = 0; 
 
   
 
  j = 0; 
 
  c = 0; 
 
  e = 0; 
 
     
 
  Usb.Task(); 
 
  
 
  //Clicking 
 
  if (lmb == 0){ 
 
    Mouse.release(MOUSE_LEFT); 
 
    } 
 
     
 
  else if (lmb == 1){ 
 
    Mouse.press(MOUSE_LEFT); 
 
    } 
 
  
 
  if (rmb == 0){ 
 
    Mouse.release(MOUSE_RIGHT); 
 
    } 
 
     
 
  else if (rmb == 1){ 
 
    Mouse.press(MOUSE_RIGHT); 
 
    } 
 
  
 
  if (mmb == 0){ 
 
    Mouse.release(MOUSE_MIDDLE); 
 
    } 
 
     
 
  else if (mmb == 1){ 
 
    Mouse.press(MOUSE_MIDDLE); 
 
    } 
    if (Serial.available() > 0) {
    //In future I plain put something here, but for now, don't consider it
  }
   else { 
 
    Mouse.move(dx, dy); 
 
    } 
}
```

There is a way what i can modify the library for work with my mouse?

rtek1000, 2023-06-15:
Try to find out if your mouse has more than one interface. I have a keyboard with built-in touchpad that doesn't work with this library. But it works with version 3.0:

https://github.com/felis/UHS30

rtek1000, 2023-06-15:
USB sniffer:

https://wiki.wireshark.org/CaptureSetup/USB

skhrlx, 2023-07-25:
I just re-mapped the arduino listening, and now it's working without bugs

closed 2023-07-25 by skhrlx

cuccune, 2024-05-27:
> I just re-mapped the arduino listening, and now it's working without bugs

How u fix