felis/USB_Host_Shield_2.0 #758: Can't Connect to BT Dongle

jsmallcomb4, 2022-12-23:
Trying to connect to PS4 Controller through ASUS USB-BT400 with no luck.  This BT dongle is listed in "confirned working" list.   Need help debugging issue or a BT dongle that works with USB_Host_Shield and PS4 controller. 

Running PS4BT.ino with Debug on returns

PS4 Bluetooth Library Started
Unknown Device Connected - VID: 0B05 PID: 17CB
BTD Init Failed, error code: D1

Running USB_desc.ino returns 

Start

01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		0B05
Product ID:		17CB
Revision ID:		0112
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01

Configuration descriptor:
Total length:		00DA
Num.intf:		04
Conf.value:		01
Conf.string:		00
Attr.:			A0
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		03
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0010
Polling interval:	01

Endpoint descriptor:
Endpoint address:	82
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Endpoint descriptor:
Endpoint address:	02
Attr.:			02
Max.pkt size:		0040
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0000
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			01
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0009
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			02
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Max.pkt size:		0011
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0011
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			03
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0019
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			04
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0021
Polling interval:	01

Interface descriptor:
Intf.number:		01
Alt.:			05
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00

Endpoint descriptor:
Endpoint address:	83
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Endpoint descriptor:
Endpoint address:	03
Attr.:			01
Max.pkt size:		0031
Polling interval:	01

Interface descriptor:
Intf.number:		02
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		FF
Intf. Protocol:		FF
Intf.string:		00

Endpoint descriptor:
Endpoint address:	84
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Endpoint descriptor:
Endpoint address:	04
Attr.:			02
Max.pkt size:		0020
Polling interval:	01

Interface descriptor:
Intf.number:		03
Alt.:			00
Endpoints:		00
Intf. Class:		FE
Intf. Subclass:		01
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		09
Type:		21
Contents:	058813400010010012

Addr:1(0.0.1)

touchgadget, 2022-12-24:
Wild guess. Try adding the VID PID of the Asus gadget to the line marked below. BTD.

BTD.cpp
```
// Check if attached device is a Bluetooth dongle and fill endpoint data structure
                // First interface in the configuration must have Bluetooth assigned Class/Subclass/Protocol
                // And 3 endpoints - interrupt-IN, bulk-IN, bulk-OUT, not necessarily in this order
                for(uint8_t i = 0; i < num_of_conf; i++) {
>>                      if((VID == IOGEAR_GBU521_VID && PID == IOGEAR_GBU521_PID) || (VID == BELKIN_F8T065BF_VID && PID == BELKIN_F8T065BF_PID)) {
                                ConfigDescParser<USB_CLASS_VENDOR_SPECIFIC, WI_SUBCLASS_RF, WI_PROTOCOL_BT, CP_MASK_COMPARE_ALL> confDescrParser(this); // Workaround issue with some dongles
                                rcode = pUsb->getConfDescr(bAddress, 0, i, &confDescrParser);
                        } else {
                                ConfigDescParser<USB_CLASS_WIRELESS_CTRL, WI_SUBCLASS_RF, WI_PROTOCOL_BT, CP_MASK_COMPARE_ALL> confDescrParser(this); // Set class id according to the specification
                                rcode = pUsb->getConfDescr(bAddress, 0, i, &confDescrParser);
                        }

```

jsmallcomb4, 2022-12-25:
Good Guess!

Added this to BTD.h

#define ASUS_BT400_VID          0x0B05
#define ASUS_BT400_PID          0x17CB

Changes BTD.cpp if statement to include ASUS doggle in work around

if((VID == IOGEAR_GBU521_VID && PID == IOGEAR_GBU521_PID) || (VID == BELKIN_F8T065BF_VID && PID == BELKIN_F8T065BF_PID)||(VID == ASUS_BT400_VID && PID == ASUS_BT400_PID)) {

Works now. 

Thanks.

closed 2022-12-25 by jsmallcomb4