felis/USB_Host_Shield_2.0 #761: Google Stadia controller

garybethel, 2023-01-22:
Is there any chance of Google Stadia controller being added to the library natively outside of USBHIDJoyStick?
I can use it with USBHIDJoyStick but the right analog stick seems to crash things if held in any position too long.

I would donate a Stadia controller if there is any interest in this and one is needed to development. Note Google also just released a firmware unlock to enable Bluetooth LE(since it would only work via USB cable).

Lauszus, 2023-01-23:
Hi @garybethel. I'll replied to your email with my address :)

garybethel, 2024-12-22:
Hey Kristian,One year later, I still owe you a stadia controller.Are you still at the same address?

Lauszus, 2024-12-22:
> Hi @garybethel. I'll replied to your email with my address :)

Send me an email lauszus@gmail.com and we will figure out the details.