felis/USB_Host_Shield_2.0 #763: Large nak_count with 2 devices on a USBHub

2jmorg, 2023-01-23:
Hi, I'm trying to use two joysticks on a USBHub connected to a USB Host Shield mini. I've adapted the HIDUniversal code to suit my needs in terms of parsing. The code works fine and I can read and parse reports from both devices fine, but when both devices are active at once I have a big delay when polling the devices, taking up to 200 ms per device.

I've been trying to pintpoint the issue and from what I can see, when entering dispatchPkt in USB.cpp nak_count goes up to about 3460. It's funny but this number is always very close to that +- 15.

When I connect only one of the joysticks, nak_count at that same point is usually at around 31-32. I don't know much about USB tbh, so I don't know what would be reasonable numbers here, but when using two devices the lag makes it not usable, whereas with only one device it's not noticeable.

Are you aware of anything like this happening when using the USBHub class in your library?

I've checked if the issue is with the hub itself, hooking it directly to the computer and checking the behaviour of both joysticks and there's no lag, so it's definitely something in the code running in the board. I'm using an nrf52840 express feather from Adafruit.

Any tips would be appreciated!

Lauszus, 2023-01-23:
You have to be aware that the MAX3421E only supports up to Full-Speed 12Mbps aka USB 1.1, thus it might not be able process all the data without causing a lag.

Another common issue is if you have any `delay/delayMicroseconds` calls in your code, as that will preventing it from calling `Usb.Task()`. Can you share your code?

2jmorg, 2023-01-23:
Hi!, thanks for the fast answer :) 
I could share my code but in regards to the library the only thing I've modified is Hiduniversal...at first I was using a different approach, connecting the adafruit to 2 separate host minis, so I had to change some stuff in usb and usbcore as well to make this work, but now I'm just using a hub and a single host (much more convenient :) )  and I don't think that's interfering in this situation. 

I changed hiduniversal to focus on a single PID so that I can create two hiduniversals that parse the same kind of device and attach them to the usb, knowing which one is which (I need to know which one's on the left and which one's on the right).

With a USB sniffer I've noticed the joysticks have several vendor specific reports that my driver is not parsing, likely used with the dedicated driver that I don't really need for my application. And three of those reports are actually somewhat big, so I think what I'm going to try is in my hiduniversal to limit which reports to actually poll for, since I do know I don't need several of those at all. The joystick reports themselves should be well within the limits of the MAX3421E.

I had been running my previous setup (with 2 host boards) before with simpler joysticks and it worked fine, but I just upgraded to a couple of premium ones and was seeing this happen and was puzzled about why it could be happening. These 2 joysticks have a bunch of secondary cosmetic and configuration functionalities, and those are likely to require more bandwidth than just sending positions of axis and buttons pressed.

I wasn't aware of the MAX3421E speed limits, you've given me at least something to try out.

I'll try and let you know if it worked, thanks!

2jmorg, 2023-01-24:
Small update:

I noticed yesterday that hiduniversal requests the report descriptor again when parsing the reports. This is something I had already noticed but since it was working for me ok with my older joysticks, I ignored it. I guess it's fine when you're trying to parse something and you don't know precisely the data you're expecting. Since it's not my case anymore because I want to work with these 2 specific joysticks I'll take the same approach as your SRWS1 example and just use a struct onto which I can dump the contents of the packets and read the data from my struct.

The lags were coming specifically from the part that does all the parsing of the descriptor, so if I can avoid that, I'll be good to go.

xxxajk, 2023-02-04:
NAK means the USB device isn't ready yet.