felis/USB_Host_Shield_2.0 #765: CHIP MAXIM3421

TeatrSveta, 2023-01-25:
Please tell me if there are any differences in the work of the library for USB HOST SHILD with the MAXIM3421EE chip?
This type of boards began to appear on the market since the middle of last year. Unfortunately, your library does not work with such boards.

Lauszus, 2023-01-26:
Please send me the exact part number? There is no such chip as `MAXIM3421EE`?

TeatrSveta, 2023-01-26:
Hello, Lauszus.
Thanks for the answer.
In attachment - foto of the board with the MAXIM chip
[image]
Exact inscription on the chip "MAXIM3421EE TJ939+NTCZ"
There is a mention of a similar problem here
https://forum.arduino.cc/t/arduino-usb-host-shield-not-registering-usb-device/1028206
Regards, Alexey.

Lauszus, 2023-01-26:
Please take a close up of the chip itself.

TeatrSveta, 2023-01-26:
[image]

Lauszus, 2023-01-26:
This is just a regular MAX3421E chip. I suggest you contact the reseller for help, as it should work.

TeatrSveta, 2023-01-26:
Thank you
I'll do

Lauszus, 2023-01-26:
Please see: https://github.com/felis/USB_Host_Shield_2.0/issues/752#issuecomment-1332769840

TeatrSveta, 2023-01-27:
Hello, Lauszus.
I have progressed in board research. At the moment, the test board_qc (from USB_Host_Shield_Library_2_0-1.6.2 ) stops while waiting for an exchange with USB.
Circuits At Home 2011
ster... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K......
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
GPIO test passed.
PLL test. 100 chip resets will be performed
Resetting oscillator

Reset number 0 Time to stabilize - 332 cycles
Reset number 1 Time to stabilize - 332 cycles
Reset number 2 Time to stabilize - 332 cycles
Reset number 3 Time to stabilize - 332 cycles
Reset number 4 Time to stabilize - 332 cycles
Reset number 5 Time to stabilize - 332 cycles
Reset number 6 Time to stabilize - 332 cycles
Reset number 7 Time to stabilize - 332 cycles
Reset number 8 Time to stabilize - 332 cycles
Reset number 9 Time to stabilize - 332 cycles
Reset number 10 Time to stabilize - 332 cycles
Reset number 11 Time to stabilize - 332 cycles
Reset number 12 Time to stabilize - 332 cycles
Reset number 13 Time to stabilize - 332 cycles
Reset number 14 Time to stabilize - 333 cycles
Reset number 15 Time to stabilize - 333 cycles
Reset number 16 Time to stabilize - 333 cycles
Reset number 17 Time to stabilize - 333 cycles
Reset number 18 Time to stabilize - 333 cycles
Reset number 19 Time to stabilize - 332 cycles
Reset number 20 Time to stabilize - 332 cycles
Reset number 21 Time to stabilize - 332 cycles
Reset number 22 Time to stabilize - 332 cycles
Reset number 23 Time to stabilize - 332 cycles
Reset number 24 Time to stabilize - 332 cycles
Reset number 25 Time to stabilize - 332 cycles
Reset number 26 Time to stabilize - 332 cycles
Reset number 27 Time to stabilize - 332 cycles
Reset number 28 Time to stabilize - 332 cycles
Reset number 29 Time to stabilize - 333 cycles
Reset number 30 Time to stabilize - 333 cycles
Reset number 31 Time to stabilize - 333 cycles
Reset number 32 Time to stabilize - 333 cycles
Reset number 33 Time to stabilize - 333 cycles
Reset number 34 Time to stabilize - 332 cycles
Reset number 35 Time to stabilize - 332 cycles
Reset number 36 Time to stabilize - 332 cycles
Reset number 37 Time to stabilize - 332 cycles
Reset number 38 Time to stabilize - 332 cycles
Reset number 39 Time to stabilize - 332 cycles
Reset number 40 Time to stabilize - 332 cycles
Reset number 41 Time to stabilize - 332 cycles
Reset number 42 Time to stabilize - 332 cycles
Reset number 43 Time to stabilize - 333 cycles
Reset number 44 Time to stabilize - 333 cycles
Reset number 45 Time to stabilize - 333 cycles
Reset number 46 Time to stabilize - 333 cycles
Reset number 47 Time to stabilize - 333 cycles
Reset number 48 Time to stabilize - 332 cycles
Reset number 49 Time to stabilize - 332 cycles
Reset number 50 Time to stabilize - 332 cycles
Reset number 51 Time to stabilize - 332 cycles
Reset number 52 Time to stabilize - 332 cycles
Reset number 53 Time to stabilize - 332 cycles
Reset number 54 Time to stabilize - 332 cycles
Reset number 55 Time to stabilize - 332 cycles
Reset number 56 Time to stabilize - 332 cycles
Reset number 57 Time to stabilize - 332 cycles
Reset number 58 Time to stabilize - 333 cycles
Reset number 59 Time to stabilize - 333 cycles
Reset number 60 Time to stabilize - 333 cycles
Reset number 61 Time to stabilize - 333 cycles
Reset number 62 Time to stabilize - 333 cycles
Reset number 63 Time to stabilize - 332 cycles
Reset number 64 Time to stabilize - 332 cycles
Reset number 65 Time to stabilize - 332 cycles
Reset number 66 Time to stabilize - 332 cycles
Reset number 67 Time to stabilize - 332 cycles
Reset number 68 Time to stabilize - 332 cycles
Reset number 69 Time to stabilize - 332 cycles
Reset number 70 Time to stabilize - 332 cycles
Reset number 71 Time to stabilize - 332 cycles
Reset number 72 Time to stabilize - 332 cycles
Reset number 73 Time to stabilize - 333 cycles
Reset number 74 Time to stabilize - 333 cycles
Reset number 75 Time to stabilize - 333 cycles
Reset number 76 Time to stabilize - 333 cycles
Reset number 77 Time to stabilize - 333 cycles
Reset number 78 Time to stabilize - 332 cycles
Reset number 79 Time to stabilize - 332 cycles
Reset number 80 Time to stabilize - 332 cycles
Reset number 81 Time to stabilize - 332 cycles
Reset number 82 Time to stabilize - 332 cycles
Reset number 83 Time to stabilize - 332 cycles
Reset number 84 Time to stabilize - 332 cycles
Reset number 85 Time to stabilize - 332 cycles
Reset number 86 Time to stabilize - 332 cycles
Reset number 87 Time to stabilize - 332 cycles
Reset number 88 Time to stabilize - 333 cycles
Reset number 89 Time to stabilize - 333 cycles
Reset number 90 Time to stabilize - 333 cycles
Reset number 91 Time to stabilize - 333 cycles
Reset number 92 Time to stabilize - 333 cycles
Reset number 93 Time to stabilize - 332 cycles
Reset number 94 Time to stabilize - 332 cycles
Reset number 95 Time to stabilize - 332 cycles
Reset number 96 Time to stabilize - 332 cycles
Reset number 97 Time to stabilize - 332 cycles
Reset number 98 Time to stabilize - 332 cycles
Reset number 99 Time to stabilize - 332 cycles

Checking USB device communication.

Waiting for device...

Please advise - how to carry out further diagnostics of the board and USB connection?
Perhaps you have encountered similar cases?
Regards, Alexey.

touchgadget, 2023-01-27:
Looks fine to me. SPI long test passed. The GPIO failure is normal. Plug in a keyboard or mouse and the program will dump out the device descriptors.

TeatrSveta, 2023-01-28:
Thank you for your reply.
I did the steps you listed.
With different keyboards and mouses.
Unfortunately the answer from the board is the same -

Checking USB device communication.

Waiting for device...

The feeling that some little things are missing - contact, settings and that sort of thing.
Which?

touchgadget, 2023-01-28:
One possibility is 5V is not reaching the USB connector. Do mouse or keyboard lights come? Use a multi-meter to check for 5V. Double check the soldering for 5V. I do not know what else would prevent this from working.

TeatrSveta, 2023-01-29:
Solved!
The jumper for VBUS turned out to be key.
All descriptions of this type of boards indicate that you need to install a "5V" jumper. In this case, USB does not work.
As soon as I broke the "5V" jumper. And put the jumper on "3.3V" everything worked. Passed the board_gс test to the end. USB-MIDI dump utility sketch successfully launched. 
I was able to test this change on 9 boards that I have Everything worked.
The only thing - do not ask me why it worked that way. ;-)
Thanks for your support and help.

touchgadget, 2023-01-29:
Wow, sounds like the 5V and 3.3V pads are labelled incorrectly. USB devices need 5V, not 3.3V. Score one more defect for cheap clone boards.

xxxajk, 2023-02-04:
Even more fun when they put on the wrong chip :-D

ecsontv, 2023-02-15:
> Solved! The jumper for VBUS turned out to be key. All descriptions of this type of boards indicate that you need to install a "5V" jumper. In this case, USB does not work. As soon as I broke the "5V" jumper. And put the jumper on "3.3V" everything worked. Passed the board_gс test to the end. USB-MIDI dump utility sketch successfully launched. I was able to test this change on 9 boards that I have. Everything worked. The only thing - do not ask me why it worked that way. ;-) Thanks for your support and help. Alexey

May i see the board after breaking the 5v jumper sir?. Thanks

TeatrSveta, 2023-02-15:
Hello.
Here is one of them..
If necessary, I can provide a video of the operation of such a board with a mouse or midi keyboard. ;-)
[image]

ecsontv, 2023-02-15:
> Hello. Here is one of them.. If necessary, I can provide a video of the operation of such a board with a mouse or midi keyboard. ;-) [image]

Very interested sir, I will subscribe your YouTube channel sir. Thanks

ecsontv, 2023-02-16:
> Hello. Here is one of them.. If necessary, I can provide a video of the operation of such a board with a mouse or midi keyboard. ;-) [image]

I do the same to put the jumper on "3.3v" and it's work for mouse or keyboard but not for usb device like guitar multi effects sir. Is this because the vbus output is not 5v sir?. sorry my english.

TeatrSveta, 2023-02-16:
I think that a Usb device such as a guitar will most likely work if you try to use the USB-MIDI dump utility sketch from the applications to the USB_Host_Shield library_2.0 or similar.

ecsontv, 2023-02-16:
> I think that a Usb device such as a guitar will most likely work if you try to use the USB-MIDI dump utility sketch from the applications to the USB_Host_Shield library_2.0 or similar.

I did sir, but no lucky. 

[image]

VBUS is 3.84v not 5v. Thanks

touchgadget, 2023-02-16:
There is another recent message thread about 3.9V out the USB host connector. This sounds to me like there is a short circuit between the 3.3V and 5V power rails. I have no idea how to fix this.

https://github.com/felis/USB_Host_Shield_2.0/issues/755

HisashiKato, 2023-02-17:
[image]

This is the shield I use.

xxxajk, 2023-02-19:
That's what you get for a crappy clone. Don't buy knockoffs.

touchgadget, 2023-02-21:
Are there authentic Circuits@Home USB host shields for sale any more? I guess the SparkFun board is authentic but it is not 100% compatible with the Circuits@Home shield. The SparkFun forum has some note on how to make it work with this library but only for Uno.

xxxajk, 2023-02-22:
Looks like I may have to produce them, as it looks like @Lauszus got out of the biz.
If I do that, expect a MUCH BETTER design than the classic.