felis/USB_Host_Shield_2.0 #766: USB name conflict with arduino esp32 code base

tomrodeheffer, 2023-01-26:
There is an unfortunate name conflict when using this library with the arduino esp32 code base.  Namely, under the default settings, this code base defines an extern symbol USB.  This happens in USB.h which is included from HardwareSerial.h which is included from Arduino.h which appears to be automatically included in all compilations. See the code base in https://github.com/espressif/arduino-esp32/tree/master/cores/esp32

It is extremely unfortunate that this extern symbol has the same name as the class USB defined in UsbCore.h  The class definition is permitted, but any attempt to use USB elsewhere causes the compiler to complain that "USB is not a type".  For instance, the USB Host Shield example sketch USB_desc does not compile for my FeatherS3.  I am assuming this is a general problem for all arduino esp32 boards.

I don't really know the right way to solve this problem.  Currently I am making progress by globally substituting a different name for USB in the USB Host Shield package.

touchgadget, 2023-01-27:
The original ESP32 does not have a USB OTG controller so I don't think there is a conflict with the USB host shield library. The ESP32-S2 and ESP32-S3 have USB OTG and their own USB host stack so that is the source of the conflicts.

tomrodeheffer, 2023-01-27:
Well, my application needs to be both a USB host (to some devices) and a USB device (to another host) at the same time, so I need to use both the on-chip USB controller and a USB host shield.  Looks like I really ought to be annoyed at the ESP32-S2/S3 code base, because the ATmega32U4 (Leonardo, Micro) which does have an on-chip USB controller does not have this name conflict.  Unfortunately, my application won't fit in an ATmega32U4.

But everything works fine if I fork the USB host shield package and merely rename the USB class, so that's my solution.