felis/USB_Host_Shield_2.0 #772: Purchase link from readme is not valid

matihuf, 2023-04-02:
Link from readme is not valid [http://shop.tkjelectronics.dk/product_info.php?products_id=43]( http://shop.tkjelectronics.dk/product_info.php?products_id=43). I bought a clone from aliexpress, but it doesn't work as expected. I discovered that instead of a 74AHC125 chip, I have two 74HCT125s. If there is currently no company that makes this board, is there a valid board schematic somewhere?
[image]

touchgadget, 2023-04-03:
See the following for information including schematics for boards that the cloners copy. 

https://chome.nerpa.tech/

matihuf, 2023-04-04:
Founded it also here https://web.archive.org/web/20230125120018/http://www.tkjelectronics.dk/uploads/usbhost/UHS_20.pdf

closed 2024-02-22 by matihuf