felis/USB_Host_Shield_2.0 #775: Unable to detect USB devices with Arduino Nano

AAJAY5, 2023-05-01:
## Desc Test

- I am running example code **USB_desc** and unable to get working. All descriptor read as 0. 
- Correct me if I am wrong. I have attached connection as well as version information.
[image]

## Board and Version Info

|||
|---|---|
|Board|Arduino Nano|
|Arduino AVR Boards|v1.8.6 (latest)|
|USB_Host_Shield_2.0|v1.6.2 (latest)|

## Connections
|Shield|Nano|
|---|---|
|MOSI|11|
|MISO|12|
|SCLK|13|
|SS|10|
|INT|9|
|RST|RST|
|VBUS|5V ( With cut on PCB)|
|VCC|3.3V|
|GND|GND|

touchgadget, 2023-05-02:
> VBUS 	5V ( With cut on PCB)

Why cut 5V? Can you verify with a meter that 5V is coming out the USB type A connector? If not the USB device, it is not receiving any power.

https://forum.arduino.cc/t/nano-and-usb-host-shield/1106732

AAJAY5, 2023-05-03:
Circuits At Home 2011
USB Host Shield Quality Control Routine
Reading REVISION register... Die revision 03
SPI long test. Transfers 1MB of data. Each dot is 64K................ SPI long test passed
GPIO test. Connect GPIN0 to GPOUT7, GPIN1 to GPOUT6, and so on
Test failed. Value written: 00 Value read: FF 
Press any key to continue...
GPIO test passed.
PLL test. 100 chip resets will be performed
Reset number 100 Time to stabilize - 467 cycles
Checking USB device communication.

Reset complete. Waiting for the first SOF...
Getting device descriptor
Descriptor Length:      00
Descriptor type:        00
USB version:            0000
Device class:           00
Device Subclass:        00
Device Protocol:        00
Max.packet size:        00
Vendor  ID:             0000
Product ID:             0000
Revision ID:            0000
Mfg.string index:       00
Prod.string index:      00
Serial number index:    00
Number of conf.:        00

xxxajk, 2023-05-06:
Apply 5v to the USB output, the cut line is for operating on devices that cooperate with 3.3v supply, which there aren't many of those, typical use case is a self powered device, such as powered hub, or DSLR camera.

There is no need to connect reset, in fact, don't connect it at all. The chip is reset via software.

AAJAY5, 2023-05-06:
Connections
[image]

rtek1000, 2023-06-15:
Hi,

I've seen other users saying this Chinese board doesn't work, I bought one and couldn't get it to work, years ago, when I have time I'll try again.

But I have the shield board for Arduino UNO which has two IC 74HC125 to convert 5V and 3.3V, this one works, I just don't remember why I put a jumper from the RESET button to pin 12 of the IC 74HC125 next to the IC MAX, the diagram already shows this connection, maybe my card was defective.

https://5.imimg.com/data5/XK/RR/MY-1833510/usb-host-shield-for-arduino.pdf