felis/USB_Host_Shield_2.0 #778: Help with RTL8153 Gigabit Ethernet Adapter on Arduino

Miguel0101, 2023-07-13:
I have a USB network adapter (Vendor ID: 0x0BDA, Product ID: 0x8153). So I need help dealing with this network adapter. Can someone help me? I thank.

Using an Arduino Uno.

xxxajk, 2023-07-17:
totally not going to happen on an UNO, not enough RAM. Perhaps UNO R4, though, but then, you would want to move to UHS30 which is going to be easier to manage, and requires more RAM too. Generally I avoid any MCU with less than 64K RAM and 8 bit anymore, since newer offerings do more for less money.

Miguel0101, 2023-07-21:
> totally not going to happen on an UNO, not enough RAM. Perhaps UNO R4, though, but then, you would want to move to UHS30 which is going to be easier to manage, and requires more RAM too. Generally I [...]

Thanks for your response!

I have a ESP32 Wrover with 8MB of flash and 8MB of RAM(PSRAM).
It should probably work for its processing power and memory.

closed 2023-07-30 by Miguel0101