felis/USB_Host_Shield_2.0 #792: Bluetooth dongle bluetooth version 5 is not supported?

kawashimaken, 2023-11-29:
Bluetooth dongle bluetooth version 5 is not supported?

This is result of board_qc
`Reset complete. Waiting for the first SOF...
Getting device descriptor
Descriptor Length:	12
Descriptor type:	01
USB version:		0110
Device class:		E0
Device Subclass:	01
Device Protocol:	01
Max.packet size:	40
Vendor  ID:		2357
Product ID:		0604
Revision ID:		0200
Mfg.string index:	01
Prod.string index:	02
Serial number index:	03
Number of conf.:	01`

My  Bluetooth dongle is new tp-link, it says that bluetooth 5.0 is supported. I tried the wii program but it dose'nt work.
I do notice there is a list of working dongles at wiki page, however most of the links are broken and I found that most of them are  "old" ones, My guess is that the "new" bluetooth dongles won't work? or are there any ongoing work to support latest bluetooth version 5.
Or do I need to by an "old" bluetooth dongle for now?

Thanks!

xxxajk, 2023-11-29:
If it requires an upload of firmware to the device, you are  out of luck.
However... class and subclass and protocol all look OK, so it "should" work.

I never bothered with bluetooth personally, but perhaps the following link will be of some use:

https://www.bluetooth.com/specifications/specs/human-interface-device-service-1-0/

Notes:
Some devices will totally break the specifications.
Some of the drivers in this library are looking for specific VID:PID of the USB device, which could be your case.

touchgadget, 2023-11-29:
Consider using bluepad32 which runs on ESP32 for Bluetooth game controllers. No need for USB host controller.

Or see this: https://github.com/felis/USB_Host_Shield_2.0/issues/758

kawashimaken, 2023-11-30:
@touchgadget Thank you for your comment! I didn't know of bluepad32 (https://github.com/ricardoquesada/bluepad32) looks really promising! I will try that out later, Thank you!

kawashimaken, 2023-11-30:
@xxxajk Thank you for your response!
I might try out [add specific VID:PID] solution sometime. Thank you for your time to share the info!

kawashimaken, 2023-12-10:
By the way,  I bought a usb dongle from ELECOM (LBT-UAN05C2) Bluetooth Ver. 4 Class2, which is working well.  I hope this info can help someone.