felis/USB_Host_Shield_2.0 #798: Hanging at USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE state when connected to certain computers.

Jtonna, 2024-01-11:
### TLDR;
The HID Mouse project I'm working on works on 2/4 machines I can regularly test on, I've done what I can to debug and gather data but I'm at my limit of knowledge/ability. The project is pretty simple, it reads mouse data from my Razer Viper V2 Pro, reformats the data, logs it to my external serial device, and submits the event to the connected device. On my dev box (Laptop 1) and Laptop 2, the USB Task State is getting stuck on USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE and does not proceed any further. 

I do find it worth noting, i did test this on a 5th device (brothers desktop) and it did get stuck on USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE until i unplugged the arduino and plugged it in again. I did this aobut 20 times sanity checking and it only failed the one time.

You may think this doesnt belong here, and you may be right BUT even with debug mode on its not telling me theres a power failure or usb connection failure which i would expect it to do if this was a powersupply related issue.

### My Project Repo
https://github.com/Jtonna/USBHostShieldMouseProxyRazerViperV2Pro

### Hardware Setup
All PC's and laptops are on windows 10
PC1 (My primary gaming desktop)
PC2 (My guest gaming desktop)
Laptop 1 - Alienware - m15 R7 (The dev box)
Laptop 2 - Acer Predator G5 (fresh install of Windows, old laptop I brought in for sanity checks)

[Arduino Leonardo Clone Board](https://www.amazon.com/KEYESTUDIO-Leonardo-Development-Board-Arduino/dp/B0786LJQ8K/ref=sr_1_4?crid=1XTJPROUE3QN8&keywords=arduino+leonardo&qid=1704957707&sprefix=Arduino+leon%2Caps%2C149&sr=8-4) which seems to follow the spec of the normal arduino leonardo
[USB Host Shield Clone Board](https://www.amazon.com/HiLetgo-Shield-Arduino-Support-Android/dp/B01MTU9OLM/ref=sr_1_3?crid=BWXC12B2A1VG&keywords=usb+host+shield+arduino+leonardo&qid=1704957766&sprefix=USB+Host+Shield%2Caps%2C141&sr=8-3) Ive made sure to solder the board on 5v, 3.3 and 5v as shown [here ](https://forum.arduino.cc/t/usb-host-shield-with-no-power/1039235/7)
[FT232RL Mini USB to TTL Serial Converter](https://www.amazon.com/HiLetgo-FT232RL-Converter-Adapter-Breakout/dp/B00IJXZQ7C/ref=sr_1_3?keywords=ft232rl&qid=1704956230&sr=8-3) So i can use Serial1 and always log to a secondary device and is [connected like this](https://i.imgur.com/JbFtDiV.jpgrl)
2x Razer Viper V2 Pro Wireless (Both Updated to latest receiver firmware 1.05.0 and mouse firmware 2.00.03)

### Local Changes to the USB_Host_Shield_2.0 Library
In [settings.h Line 42](https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L42) ive set `#define ENABLE_UHS_DEBUGGING 1`
In [settings.h Line 49](https://github.com/felis/USB_Host_Shield_2.0/blob/master/settings.h#L42) ive set `#define USB_HOST_SERIAL Serial1` so I can view logs through my FT232RL on the devbox.

### IDE Version & Custom Boards.txt
I am primarily using IDE version 1.8.18 BUT i have used the latest 2.2.1 and uploaded the sketch with the same result of hanging at USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE across 2 devices.

[boards.txt](https://github.com/felis/USB_Host_Shield_2.0/files/13900151/boards.txt) - I removed all entries except the 3 ive used, 1 is default and 2 are custom. All 3 of them give the same result across all devices but with different vendor and product id's.

### Observations of the issue
As stated in the TLDR its as if the arduino isnt starting the device connected to the host shield on my Laptop 1 (dev box) and Laptop 2, Ive attached some startup logs from failures and successes. Ive made sure in failing cases the mouse is powered on and by the blinking indicator is looking for the receiver. Ive also attached a log from [Device Monitoring Studio](https://www.hhdsoftware.com/device-monitoring-studio)'s which has the same information as the image below.
[Packet View.txt](https://github.com/felis/USB_Host_Shield_2.0/files/13899327/Packet.View.txt)
[image]

Logs on Failure from Laptop 1 & Laptop 2 (They are the same so only 1 photo) (
[image]
Logs on success from PC 1 & 2 (they are the same so only 1 photo)
[image]

### Sanity Checks Performed
- Compared driver's being used
- Ive tried using the Red Leonardo reset switch on all machines to see if machines that previously failed would work and vice versa but the outcome is always the same, PC 1&2 work flawlessly and Laptop 1&2 fail.
- I've tried every combination of ports on failing machines from 2.x and 3.x usb ports with the same result's
- I've tried every combination of ports on working machines from 2.x and 3.x ports with all functioning results.
- Deleting all mouse-related drivers on all machines, and performing a hard shutdown and start, i see the device gets installed and then shows in device managers but i get the same outcome on failing machines.
- As a test on Laptop 2, i did a fresh install of Windows, connected the arduino and it installed but the logs showed it was stuck on the same USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE.
- The reason im unsure of this being a power delivery issue is in the above screenshots/logs you can see DevicePowerStateD0 which according to the docs here https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/device-power-states https://learn.microsoft.com/en-us/windows-hardware/drivers/kernel/device-working-state-d0 which imply to me that the device is fully powered and operational. 

### Used Driver Comparison
[Universal Serial Bus Viewer](https://learn.microsoft.com/en-us/windows-hardware/drivers/debugger/usbview)
Below ive attached the following
Screenshot of drivers used on both Laptop (dev box) and PC1 (primary desktop) which all seem to be using 10.0.1904.1
Screenshots of Universal Serial Bus View for the device
Logfile from Universal Serial Bus View with extra information not visible in screenshots

Laptop 1 (Dev Box, looks the same as secondary laptop)
[Laptop1 - USBViewAll.txt](https://github.com/felis/USB_Host_Shield_2.0/files/13899410/Laptop1.-.USBViewAll.txt)
[image]
<p float="left">
[image]
[image]
</p>

PC 1 (Primary PC, looks the same as secondary PC)
[PC1 USBViewAll.txt](https://github.com/felis/USB_Host_Shield_2.0/files/13899426/PC_1_-_USBViewAll.txt)
[image]
<p float="left">
[image]
[image]
</p>

### Board QC
I made a change to board_qc to use serial1 instead of serial.
Logs for Laptop (hanging, doesnt matter how many times i unplug and plugin the usb receiver, or just let it sit)
[Laptop1 - board_qc.txt](https://github.com/felis/USB_Host_Shield_2.0/files/13900663/Laptop1.-.board_qc.txt)
Logs for Desktop (works but i need to unplug the usb and plug it in when its on "waiting for device....." section )
[Desktop1 - board_qc.txt](https://github.com/felis/USB_Host_Shield_2.0/files/13900665/Desktop1.-.board_qc.txt)

### General Questions

1. I dont have a multimeter at the moment (i will this weekend) so i cant check voltages to the USB receiver on the host shield but im under the assumption that OSC would fail to start if this was a power supply issue?
2. In the event of it being a power supply issue, i purchased [this 5V 2A supply](https://www.amazon.com/gp/product/B07TZF37DY/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&th=1), i had a 20V and fried a prior leonardo today so ive gotta ask, if i plug this one in that conforms the the spec of the device and this is a power related issue this should resolve that correct? Ive got extra leonardos on the way just in case this frys another one.
3. What can i do to get more debug information or figure thats going on?
4. I did end up reading this article https://www.printed-droid.com/kb/usb-host-shield-repair/ which shows that some clones come with two 74HC125 IC's (which both of my shields have two of) and it states i should replace them with a 74AHC125 BUT i have no clue if i should buy some from ebay/amazon. I see alot of listings with 74AHC125 in the name on ebay and amazon but none are an exact match of 74AHC125 so if someone would be kind enough to dig one up id be more than happy to try a replacement on one of my spare shields.
5. I did take a look at https://www.sparkfun.com/products/retired/9628 for the shield most people are using but it says its retired and i cant find any locally, i see theres a USBc one so would getting that and a usbc to usb adapter work for my usecase?

Im more than happy to add debug logs and retest everything everywhere if anyone has any suggestions; after posting this ill reply to this with a link to a discussion about this im making on the Arduino discord so if anyone has time to hop on a call with me id be more than grateful to reimburse you for your time.

Jtonna, 2024-01-11:
Discussion Link for the the Arduino Discord -> https://discord.com/channels/420594746990526466/1194936441545965598/1194936441545965598

OR join manually from https://support.arduino.cc/hc/en-us/articles/4405329164178-Join-the-official-Arduino-Discord-server
Under Hardware Help ->search for Hanging at USB_DETACHED_SUBSTATE_WAIT_FOR_DEVICE state when connected to certain computers.

I would prefer to keep resolutions in this thread not on discord, but i would like extra coverage and support on this issue

xndhn, 2024-05-29:
How can I customize my own wireless mouse? Can you provide a specific tutorial?

Jtonna, 2024-05-29:
> How can I customize my own wireless mouse? Can you provide a specific tutorial?

I cant provide a tutorial but look at my linked repository i put some information there to put you on the right track

closed 2024-06-09 by Jtonna