felis/USB_Host_Shield_2.0 #803: Midi keyboards

AlfredR1, 2024-01-29:
One Midi keyboard (E-Keys49 from Evolution) sends VID 0A4D, PID 0099 and receives all keystroke date via USB correctly.
Second Midi keyboard (Oxygen49 from M-Audio) sends VID 0763, PID 3102. But no keystrokes or other Midi data are received.

I use Mini USB Shield 2.0 with MAX 3421E on ESP32 controller.
Any idea or workaround?

YuuichiAkagawa, 2024-01-30:
Pleae run  [USB_MIDI_desc ](https://github.com/YuuichiAkagawa/USBH_MIDI/blob/master/examples/USB_MIDI_desc/USB_MIDI_desc.ino) and report the results.

I had a power issue with the my M-Audio Keystation 49 MK3. Is the power supply enough?
https://github.com/YuuichiAkagawa/USBH_MIDI/issues/73

AlfredR1, 2024-01-30:
That is whaz I receive:

 

ets Jun  8 2016 00:22:57

 

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1344

load:0x40078000,len:13964

load:0x40080400,len:3600

entry 0x400805f0

Start

 

 

01

--

String Descriptors:

Manufacturer:                  M-Audio

Product:                              Oxygen 49

 

Device descriptor: 

Descriptor Length:          12

Descriptor type:               01

USB version:                      0110

Device class:                      00

Device Subclass:               00

Device Protocol:               00

Max.packet size:              08

Vendor  ID:                         0763

Product ID:                         3102

Revision ID:                        0110

Mfg.string index:             01

Prod.string index:            02

Serial number index:      00

Number of conf.:             01

 

Configuration descriptor:

Total length:                      0065

Num.intf:                            02

Conf.value:                        01

Conf.string:                        00

Attr.:                                     80

Max.pwr:                            32

 

Interface descriptor:

Intf.number:                     00

Alt.:                                       00

Endpoints:                          00

Intf. Class:                           01

Intf. Subclass:                    01

Intf. Protocol:                    00

Intf.string:                          00

 

<<<AUDIO_CONTROL>>>

Unknown descriptor:

Length:                09

Type:                    24

Contents:           01000109000101

 

Interface descriptor:

Intf.number:                     01

Alt.:                                       00

Endpoints:                          02

Intf. Class:                           01

Intf. Subclass:                    03

Intf. Protocol:                    00

Intf.string:                          00

 

<<<MIDIStreaming>>>

 

MS Interface descriptor

07240100014100

DescriptorSubtype: 01

USB_DESCRIPTOR_CS_INTERFACE :         MS_HEADER

MIDIStreaming SubClass Specification Release number: 0100

wTotalLength:                  0041

 

MS Interface descriptor

062402010103

DescriptorSubtype: 02

USB_DESCRIPTOR_CS_INTERFACE :         MIDI_IN_JACK(Embedded)

JackType:            01

JackID:                 01

 

MS Interface descriptor

062402020200

DescriptorSubtype: 02

USB_DESCRIPTOR_CS_INTERFACE :         MIDI_IN_JACK(External)

JackType:            02

JackID:                 02

 

MS Interface descriptor

092403010901020103

DescriptorSubtype: 03

USB_DESCRIPTOR_CS_INTERFACE :         MIDI_OUT_JACK(Embedded)

JackType:            01

JackID:                 09

NrInputPins:      01

020103

 

MS Interface descriptor

092403020A01010100

DescriptorSubtype: 03

USB_DESCRIPTOR_CS_INTERFACE :         MIDI_OUT_JACK(External)

JackType:            02

JackID:                 0A

NrInputPins:      01

010100

 

Endpoint descriptor:

Endpoint address:           81

Attr.:                                     02(Bulk)

Max.pkt size:                     0040

Polling interval: 00

 

MS Endpoint descriptor

0525010109

USB_DESCRIPTOR_CS_ENDPOINT :          MS_GENERAL

bNumEmbMIDIJack: 01

09

 

Endpoint descriptor:

Endpoint address:           02

Attr.:                                     02(Bulk)

Max.pkt size:                     0040

Polling interval: 00

 

MS Endpoint descriptor

0525010101

USB_DESCRIPTOR_CS_ENDPOINT :          MS_GENERAL

bNumEmbMIDIJack: 01

01

 

 

Addr:1(0.0.1)

Can you see what is wrong?

 

Alfred

 

YuuichiAkagawa/USBH_MIDI#73 <https://github.com/YuuichiAkagawa/USBH_MIDI/issues/73>

YuuichiAkagawa, 2024-01-30:
There are no problems with the descriptor.
Does the [USBH_MIDI_dump](https://github.com/felis/USB_Host_Shield_2.0/blob/master/examples/USBH_MIDI/USBH_MIDI_dump/USBH_MIDI_dump.ino) also not work?

AlfredR1, 2024-01-30:
Thanks for your attention and support

 

Midi dump  shows VID/PID only with M-Audio, even with keystrokes oder slider-movements::

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1344

load:0x40078000,len:13964

load:0x40080400,len:3600

entry 0x400805f0

VID:0763, PID:3102

But a lot of data with E-Key49:

 

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0030,len:1344

load:0x40078000,len:13964

load:0x40080400,len:3600

entry 0x400805f0

VID:0A4D, PID:0090

00000FD9:  4: 09 90 2D 13 00 00 0D 80 00 53 FC 3F 74 1C FC 3F ED A4 03 45 B0 5C FB 3F 00 00 00 00 00 00 00 00 01 00 00 00 38 1D FC 3F 00 52 FC 3F 03 00 00 00 23 00 06 00 2C 44 0D 80 20 53 FC 3F 00 00 00 00

00001061:  4: 09 90 2B 30 00 00 0D 80 00 53 FC 3F 74 1C FC 3F ED A4 03 45 B0 5C FB 3F 00 00 00 00 00 00 00 00 01 00 00 00 38 1D FC 3F 00 52 FC 3F 03 00 00 00 23 00 06 00 2C 44 0D 80 20 53 FC 3F 00 00 00 00

00001072:  4: 09 90 2D 00 00 00 0D 80 00 53 FC 3F 74 1C FC 3F ED A4 03 45 B0 5C FB 3F 00 00 00 00 00 00 00 00 01 00 00 00 38 1D FC 3F 00 52 FC 3F 03 00 00 00 

 

The supply on M-Audio seems ok, because all changes on setup and CC show correct reactions in the display

 

I am puzzeled

Alfred

YuuichiAkagawa, 2024-01-30:
The Keystation 49 MK3 does not lack power supply to the keyboard.
The operation is abnormal because a situation of insufficient power supply to the microcontroller occurs when the keyboard starts up, causing it to reset.
If the ESP32 is reset after the keyboard is connected and the display is completed, does it work?

AlfredR1, 2024-01-30:
No sorry, it does notVon meinem Smartphone gesendet.

AlfredR1, 2024-01-30:
May I have some additional information:

1.	I use the USBMiniShield with some modifications in order to supply it with 3V3 and USB with 5V I cutted one connection on the PCD (see picture 1). Now I can feed 5V on Kathode or Anode of the Schottly Diode
2.	Maybe the Reset (R 1k) from 5V to RST is not sufficient?
3.	INT output is now not supplied with 3V3 via 22k. But I do not connect INT signal to ESP32. Is it required?
(Instead of ESP32 WROOM I use an AudioKit A1S and USB Mini Shield only)

But first I like to wish a good night to Tokyo from Germany

Alfred

YuuichiAkagawa, 2024-01-30:
There seems to be no problem with the connection to USBHost Shield as it is working with E-Keys49.
I checked to see if there is sufficient power supply to isolate the problem. For example, if the power supply is more than 1A, I consider it is not a problem.
Since VID and PID are output when USBH_MIDI_dump is executed, the device is recognized as a MIDI device.

In previous cases, the following two factors have caused the failure to operate.
- Insufficient power ([issue](https://github.com/YuuichiAkagawa/USBH_MIDI/issues/73#issuecomment-1477600453))
- Toggle issue (#438)

AlfredR1, 2024-01-31:
Thanks a lot for your support. Now I know that it is no software issue. 
I will try to improve 5V supply and to stabilize it. This will take some time. If I will succeed I will leave a note here!

Best Regards