felis/USB_Host_Shield_2.0 #812: Is there any way to setup the SPI_SS pin and INTR pin from the main sketch?

mSlider, 2024-05-01:
I know that for change SPI_SS pin and INTR pin has to be done in "UsbCore.h", but this has the inconvenient that if I use diferent hardware boards with the same processor I have to modify the library ("UsbCore.h") each time I compile for one of those boards.

Is there any way to setup or change SPI_SS pin and INTR pin from the main sketch or setup function?

Thanks!!!

xxxajk, 2024-05-02:
You can do that in UHS3, but if UHS3 willl fit what you want to do, or needs a specific driver, right now that's on you.

mSlider, 2024-05-02:
Thank you for the info about UHS3 but I'd prefer to stay with UHS2 for now because it's running really stable.

If you have another solution to modify the library to make a function to call from main sketch for change SPI_SS and INTR pin please feel free to post here.

xxxajk, 2024-05-03:
Unfortunately the only recourse would be to either make a separate library (which is a pretty awful way to do it) or to make a second user account on your computer, change what you need to change, and do the compiles on that account.

UHS3 *is* stable for everything I have used it for. I don't even use 2.0 or the smaller AVRs anymore.
What board are you using?

mSlider, 2024-05-03:
Thanks xxxajk, I'm going to try to the UHS3 as you suggest.

One of the boards actually I use is a Leonardo with ATMEGA32U4 AVR processor and the application is for reading a usb gamepad and send some of the buttons to a computer as a HID Keyboard.

mSlider, 2024-05-03:
> UHS3 _is_ stable for everything I have used it for. I don't even use 2.0 or the smaller AVRs anymore. What board are you using?

One of my boards is a Leonardo

mSlider, 2024-05-03:
I've tried the UHS3 library with Leonardo and nothing happens with the HID_RAW example.

Also I see the sketch ocupped near 90% of the memory space, so I't can fit all the rest of routines in my programs.

If there is no solution for my request please feel free to close the issue!