felis/USB_Host_Shield_2.0 #813: Elagoo Arduino UNO R3 Waiting for Device...

Es0ter1ck, 2024-05-19:
I have an Arduino UNO R3 from Elagoo

and purchased this Host Shield
https://www.amazon.com/Shield-Arduino-Support-Google-Android/dp/B0CKSZ81CS/ref=sr_1_1?crid=3VZ6BXT3RCI29&dib=eyJ2IjoiMSJ9.ShK4RulJmdNYyfxWDz_a0YQFLY2INYhXkB-UkJnNkoVQqzqIP89CDnn4wesQxOhbUOMXdBLLhPiCzyEjgVBnqFMrcEcS6HuEklEevlf1is0q5mFqC7r7X5niBVQqJSy_mroWW-T-JVNMB3_uNkwoxiXOVJ8ePt_0kr1c1f8ZPiO4wjT4FgpNU9fZn3QvmSLtWIM3LeruG52VFC-DHqKjVe9-mrYHYg66sJ1JmAUI_y4.Nrz-zfG_AQ3l2H1owTHktDz6hXAUeTI8ex_GA194FeU&dib_tag=se&keywords=USB+Host+Shield+for+Arduino&qid=1715629500&sprefix=usb+host+shield+for+arduino%2Caps%2C123&sr=8-1

My Problem:
When running board_qc the Serial Monitor stays on Waiting for Device... forever never recognizing the device

What I have tied:
I have soldered the 3.3V and 5V input and the 5V for the VBUS power

this allows the board_qc to show Die revision 03 with my plugged in keybaord

I switched and tried 3V instead for the VBUS and that doesnt run past Die revision 00

Please help!

xxxajk, 2024-05-23:
plug in any kind of usb device to complete the test.

Es0ter1ck, 2024-05-24:
> plug in any kind of usb device to complete the test.

thank you for the response. The issue is that the keyboard is connected, I have tested with a bluetooth dongle as well

jacobsoftwares, 2024-05-24:
"I have soldered the 3.3V and 5V input" - either 3.3v or 5v not both

Es0ter1ck, 2024-05-24:
> "I have soldered the 3.3V and 5V input" - either 3.3v or 5v not both, power the board via VIN For power to USB via VBUS unsolder both 3.3 and 5v.

[image]

Attached is a close up of my host shield. Thank you for your reply! 

Sorry but maybe I am a bit confused you mean to unsolder the blue circle joints? Is the 5V join being soldered to power the VBUS ok?

jacobsoftwares, 2024-05-24:
It looks like you are not connected with arduino. Try this
[image]

Es0ter1ck, 2024-05-24:
With the current setup as shown I get 5V reaing witht he multimeter in the green circle VBUS, and 3.9V in the 3.3V joint and 5V in the 5V join in blue circle

The keyboard is lighting up so when connecting to the host shield it is powering it uusing the power connected to th Arduino so that seems to work.

How would I make sure VIN is working? it should be in this case as mentioned above right?

jacobsoftwares, 2024-05-24:
Then please recheck for MISO, MOSI, GND connections. 
[image]

jacobsoftwares, 2024-05-24:
solder these
[image]

Es0ter1ck, 2024-05-24:
you mean check the power? of those

jacobsoftwares, 2024-05-24:
SS, INT, MISO, MOSI, GND should be connected with the microcontroller for the board to work

Es0ter1ck, 2024-05-24:
Again thank you so much for the help. I have soldered the SS and the INT and checked they receive power, they do.  

I then read the pins you mentioned before and labled those readings.

Doing this though its still stuck on Waiting for Device...

[image]

jacobsoftwares, 2024-05-24:
desolder on SS, but keep the INT and try

Es0ter1ck, 2024-05-24:
Do the 0V on the right matter?

jacobsoftwares, 2024-05-24:
[image]
and try again..

jacobsoftwares, 2024-05-24:
Upload basic keyboard example and try

jacobsoftwares, 2024-05-24:
Have you installed USB HOST SHEILD library?

Es0ter1ck, 2024-05-24:
I have the library as the board_qc compiles without error. The USB host shield should be recognized using the board_qc sketch but its not, so when I uploaded the above code, no input was received,

jacobsoftwares, 2024-05-24:
I am using the sketch which I shared with USB HOST SHIELD library.. I had no problems.. Did you try the sketch I shared?

jacobsoftwares, 2024-05-24:
Does you USB devices require its drivers to be installed manually on the desktop? If you have installed their drivers manually then they might not work as desired.

jacobsoftwares, 2024-05-24:
Jumper is for sparkfun boards… send the pic of Arduino uno…

Compile sketch boardqc with debugging on.. so you will get detailed report via serial

jacobsoftwares, 2024-05-24:
Try with USB mouse

Es0ter1ck, 2024-05-24:
> Jumper is for sparkfun boards… send the pic of Arduino uno…
> 
> Compile sketch boardqc with debugging on.. so you will get detailed report via serial

[image]

Im not 100% what you mean by the drivers but when connected the devices I never had to manually download drivers or anything like that.

I tried with mouse, to no avail :( 

I will most likely purchase the official Arduino Uno R3 board, and another USB host shield or two and will report back the results

I just wanted to avoid that if I could.

jacobsoftwares, 2024-05-25:
the board is ok..

I have connected ATPMEGA328p with USB HOST shield on breadboard... So that should not be an issue.

Es0ter1ck, 2024-05-25:
Yes I have, to make sure I did it correctly this is what I did

I went to Documents->Arduino->Libraries->USB Host 2.0->Settings.h->ENABLE_UHS_DEBUGGING 1

But I don't see anything different in the Serial monitor is it somewhere else

jacobsoftwares, 2024-05-25:
I am sorry Arduino Uno board doesnot have debugging... I am working on ESP32 which has debugging feature... I was thinking Uno also has the same.. Sorry for misdirection.. I am actually out of ideas!

Es0ter1ck, 2024-05-25:
No worries, thank you so much for all of your help. There was this link that was given from the official Arduino forum and I was wondering if it makes more sense to you than me and if it could potentially be a solution:
https://forum.arduino.cc/t/usb-host-shield-not-working-wrong-smd-chip-populated/1180071