felis/USB_Host_Shield_2.0 #814: Hotone Ampero 2 Stomp USB MIDI Control

Jonachoda, 2024-05-19:
Hi, I was use a Arduino Nano with a duinofun USB Host Shield and program it to send Midi System Message to a Boss Pedalboard GT-1 with succefull, I could send and receive message from the pedalboard what I receive information of the effects are ON or OFF. I bought a new Pedalboard Hotone Ampero 2 Stomp, and the System message are so complex to descifer, then I think I can use Midi.SendData with the struct of a Midi Control Change #CC, for 4 days I'm trying with no results. Then I look into the response of the SendData, with anything connected I receive an error number 0x0E that the device is disconected, but with this pedalboard connected I receive an Erro number 0x0D that represents an JSTATE issue May I think what the problem is the speed of the Data, what do you think?

YuuichiAkagawa, 2024-05-19:
Please run [USB_MIDI_desc](https://github.com/YuuichiAkagawa/USBH_MIDI/tree/master/examples/USB_MIDI_desc).
Hotone Ampero 2 Stomp may not support USB Full Speed communication. In that case, you cannot connect using USB Host Shield.In past cases, Line6 HX Stomp was not supported.
https://github.com/YuuichiAkagawa/USBH_MIDI/issues/75

Jonachoda, 2024-05-20:
21:04:16.000 -> Start
21:04:28.186 -> 
21:04:28.186 -> 
21:04:28.186 -> 01
21:04:28.186 -> --
21:04:28.186 -> 
21:04:28.186 -> Request error. Error code:	0D
21:04:28.186 -> 
21:04:28.186 -> Addr:1(0.0.1)

Jonachoda, 2024-05-20:
When I connect the USB cable the Arduino reboots, the only way to get this message if I restart the pedalboard wth the ON/OFF button

YuuichiAkagawa, 2024-05-20:
It is probably only compatible with Hi-Speed.
Hotone Ampero 2 Stomp seems to be equipped with a MIDI port, so why not control it with serial MIDI?
I don't know what caused the restart, but it may have become unstable as a result of the power being supplied to the pedalboard.

Jonachoda, 2024-05-20:
Yes, have 2 MIDI port, IN and OUT, but the DIN conector are so big, In addition to the fact that I already had the controller and it was only reprogram the MIDI messages. I go with MIDI messages trought MIDI serial with DIN connectors. TThe problem is that I can't receive messages to check if the effect is on or off.