felis/USB_Host_Shield_2.0 #815: Recommended 5V target board that doesn't require a shield?

jdfr228, 2024-05-20:
I am a bit overwhelmed while trying to find a 5V-compliant board that has everything required built-in (IE includes USB Host support).

I will be connecting to a retro 80's computer and want full 5V tolerance on all I/O and will be powering the board directly from the computer's proprietary connector's 5V pin.

Initially I bought an Arduino Nano not understanding that having a USB port doesn't mean the device can be a USB Host.

Could anyone recommend anything in particular to me, or point me to a better place to ask (sorry to open this as a GitHub issue)?  Something like the Teensy 3.2 looks great, but it's long since discontinued and the newer ones aren't 5V tolerant.

Thank you!

touchgadget, 2024-05-20:
Arduino Mega 2560 ADK has a built-in MAX3421E chip and is supported by this lib. No shield needed. 5V. I have never used this board this is about all I know.

xxxajk, 2024-05-23:
There are also 5v tolerant MCU boards as well. Some pack an awesome punch!
This has always been one of my complaints as well. The companies that produce the chips never have everything in total that you want. They'll have subsets, and always seem to leave just one thing out. Sometimes they leave out USB host or otherwise. 

Teensy 3.2 is pretty close to what you might want, but... by far the best solution, which has come closest to my own needs has been the Teensy 4.1, as far as a board goes. It *nearly* has everything in a dev board, but unfortunately left out the built-in camera/display interfaces, possibly due to the constraints on the size of the board.

You can make the pins 5v tolerant with 2 very inexpensive parts. A diode, and a resistor. You can get this all the way down to only a diode if the internal pull-up to 3v3 on the pin is strong enough. Basically you end up converting the output from the 5v side to something sort-of like an open collector, which can only pull the signal low.

If you need the signal to switch from input to output, this is pretty simple to do as well, and it all happens without having to use a data direction pin. Requires 2 parts, (resistor and diode) but the diode (3v3 Zener instead of signal) ends up a little more expensive, and can add to current draw on the 5v part side, That's the trade-off. In this version, the Zener clamps the voltage on the wire to 3v3, the resistor goes on the 5v side to limit current. What's actually nice about this setup, if done right, is that signal reflections are eliminated by the resistor. This works because on the 5v side, 3v3 is enough to register as a "1". Unfortunately I can't really give you the value for the resistor, since it depends on the application's needs. As a starting value, say your maximum current output on both is 10mA. the difference between the voltages is 1.7V, so using Ohm's law, the value should be no less than 170ohms. 200 might be better, depending on the type of signals, and is something you have to experiment with.

If none of the above makes sense, here's an ascii-art drawing of both circuits.
```
5v out to 3v3 in

5v device        | /|      3v3 device
5v signal out ---|< |---+- 3v3 signal in
                 | \|   |
                        `--/\/\/--> +3v3
                            1k
```
and...
```
Bidirectional clamp ~8mA

           200ohms
5v device --/\/\/--+-- 3v3 device
                   |
                   |  .-  
                   |  | /| 3v3 zever diode
                   `--|< |--.
                      | \|  |
                     -'     |
                          -----
                           ---
                            -
```