felis/USB_Host_Shield_2.0 #816: Usb.vbusPower(vbus_off);

guldevec, 2024-05-27:
I have an active barcode reader as a HID device working in presentation mode with the shield and working well.
I tried **Usb.vbusPower(vbus_off);** to try to turn the power to the device off so to save the reader when not in use, but could not work it out. 
Is there a way to turn the connected USB device off correctly?

xxxajk, 2024-05-28:
You bought a cheap clone without the ability, that's why it isn't working.

guldevec, 2024-05-28:
Have tried 2 different shields manufacture by different parties, same result. Could you please point me to a "genuine" source of supply or what exact one to purchase please.

USB     Usb;
USBHub     Hub(&Usb);
HIDBoot<USB_HID_PROTOCOL_KEYBOARD>    HidKeyboard(&Usb);
KbdRptParser Prs;

void setup() {
 if (Usb.Init() == -1) Serial.println("OSC did not start.");
// disable until button press on
 Usb.vbusPower(vbus_off);
....