felis/USB_Host_Shield_2.0 #817: Request to Release `1.7.0` + Publish to PlatformIO Registry

plypaul, 2024-07-09:
There have been a couple of helpful PRs that were recently merged to add support for new boards. In particular, this PR added support for the latest official Arduino UNO R4 boards: https://github.com/felis/USB_Host_Shield_2.0/pull/779

Since it has been some time since the last release (`1.6.2` on Jun 9, 2022), what do you all think about tagging a new release and publishing the new version to the PlatformIO registry?

https://registry.platformio.org/libraries/felis/USB-Host-Shield-20

Lauszus, 2024-07-23:
I have now published it here on Github: https://github.com/felis/USB_Host_Shield_2.0/releases/tag/1.7.0, but I do not have access to the PlatformIO registry.

@felis can you publish an update: https://docs.platformio.org/en/latest/core/userguide/pkg/cmd_publish.html#cmd-pkg-publish?

felis, 2024-07-24:
I don't seem to be able to do this atm, nor change access for @Lauszus. I think I know why, need some time to fix it.

Lauszus, 2024-07-29:
@felis any updates? :)