felis/USB_Host_Shield_2.0 #828: Shield does not recoqnize devices

Vin7201, 2024-12-20:
I am fairly new to arduino and trying to get the USB host shield to work with a PS4 controller. I lent my Arduino Uno + Host shield to an aquaintance who got it to work with the HID joystick example. On my PC I cannot get it past waiting for device... in de board_qc with a mouse, bluetooth dongle or PS4 controller. The script does compile and upload perfectly.

I have 5,12V on the 5V pin and 4,02V on the 3,3V pin.

I tried the following script:

/*
 Example sketch for the PS4 USB library - developed by Kristian Lauszus
 For more information visit my blog: http://blog.tkjelectronics.dk/ or
 send me an e-mail:  kristianl@tkjelectronics.com
 */

#include <PS4USB.h>

// Satisfy the IDE, which needs to see the include statment in the ino too.
#ifdef dobogusinclude
#include <spi4teensy3.h>
#endif
#include <SPI.h>
int a = 5;
int b = 5;
USB Usb;
PS4USB PS4(&Usb);

bool printAngle, printTouch;
uint8_t oldL2Value, oldR2Value;

void setup() {
  Serial.begin(115200);
#if !defined(__MIPSEL__)
  while (!Serial); // Wait for serial port to connect - used on Leonardo, Teensy and other boards with built-in USB CDC serial connection
#endif
  if (Usb.Init() == -1) {
    Serial.print(F("\r\nOSC did not start"));
    while (1); // Halt
  }
  Serial.print(F("\r\nPS4 USB Library Started"));
}

void loop() {
  Usb.Task();

Serial.print(" ");
a = PS4.getAnalogHat(LeftHatX);
Serial.print(a);
Serial.print(" ");
b = PS4.getAnalogButton(L2);
Serial.print(b);
delay(1000);
}

Which gives me the following output in the serial monitor no matter if a PS4 controller is connected or not.

PS4 USB Library Started 127 0 127 0 127 0 127 0 127 0 127 0 127 0 127 0 127 0 127 0

The arduino does not react to any input on the  controller.

controllercustom, 2024-12-21:
>  4,02V on the 3,3V pin.

https://forum.arduino.cc/t/usb-host-shield-3-3v-became-4v-after-soldering/

netlcod, 2025-02-01:
You should solder 3.3V instead of 5V (yellow square).