felis/USB_Host_Shield_2.0 #830: Connecting to a Goldmund audio processor

BobHD, 2025-01-07:
Hi,

I am trying to use an Aduino Uno + ARCELI USB Host Shield to send serial commands to a Goldmund audio processor.

These commands when I use a C++ program on Windows 10 with code like:

` WriteFile(hSerial, dataToSend.data(), dataToSend.size(), &bytesWritten, nullptr)`

work well.

I had to use a driver named "FTDI CDM Drivers 2.08.28".

My Arduino program uses:

```

#define ENABLE_UHS_DEBUGGING 1
#include <cdcacm.h>
#include <usbhub.h>                                                                                                            
// Global objects for USB communication
USB Usb;
CDCAsyncOper AsyncOper;
ACM Acm(&Usb, &AsyncOper);

void setup()
{

  Serial.begin(115200);

  while(Usb.Init() == -1)
   { Serial.println("OSC did not start.");
     delay(250);
   }
}
void loop()
{
  Usb.Task(); // Required to maintain USB functionality
  USB_DEVICE_DESCRIPTOR desc_buf;
  if( Usb.getUsbTaskState() == USB_STATE_RUNNING )
   { uint8_t rcode = Usb.getDevDescr(1, 0, 0x12, (uint8_t *)&desc_buf);
     if (rcode)
      { Serial.print(F("\r\nError reading device descriptor: "));
        Serial.println(rcode, HEX);
      }
     else
        Serial.print(F("\r\nDevice descriptor: "));
     delay(10000);
   }
  else
     Serial.println(Usb.getUsbTaskState());
..............
 
Usb.Task();
if (Acm.isReady())
   Acm.SndData(len_serial, buff_serial);
else
   Serial.println("Acm is not ready);
..............
}

Now:

1) Usb.Init() works.

2) Usb.getUsbTaskState() return 18.

3) Acm is not ready.

What can I do to debug this?

```

BobHD, 2025-01-09:
My error was that FTDI is not CDC ACM.

Si I rewrote my program using FTDi:
The PC reference was just to show how the processor reacts to USB commands.

I am trying to replicate what works from a PC using an Adruino Uno + USB shield, with the USB Host Shield Library 2.0.

There is one cable from the Shield to the Processor.

During debugging phase, there is one cable from the PC to tje Arduino, but that's just to upload the program and debug it.

With the help of grok 2, I am trying now:

```
#include <Usb.h>
#include <cdcftdi.h>
 
USB Usb;
FTDI *Ftdi = NULL; // Use a pointer to handle dynamic allocation
 
void setup() {
  Serial.begin(115200);
 
  // Initialize USB
  while (Usb.Init() == -1) {
    Serial.println("OSC did not start.");
    delay(200);
  }
 
  // Manually find and initialize FTDI device
  Serial.println("Line 17");
  Usb.Task();
  Serial.println("Line 18");
  uint8_t devCount = Usb.getUsbTaskState();
  //if (devCount == 0)
      Serial.println(devCount);
  Serial.println("Line 23");
  for (uint8_t i = 0; i < devCount; i++) {
    USB_DEVICE_DESCRIPTOR devDescr;
    if (Usb.getDevDescr(i, 0, 0x12, (uint8_t*)&devDescr) == 0) {
      Serial.println("Line 27");
      // Check if this device is an FTDI device by its VID/PID
      if (Usb.getUsbTaskState() == USB_STATE_RUNNING) {
        if (devDescr.idVendor == 0x0403 && devDescr.idProduct == 0x6001) { // Example VID and PID for FTDI FT232R
          Ftdi = new FTDI(&Usb, i); // Assuming FTDI constructor takes USB instance and device index
          if (Ftdi == NULL) { // Check if allocation failed
            Serial.println("Failed to allocate FTDI instance.");
          } else {
            break; // Found FTDI device
          }
        }
      }
    }
  }
 
  if (Ftdi == NULL) {
    Serial.println("No FTDI device found.");
    while (1);
  }
 
  // Set baud rate to 115200
  if (Ftdi->SetBaudRate(115200) != 0) {
    Serial.println("Failed to set baud rate.");
    while (1); // Halt if baud rate cannot be set
  }
  Serial.println("USB FTDI Initialized");
}
 
void loop() {
  Usb.Task(); // Maintain USB functionality
 
  if (Ftdi->isReady()) {
    const char* cmd = "DE4212";
    uint16_t len = strlen(cmd);
 
    // Write the command string to the FTDI device
    uint8_t rcode = Ftdi->SndData(len, (uint8_t*)cmd);
 
    if (rcode) {
      Serial.print("Send Data Error: ");
      Serial.println(rcode, HEX);
    } else {
      Serial.println("Data sent successfully.");
      delay(1000); // Wait a bit before sending again or looping
    }
  } else {
    Serial.println("FTDI device not ready.");
  }
}
```
that goes straight to:

```Serial.println("No FTDI device found.");```

Any idea?

xxxajk, 2025-01-19:
UHS30 treats all serial (including FTDI) as if they are CDCACM by providing the equivalent responses via a small abstraction layer. Have you tried to see if that will work for you?

BobHD, 2025-01-20:
I did not know about the USB Host Shield 3.0 (just looked it up), I got the first one I found, a DollaTek ESP-12E ESP8266 UART.

I found the library here:

```https://github.com/felis/UHS30```

I'm not sure about the hardware compatible with the UNO REV3 though, is this the one?

```https://www.sparkfun.com/sparkfun-usb-c-host-shield.html```

Would you have a link in Europe?

Thanks.

xxxajk, 2025-01-20:
The sparkfun board is wired wrong, and only old UNO will work with it without modifications, and it doesn't have enough RAM to operate with UHS30 because it uses the actual USB specifications instead of shortcut hacks. 
I have to look into it, but I should be able to get a PJRC Teensy 4.1 running with it, and does not require any additional boards.

BobHD, 2025-01-20:
Let me know please.

Also, what might be a problem is that I also run the IRLib library. 

I know for example that it does not compile with the Arduino UNO WIFI, that I just tested needing to add Ethernet capability. I'm going to try an Ethernet shield on the UNO, but I'm close to max memory with it.

Any suggestion to replace the UNO with another board that will have more memory and work with IRLIB?

xxxajk, 2025-01-20:
Excellent. Don't forget the usb cable (if you got one from an old motherboard for front panel, these work fine) and the ethernet leadout that contains the magnetics/jack. 
Teensy4.1 is a 600MHz beast, and you won't be disappointed, especially if you need a lot of RAM for something. You can add on an extra 16MB of PSRAM (same 8MB on the ESP32) for some fantastic possibilities.

BobHD, 2025-02-09:
> Excellent. Don't forget the usb cable (if you got one from an old motherboard for front panel, these work fine) and the ethernet leadout that contains the magnetics/jack. Teensy4.1 is a 600MHz [...]

Hi,

I got my Teensy 4.1. Pretty nice for the size!

I managed to connect to my Goldmund with this sketch I got from deepsek AI:

```#define CMD  "DE4409"  // Set Input 9
//#define CMD  "DE4212"  // Set Volume 0x11 18
//#define CMD  "DE41"   // Get Volume
                                                                                                                 
#include <USBHost_t36.h>
                                                                                                                 
// Create an instance of the USBHost class
USBHost myusb;
                                                                                                                 
// Create an instance of the USBHub class (optional, if you have a hub connected)
//USBHub hub1(myusb);
                                                                                                                 
// Create an instance of the USBSerial class for the FTDI device
USBSerial userial(myusb);
                                                                                                                 
void setup() {
  // Start the USB host
  myusb.begin();
                                                                                                                 
  // Initialize serial communication for debugging
  Serial.begin(115200);
  while (!Serial); // Wait for the Serial monitor to open (Teensy only)
                                                                                                                 
  Serial.println("Teensy 4.1 USB Host - FTDI Example");
}
                                                                                                                 
void loop() {
  // Check if the USB host has detected the FTDI device
  if (userial) { // Check if the USBSerial object is valid
    // Send an arbitrary string to the FTDI device
    userial.print(CMD);
                                                                                                                 
    // Check if data is available from the FTDI device
    if (userial.available()) {
      // Read and print the response from the FTDI device
      String response = userial.readString();
      Serial.print("Response from FTDI device: ");
      Serial.println(response);
    }
  } else {
    Serial.println("FTDI device not connected.");
  }
                                                                                                                 
  // Delay to avoid spamming the FTDI device
  delay(1000);
}```

But the G does not react (that works with the same string sent from a C++ program running on Windows), and all I get in return is the echo of the character sent:

```Opening port
Port open
VMDPR_Teensy 4.1 USB Host - FTDI Example
FTDI device not connected.
Response from FTDI device: DE4212```

Any suggestion?

xxxajk, 2025-02-10:
First, doesn't look like UHS30 code. It should be using EHCI
Second, "AI" which is actually LLM and not really AI, has never generated code worth a shit, don't use it.
Third, I don't do any code on windows, and never will, so for that, I can't help you, or use something more universal like python + pyserial for the pc side.

For the teensy4.1 please see here:
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_KINETIS_EHCI/examples/KINETIS_EHCI_TEST/KINETIS_EHCI_TEST.ino
and move the conversation to UHS30.

BobHD, 2025-02-10:
Thanks a lot. I'll try and report. 

LLMs (ok, not Skynet :) can be useful, the VS2022 code above worked like a charm and saved me a lot of time. It could certainly be better written, but for testing purposes, it was enough. 

I posted it just to show what works, although the FTDI part is hidden since this program only sees a COM port. The same program works the same whether it drives a real COM port or an USB derived one. I just know that Windows needed an FTDI driver to see this USB device as a COM one.

But I agree that they are often wrong, you have to be very cautious. I got recently a 40 lines generated code, but after examining it, it could be reduced to 4! Still, it was a start that helped.

Ultimately, forums like this one are far better, but using LLMs allows to at least explore something when you are a beginner (it's my case for programming a teensy) and ask the good questions.

BobHD, 2025-02-10:
> First, doesn't look like UHS30 code. It should be using EHCI Second, "AI" which is actually LLM and not really AI, has never generated code worth a shit, don't use it. Third, I don't do any code on [...]

From what I read: 

The USB Host Shield 3.0 (UHS30) is an external USB host solution that works with microcontrollers like Arduino or older Teensy models without built-in USB host.

For the Teensy4.1 that does not need a shield, I have to use indeed the EHCI (Teensy USB Host) library. I think I did with the example that uses USBHost_t36.h, the include for the built-in USB host port. Moving to UHS30 does not seem appropriate, or am I missing something obvious?

I tried to make a post on the Teensy Forum (https://forum.pjrc.com/index.php), but I registered a few days ago, and it still says "Your account is currently awaiting approval by an administrator. You will receive an email when a decision has been taken. "

I'll report if I can get some help there.

But the code using EHCI seems trivial, I am afraid that this library just does not work with the Goldmund.

xxxajk, 2025-02-10:
UHS30 is not just for the shield, it's for some MCUs that have the USB on-chip.
Again, see example, move to UHS30.

BobHD, 2025-02-10:
Ok, will do. Thanks for the clarification.

xxxajk, 2025-02-12:
All you need is this cable:

[image]

Commonly used on older PC's, but PJRC does sell them.

BobHD, 2025-02-12:
Yes, it's what I am using. No joy so far, but I haven't given up yet.

I have yet to find UHS30 examples and test them.

If it works, I'll keep my current Uno Wifi R4 (I managed to make everything else work on it, including IR decoding and LEDBACKPACK), and plug the TX pin to the teeny RX one to convert serial to USB. Using a step down voltage converter to send 3.3V on the teeny of course.

BobHD, 2025-02-14:
Hi,
 
I found a suitable example for the USBHost_t36.h library in:
```
C:\Users\m1\AppData\Local\arduino15\packages\teensy\hardware\avr\1.59.0\libraries\USBHost_t36
```
which I think is what you were taking about.
 
Using this sketch:
                                                                                                                 
```
C:\Users\m1\AppData\Local\arduino15\packages\teensy\hardware\avr\1.59.0\libraries\USBHost_t36\examples\Serial\USBtoUSBHostSerial\USBtoUSBHostSerial.ino
```
                                                                                                                 
I changed it to:
 
```
//#define CMD  "DE4409"  // Set Input 9
#define CMD  "DE4212"  // Set Volume 0x12 18
//#define CMD  "DE41"   // Get Volume
 
//=============================================================================
// Simple USBHost USBSerial test
// This sketch is very much like the main teensy example: USBtoSerial.ino
// but instead of going to hardware serial port it forwards the data
// between the USB Serial and a USB Serial device connected to the USB Host
// port.
//
// This example is in the public domain
//=============================================================================
 
#include <USBHost_t36.h>
 
//=============================================================================
// Options
//=============================================================================
// uncomment the line below to output debug information
//#define DEBUG_OUTPUT
 
// Uncomment the line below to print out information about the USB devices that attach.
#define PRINT_DEVICE_INFO
#define USBBAUD 1000000 //115200
//#define USBBAUD 115200
 
uint32_t baud = USBBAUD;
uint32_t format = USBHOST_SERIAL_8N1;
USBHost myusb;
 
//=============================================================================
// USB Objects
//=============================================================================
 
// Optional if you use are possibly going to plug your USB Serial device
// into a USB Hub, you should include one or more USB Hub objects.
// How many? depends as some HUB chips support lets say 4 ports, so if the HUB
// actuall has more than this, than internally it may be made up using multiple
// HUBs and you may need one of the hub objects for each one of these.
USBHub hub1(myusb);
//USBHub hub2(myusb);
//USBHub hub3(myusb);
 
// There is now two versions of the USBSerial class, that are both derived from a common Base class
// The difference is on how large of transfers that it can handle.  This is controlled by
// the device descriptor, where up to now we handled those up to 64 byte USB transfers.
// But there are now new devices that support larger transfer like 512 bytes.  This for example
// includes the Teensy 4.x boards.  For these we need the big buffer version.
// uncomment one of the following defines for userial
//USBSerial userial(myusb);  // works only for those Serial devices who transfer <=64 bytes (like T3.x, FTDI...)
USBSerial_BigBuffer userial(myusb, 1); // Handles anything up to 512 bytes
//USBSerial_BigBuffer userial(myusb); // Handles up to 512 but by default only for those > 64 bytesUSBHost myusb;
[... 236 lines cut ...]
```
 
I get these results:
 
```
Opening port
Port open
*** Device USERIAL 403:6001 - connected ***
manufacturer: FTDI
product: USB <-> Serial
Sendig command DE4212
No Reply.
Sendig command DE4212
No Reply.
Sendig command DE4212
No Reply.
```
                                                                                                                 
So, the teensy is connecting to an FTDI device named "USERIAL 403:6001", obviously the Goldmund.
                                                                                                                 
But that does not work, unlike when sending the same DE4212 string from a Windows program the same way.
                                                                                                                 
Any guess?
                                                                                                                 
I fear that the teensy drivers are just not apropriate, even though they do see the Goldmund as an FTDI device.

xxxajk, 2025-02-22:
looks like uhs2.0 code...
Please see this example in the uhs3.0 lib, and use the uhs3.0 lib.
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/UHS_KINETIS_EHCI/acm_terminal/acm_terminal.ino

uhs2 treats each serial device differently, and exposes all the odd quirks and bugs, expecting you to know what you are doing. I never really liked it for that alone, and it has other issues that can't be solved on a smaller mcu anyway.
uhs3 treats ALL serial as cdc-acm, instead of doing different code for specific serial, hiding any strangeness from you so you don't have to deal with it.

uhs2 requires you to poll the usb subsystem, and if you block too long in your code, bad things will happen.
uhs3 doesn't require you to do anything to maintain the usb subsystem, since it runs on an interrupt from the USB control chip, so you never call task, or have to worry about calling a task, instead, you just use it.

BobHD, 2025-02-22:
Thanks, that's a good idea to test this more advanced library.

I tried the example you indicated with a Teensy 4.1, but it loops displaying over and over:

```
Starting CDC-ACM test program...
Waiting for Connection...
VMDPR_
```
That's weird, setup() is supposed to be called only once. Here, it keeps restarting.

loop() is never called.

If I set a breakpoint at "printf_P(PSTR("\r\n\r\nWaiting for Connection...\r\n"));", and I don't hit F5 staying paused, I still see continuously what's above.

Any idea?

BobHD, 2025-02-22:
I just found out:

```
UHS30 is designed specifically for the MAX3421E USB controller on a USB Host Shield, not for native USB host ports like the one on the Teensy 4.1
```
I'll have to try with an USB shield instead of the native host USB port.

xxxajk, 2025-02-23:
That is incorrect, where did you read this?

xxxajk, 2025-02-23:
Here is what is currently supported: 

USB HOST SHIELD -- MAX3421E on a proper USB host shield (WARNING: NOT ALL CLONES ARE THE SAME!!!!) https://github.com/felis/UHS30/tree/master/libraries/UHS_host/USB_HOST_SHIELD

ADK -- MAX3421E on the arduino mega ADK https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_ADK

KINETIS FS HOST -- Kinetis USB classic USB low/full speed IP, which is on most Kinetis chips. Some competitor chips offer a VERY SIMILAR interface, down to the same register naming and bits. https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_KINETIS_FS_HOST

KINETIS EHCI -- Kinetis USB using EHCI spec, supporting low/full/high speed USB, common on higher-end Kinetis chips. SHOULD work with other EHCI implementations on other chips with a few tweaks. EHCI is an industry standard. https://github.com/felis/UHS30/tree/master/libraries/UHS_host/UHS_KINETIS_EHCI

So, no it is not "Just for the MAX3421E" add-on chip.

BobHD, 2025-02-23:
All this is new for me. 

The info I got was from Grok 3, another A.I. 

And by the way, an LLM is an A.I., just a specific one that only deals with text. How do I know that? ChatGPT told me, so it must be true. :)

I am a software programmer, mainly in C and REXX, hardware specifics like what you described is not something I'm familiar with.
That's good, it's always nice to learn something new.

Does that mean that the native Teensy 4.1 host USB port is supported?

The unusual was setup() loops and loop() is never called made me think that there was a compatibility problem, so I asked Grok.

Obviously, you know better. LLMs are useful, but as said before they make mistakes.

But I'm still stuck, the example does not work.

xxxajk, 2025-02-24:
Furthermore, 
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/UHS_KINETIS_EHCI/acm_terminal/acm_terminal.ino 
DOES DO Setup() and Loop().

Look at the code, gee whiz man.

xxxajk, 2025-02-24:
By the way, I did try to look up how the rs232/usb port on the device works, and couldn't find anything. 
No example code, no command listings, no protocol information whatsoever.
So, that said, how do you know how to operate it via the serial port anyway?

BobHD, 2025-02-24:
> Furthermore, https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/UHS_KINETIS_EHCI/acm_terminal/acm_terminal.ino DOES DO Setup() and Loop().
> 
> Look at the code, gee whiz man.

I just tried this code, and setup() loops with the Teensy connected to the Goldmund. 

I did look at the code, and setting a breakpoint at:

```
printf_P(PSTR("\r\n\r\nWaiting for Connection...\r\n"));
```
shows that it's looping there.

A beakpoint here:

```
void loop() {
 
        if(Acm.isReady()) { 
````
is never reached.

The code is pretty straightforward, the only reason I see for behaving this way is that the library does not work with the Teensy4.1.

I did install this:

```
https://www.pjrc.com/teensy/package_teensy_index.json
```

If I am wrong, and I would be delighted to be so, it would be nice to know what to do make the code work.

Modifying the library itself is out of my limited expertise in this domain.

> By the way, I did try to look up how the rs232/usb port on the device works, and couldn't find anything. No example code, no command listings, no protocol information whatsoever. So, that said, how do you know how to operate it via the serial port anyway?

I have the Goldmund doc to use the RS/232 port.

For example, sending "DE4212" sets the volume at 0x12 or 18.

Initially, I was using the serial output of the UNO to do that, but the Goldmund serial port is no longer working.

On Windows, I installed an FTDI drive that makes the Goldmund USB port look like a com port. 

The same program that used to work when connecting the PC to the Goldmund RS/232 port (that no longer works) works perfectly when I connect it to the virtual COM device.

The PC is able to pilot the Goldmund using its USB input.

You're right that Goldmund does not document it, but obviously both the USB and the RS/232 port are on the same card, and its firmware reacts the same way whether you use the USB or the RS/232 port to send commands.

I know that LLMs are often wrong, but they allowed me to finish the rest of this project by pointing me in the right direction.

I'm asking here because they are not enough, but they help.

Sometimes it's surprising: I needed to convert an old C program using an obsolete REGEXP library to something more modern running on both Linux and Windows. It gave me all the information needed to use the ECMAScript library on Windows, and the PCRE one on Linux.

And converted my existing code in 2s.

The conversion was flawed, it needed work, but it saved me days of research.

Nothing wrong to use LLMs as long as you are aware of its limits.

The readme that I did read says:

```
Current working USB interface drivers

    HID RAW
    MIDI
    Hub
    Mass storage
    CDC-ACM serial
    XR21B1411 serial
    PROLIFIC serial
    FTDI serial
```
That confirms that it should support the FTDI interface needed with the Goldmund (that works on a PC), but it does not say anything about the Teensy4.1 native host USB port support.

I'm likely missing something, that's why I ask. Not by laziness!

xxxajk, 2025-02-24:
Or the VID:PID on the USB isn't FTDI
What is the VID:PID?

xxxajk, 2025-02-24:
The driver will test for:
UHS_VID_FUTURE_TECHNOLOGY_DEVICES_INTERNATIONAL (0x0403U)
If the device maker changed the VID (they often do to be a pain in the ass) then you need to let me know, so I can fix it.

xxxajk, 2025-02-24:
How it works:
The default VID/PID in the driver is 
0403/6001 for the 'AM', 'BM' and 'R' chips, 
0403/6010 for the FT2232D and FT2232H chips and 
0403/6011 for FT4232H chips.

UHS30 only checks the VID (0403)
Then to sort out the various features/quirks checks
bcdDevice values that are known:
FT232AM 0200
FT232BM 0400
FT2232C 0500
 FT232R 0600
FT2232H 0700
FT4232H 0800
 FT232H 0900
 FT230X 1000

BobHD, 2025-02-24:
I get (I think) that these are the various protocols implemented with various chips.

But I have no way to know what the Goldmund uses. The company is very tight-lipped about their products, they never say anything about the internals. 

And I'm probably their only customer even knowing what an Arduino is!

The sketch should be able to retrieve with the appropriate calls (that I don't know for the moment) but since the loop() does not even start, that would not work either.

xxxajk, 2025-02-24:
Turn on the debug, it will spill out the information on what has connected.
Add here:
https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/UHS_KINETIS_EHCI/acm_terminal/acm_terminal.ino#L10
#define DEBUG_PRINTF_EXTRA_HUGE_ACM_HOST 1

BobHD, 2025-02-25:
Your demo adding the define prints returns:

```
Starting CDC-ACM test program...
Waiting for Connection...
VMDPR_
Starting CDC-ACM test program...
Waiting for Connection...
```
So, the UHS30 library fails to connect, while the USBHost_t36 does.

I managed to send a command to the Goldmund with this lib using:

```
    // Send commands periodically
    static unsigned long lastCmd = 0;
    if (millis() - lastCmd >= 5000) {
      const char* cmds[] = {"DE4409\r", "DE4212\r", "DE41\r"};
      for (int i = 0; i < 1; i++) {
        sendCommand(cmds[i]);
        delay(1000); // Small delay as in original
                                                                                                                 
         // Read from Goldmund with buffering
        static char buffer[64];
        static int bufPos = 0;
        while (userial.available()) {
          char c = userial.read();
          if (c == '\n' || c == '\r' || bufPos >= 63) {
            buffer[bufPos] = '\0';
            Serial.print("Received: ");
            Serial.println(buffer);
            bufPos = 0;
          } else {
            buffer[bufPos++] = c;
          }
        }
      }
      lastCmd = millis();
    }
```
The Goldmund responds with:
```
Sent: DE4409
Received: err: de4409: command not implemented
```
This DE4409 commands works when either sending it from the Arduino over RS-232 (before the input stopped working on the Goldmund), or now when sending it from the PC seeing the Goldmund as a virtual COM port thanks to the FTDI driver.

That seems to be because here I am making the mistake to sending 6 ASCII characters instead of sending 3 bytes 0xDE 0x44 0x09

But when I try to send the correct string with:

```
      uint8_t cmd1[] = {0xDE, 0x44, 0x09, 0x0A};
      len_serial = 4;
      memcpy(buff_serial, cmd1, len_serial);
      sendCommand(buff_serial, len_serial);
      delay(1000); // Longer delay for Goldmund
```
this time the Goldmund does not answer. 

The Goldmund doc says it should echo the command when it's ok, or send back 0xFF if there is an error. I think it's sending " err: de4409: command not implemented" because I sent something else than the start byte that should be sent for these basic commands: 0xDE. And yet, sending 0xDE with a correct following command does not work at all!

I ordered a SparkFun USB-C Host Shield that I will try with the Teensy and the UHS30 library hoping it will work better than the native host port.

xxxajk, 2025-02-26:
VID: 0x0403 PID: 0x6001 is supported. Good.

You are using the Teensy 4.1 correct?
Send me the full sketch so I can take a look at it please.

BobHD, 2025-02-27:
Yes, Teensy4.1.

Here is the sketch that sends 6 ASCII characters instead of sending 3 bytes.

With this one the Goldmund answers wrong command:

```
//#define CMD  "DE4409"  // Set Input 9
// #define CMD  "DE4212"  // Set Volume 0x12 18
//#define CMD  "DE41"   // Get Volume
//  userial.write('0x0A', 1);   // \r 0x0D   \n 0x0A
                                                                                                                 
#include <USBHost_t36.h>
                                                                                                                 
// USB Host objects
USBHost myusb;
USBHub hub1(myusb);
USBHub hub2(myusb);
USBSerial userial(myusb);
USBHIDParser hid1(myusb);
uint8_t buff_serial[20];
                                                                                                                 
void setup() {
  while (!Serial);
  Serial.begin(115200);
  myusb.begin();
  Serial.println("USB Host Started");
}
                                                                                                                 
void sendCommand(const char* cmd) {
  int len = strlen(cmd);
  userial.write((const uint8_t*)cmd, len);
  Serial.print("Sent: ");
  Serial.println(cmd);
}
                                                                                                                 
void loop() {
  myusb.Task();
                                                                                                                 
  if (userial) {
    static bool printed_serial = false;
    if (!printed_serial) {
      Serial.println("CDC-ACM Device Connected:");
      Serial.printf("  VID: 0x%04X\n", userial.idVendor());
      Serial.printf("  PID: 0x%04X\n", userial.idProduct());
      printed_serial = true;
    }
                                                                                                                 
    // Echo Serial input to Goldmund
    if (Serial.available()) {
      char c = Serial.read();
      userial.write(c);
      Serial.write(c);
    }
                                                                                                                 
    // Read from Goldmund with buffering
    static char buffer[64];
    static int bufPos = 0;
    while (userial.available()) {
      char c = userial.read();
      if (c == '\n' || c == '\r' || bufPos >= 63) {
        buffer[bufPos] = '\0';
        Serial.print("Received: ");
        Serial.println(buffer);
        bufPos = 0;
      } else {
        buffer[bufPos++] = c;
      }
    }
                                                                                                                 
[... 170 lines cut ...]
      delay(1000); */
                                                                                                                 
      lastCmd = millis();
    }
  }
                                                                                                                 
  if (hid1) {
    static bool printed_hid = false;
    if (!printed_hid) {
      Serial.println("HID Device Connected:");
      Serial.printf("  VID: 0x%04X\n", hid1.idVendor());
      Serial.printf("  PID: 0x%04X\n", hid1.idProduct());
      printed_hid = true;
    }
  }
                                                                                                                 
  delay(10);
}
```

Both were written by Grok 3. It got then wrong at first, with compile errors. But with a little back and forth sending it back the errors, it made then work. Grok also made the mistake of sending ASCII when I gave it the PC C program that sends bytes as an example that was working, I spotted it. So you're right they often give very wrong answers, but for me it was a start on a subject I know nothing about.

With USB30, there is no communication at all, the VID is not detected.

The "Manual input part" is useless.

xxxajk, 2025-02-27:
Neither of these is UHS30

BobHD, 2025-02-27:
Not at all, I did install UHS30 and tested your example. It just does not work.

And I did say the examples above were using USBHost_t36, not UHS30, that was the point.

UHS30 can't even find the VID number.

I'm not ignoring you, on the contrary. What you say is helpful, but UHS30 just does not work with the Goldmund.

Now, if you want to post here an USS30 sketch that just polls for the VID number and prove me wrong, I'll be more than happy to be so.

xxxajk, 2025-03-03:
So, seems that with the teensy 4.x, bitbanding isn't available (BOOOOOOO!) and I have fixed it.
The updated code will be uploaded after a few more things are verified.

xxxajk, 2025-03-03:
You should see something similar to the following using the CDC-ACM test:

```01:30:29.159 -> 
01:30:29.159 -> 
01:30:29.159 -> Waiting for Connection...
01:30:29.159 -> 
01:30:29.159 -> 
01:30:29.159 -> 
01:30:29.159 -> Configuring: parent = 0, port = 1, speed = 1
01:30:29.358 -> $$$$$$$$$$$$$$$$$$$$$$$$$ ADDRESS 1: 0 retries.
01:30:29.953 -> $$$$$$$$$$$$$$$$$$$$$$$$$ ADDRESS 1: 0 retries.
01:30:29.953 -> USBHub: checking numep 2, klass 00, interface.klass ff, protocol ff/00-00?01
01:30:29.953 -> USBHub: checking numep 2, klass 00, interface.klass ff, protocol ff/00-01?01
01:30:29.953 -> 
01:30:29.953 -> 
01:30:29.953 -> Connected.
01:30:29.953 -> 
```

BobHD, 2025-03-03:
> So, seems that with the teensy 4.x, bitbanding isn't available (BOOOOOOO!) and I have fixed it. The updated code will be uploaded after a few more things are verified.

That's great, thanks a lot. 

I tested:

https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/UHS_KINETIS_EHCI/acm_terminal/acm_terminal.ino#L10

after redownloading:

https://github.com/felis/UHS30/tree/master

I'm not sure if it has already been updated or not. Some files are more recent than on the one I previously downloaded.
 If not, no hurry. Please take your time.

For the moment I get:

```
Starting CDC-ACM test program...
Waiting for Connection...
08:29:03.332    GoldCont.ino, line 100		millis()=31405
     Millis is currently 31405.
08:29:03.569    GoldCont.ino, line 100		millis()=31655
     Millis is currently 31655.
```
A break point after "if(Acm.isReady()) {" is never triggered.

Also although I am using (with Visual Micro/VS2022) the "Arduio 2 - Teensy 4.1" setting, I get when I upload:

```
Compiling debug version of 'GoldCont' for 'Teensy 4.1 (teensy41)'
Build Folder: "file:///C:/Users/m1/AppData/Local/Temp/VMBuilds/GoldCont/teensy41/Debug"
 
UHS_KINETIS_EHCI.h:316: In file included from
uhs_host.h:68: from
GoldCont.ino:43: from
UHS_KINETIS_EHCI_INLINE.h: In member function virtual int16_t UHS_KINETIS_EHCI::Init(int16_t)
UHS_KINETIS_EHCI_INLINE.h: 360:15: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'uhs_kehci_qtd_t' {aka 'struct _uhs_kehci_qtd'} with no trivial copy-assignment; use value-initialization instead [-Wclass-memaccess]
   360 |         memset(&qHalt, 0, sizeof (qHalt))
   |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
 
uhs_host.h:68: In file included from
GoldCont.ino:43: from
UHS_KINETIS_EHCI.h:162: note  uhs_kehci_qtd_t {aka struct _uhs_kehci_qtd} declared here
   162 | typedef struct _uhs_kehci_qtd {
   |                ^~~~~~~~~~~~~~
 
Warning*: for Arduino 1.8.x, Teensyduino must be installed with a special installer
   Boards Manager installation give incomplete setup!
  Download the Installer at https*: www.pjrc.com\teensy\td_download.html
teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:55100, data:7208, headers:8344   free for files:8055812
teensy_size:    RAM1: variables:26144, code:52400, padding:13136   free for local variables:432608
teensy_size:    RAM2: variables:12416  free for malloc/new:511872
 
Uploading 'GoldCont' to 'Teensy 4.1 (teensy41)' using 'usb:0/140000/0/6/4/4/3'
Warning: for Arduino 1.8.x, Teensyduino must be installed with a special installer.
  Boards Manager installation give incomplete setup!
  Download the Installer at https://www.pjrc.com/teensy/td_download.html
```
The Arduino IDE shows that the Teensy 2.04 version 1.59.0 board is installed, so I don't understand this warning.

xxxajk, 2025-03-03:
Current dev versions is:
Arduino 1.8.13
Teensyduino 1.54
Yes, both are old, but shouldn't matter.

xxxajk, 2025-03-03:
Pushed.

BobHD, 2025-03-03:
> Pushed.

Thanks. 

I downloaded the zip file again from https://github.com/felis/UHS30/tree/master, deleted manually "C:\Users\m1\Documents\Arduino\libraries\UHS30-master" to be sure and then extracted the zip file there.

I did a full re-build, but I am getting the same results: a break point after "if(Acm.isReady()) {" is never triggered, all I get is "Waiting for Connection..."

I tried again to use the PC to check that the cable is working and the Goldmund responds to commands. That still works.

Anything specific I can do to debug this?

xxxajk, 2025-03-03:
where did you unzip it?
Were there any compile errors or warnings?

BobHD, 2025-03-03:
> where did you unzip it? Looks OK. Were there any compile errors or warnings?

My username being "m1", I unzipped it in:

```
C:\Users\m1\Documents\Arduino\libraries
```
There was already a library with the same name and same subdirectories there, that I deleted first.

I got these warnings first:

```
Visual Micro Tutorial Mode: Board packages were loaded on-demand because a toolbar or project required them. Tip: To improve the speed that Visual Studio loads, use Visual Studio's toolbar visibility or vMicro's auto toolbar visibility to control when the board packages are loaded into cache. The small VMicro toolbar can always be visible, it does not cause load until clicked. Other toolbars might cause load. (Tutorial mode can be switched off using the vMicro menu, it is recommended to keep it on initially when using this product)
Compiling debug version of 'GoldCont' for 'Teensy 4.1 (teensy41)'
Build Folder: "file:///C:/Users/m1/AppData/Local/Temp/VMBuilds/GoldCont/teensy41/Debug"
 
UHS_KINETIS_EHCI.h:316: In file included from
uhs_host.h:68: from
GoldCont.ino:43: from
UHS_KINETIS_EHCI_INLINE.h: In member function virtual int16_t UHS_KINETIS_EHCI::Init(int16_t)
UHS_KINETIS_EHCI_INLINE.h: 360:15: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'uhs_kehci_qtd_t' {aka 'struct _uhs_kehci_qtd'} with no trivial copy-assignment; use value-initialization instead [-Wclass-memaccess]
   360 |         memset(&qHalt, 0, sizeof (qHalt))
   |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
 
uhs_host.h:68: In file included from
GoldCont.ino:43: from
UHS_KINETIS_EHCI.h:162: note  uhs_kehci_qtd_t {aka struct _uhs_kehci_qtd} declared here
   162 | typedef struct _uhs_kehci_qtd {
   |                ^~~~~~~~~~~~~~
 
Warning*: for Arduino 1.8.x, Teensyduino must be installed with a special installer
   Boards Manager installation give incomplete setup!
  Download the Installer at https*: www.pjrc.com\teensy\td_download.html
   Opening Teensy Loader
teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:55228, data:7208, headers:8216   free for files:8055812
teensy_size:    RAM1: variables:26144, code:52528, padding:13008   free for local variables:432608
teensy_size:    RAM2: variables:12416  free for malloc/new:511872
 
Uploading 'GoldCont' to 'Teensy 4.1 (teensy41)' using 'usb:0/140000/0/6/4/4/3'
Warning: for Arduino 1.8.x, Teensyduino must be installed with a special installer.
  Boards Manager installation give incomplete setup!
  Download the Installer at https://www.pjrc.com/teensy/td_download.html
```
These are the same as before I think, and they disappear when the program is loaded, all I can see then is:
```
Launching Debugger...
   - Serial debug will start immediately. WiFi (or gdb) debug will start a few seconds or a short time after the remote device connects to the network.
   - Switch on 'vmicro>debugger>full speed' to prevent debugger message throttling. (read the docs)
   - If 'in-break live watch' is disabled, continue an existing paused debug session by sending the character 'c' to the board using the Serial window.
   - If uploading on WiFi but debug is on Serial then it is possible the cpu will be in debug break when OTA upload is attempting connect. If this is the case then sending char 'c' using the serial/debug terminal will force the mcu to continue (like F5).
   - WiFi debug? Wifi debug is not fully supported yet. Use the vMicro>Debugger menu to set an alternative debug port such as Serial (default speed is 115k)
   - To view debug messages without trace messages use the 'message windows' option on the monitor window.
   - Switch off this message and other Tutorial messages using the 'vMicro>General' menu
11:07:26.461    GoldCont.ino, line 97	setup()
GoldCont is paused at GoldCont.ino, line 97, setup()
```
line 97 is " printf_P(PSTR("\r\n\r\nWaiting for Connection...\r\n"));" which is indeed displayed, but then "Acm.isReady()" is never true.

xxxajk, 2025-03-03:
Line 360's warning is interesting, and a new one...
I'll look into it.

xxxajk, 2025-03-03:
update again, warning should be fixed.
Try also with a plain FTDI adapter as well.

xxxajk, 2025-03-03:
Session from  regular old arduino:
```
06:52:26.455 -> 
06:52:26.455 -> 
06:52:26.455 -> Starting CDC-ACM test program...
06:52:27.579 -> 
06:52:27.579 -> 
06:52:27.579 -> 
06:52:27.579 -> Configuring: parent = 0, port = 1, speed = 1
06:52:28.637 -> $$$$$$$$$$$$$$$$$$$$$$$$$ ADDRESS 1: 0 retries.
06:52:29.232 -> $$$$$$$$$$$$$$$$$$$$$$$$$ ADDRESS 1: 0 retries.
06:52:29.232 -> USBHub: checking numep 2, klass 00, interface.klass ff, protocol ff/00-00?01
06:52:29.232 -> USBHub: checking numep 2, klass 00, interface.klass ff, protocol ff/00-01?01
06:52:29.232 -> 
06:52:29.232 -> 
06:52:29.232 -> Connected.
06:52:29.232 -> 
06:52:47.979 -> testing 1,2,3
06:53:02.129 -> works for me....
```
I entered the testing 1,2,3 hit send, then works for me, send
FTDI has TX and RX connected together so that the output loops right back.

BobHD, 2025-03-03:
> update again, warning should be fixed. 

I redownloaded UHS30-master.zip.

Just to check it's the last one, checksums are:

```
  File: UHS30-master.zip
CRC-32: c6a5b50a
   MD4: 29e7b5b747dae367ab75dbcecc74756b
   MD5: 8d1c20ea7da7663f57e03f8526de25b9
 SHA-1: 4877d25fe53ed1133ec80c7d2e1cb36fe8ece85d
```
I am still getting doing a full rebuild:
```
Compiling debug version of 'GoldCont' for 'Teensy 4.1 (teensy41)'
Build Folder: "file:///C:/Users/m1/AppData/Local/Temp/VMBuilds/GoldCont/teensy41/Debug"
 
UHS_KINETIS_EHCI.h:316: In file included from
uhs_host.h:68: from
GoldCont.ino:43: from
UHS_KINETIS_EHCI_INLINE.h: In member function virtual int16_t UHS_KINETIS_EHCI::Init(int16_t)
UHS_KINETIS_EHCI_INLINE.h: 360:15: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'uhs_kehci_qtd_t' {aka 'struct _uhs_kehci_qtd'} with no trivial copy-assignment; use value-initialization instead [-Wclass-memaccess]
   360 |         memset(&qHalt, 0, sizeof (qHalt))
   |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
 
uhs_host.h:68: In file included from
GoldCont.ino:43: from
UHS_KINETIS_EHCI.h:162: note  uhs_kehci_qtd_t {aka struct _uhs_kehci_qtd} declared here
   162 | typedef struct _uhs_kehci_qtd {
   |                ^~~~~~~~~~~~~~
 
Warning*: for Arduino 1.8.x, Teensyduino must be installed with a special installer
   Boards Manager installation give incomplete setup!
  Download the Installer at https*: www.pjrc.com\teensy\td_download.html
teensy_size: Memory Usage on Teensy 4.1:
teensy_size:   FLASH: code:55100, data:7208, headers:8344   free for files:8055812
teensy_size:    RAM1: variables:26144, code:52400, padding:13136   free for local variables:432608
teensy_size:    RAM2: variables:12416  free for malloc/new:511872
```
Did I miss something?

I got the same results as before:

```
Starting CDC-ACM test program...
Waiting for Connection...
```
I takes along time to load, around 60s. Is that normal?

> Try also with a plain FTDI adapter as well.

Do you mean something else than the Goldmund?

Aside a keyboard, the only USB device I have is an HDD USB dock.

It's not an FTDi device I think?

Anyways, results are the same: no connection.

xxxajk, 2025-03-04:
doesn't look like it updated.

BobHD, 2025-03-04:
> doesn't look like it updated.

Nope.
I downloaded the ZIP again after seeing your post, it's the same.

This time, just to see if it makes a differences, I made the library update using the Visual Micro interface for that.

I got the same results as unzipping manually into the library folder.

xxxajk, 2025-03-04:
Here's a test using a USB thumb drive:
```11:05:39.380 -> 
11:05:39.380 -> 
11:05:39.380 -> Start.
11:05:39.380 -> 
11:05:39.380 -> SWI_IRQ_NUM 71
11:05:40.405 -> 
11:05:40.405 -> 
11:05:40.405 -> USB HOST READY.
11:05:40.405 -> No media. Waiting to mount /
11:07:52.030 -> 
11:07:52.030 -> 
11:07:52.030 -> 
11:07:52.030 -> Configuring: parent = 0, port = 1, speed = 2
11:07:53.088 -> $$$$$$$$$$$$$$$$$$$$$$$$$ ADDRESS 1: 0 retries.
11:07:53.650 -> $$$$$$$$$$$$$$$$$$$$$$$$$ ADDRESS 1: 0 retries.
11:07:53.650 -> USBHub: checking numep 2, klass 00, interface.klass 08, protocol 50/00-00?01
11:07:53.650 -> BulkOnly: checking numep 2, klass 00, subklass 00
11:07:53.650 -> BulkOnly: checking protocol 00, interface.klass 08, interface.subklass 06
11:07:53.650 -> BulkOnly: checking interface.protocol 50
11:07:53.650 -> USBHub: checking numep 2, klass 00, interface.klass 08, protocol 50/00-01?01
11:07:53.650 -> BulkOnly: checking numep 2, klass 00, subklass 00
11:07:53.650 -> BulkOnly: checking protocol 00, interface.klass 08, interface.subklass 06
11:07:53.650 -> BulkOnly: checking interface.protocol 50
11:07:53.650 -> BS SetInterface
11:07:53.650 -> ep: 0x00 bmAttributes: 0x02 index: 1
11:07:53.650 -> 
11:07:53.650 -> ep: 0x01 bmAttributes: 0x02 index: 2
11:07:53.650 -> 
11:07:53.650 -> BS Start, speed: 2
11:07:53.650 -> BS Start
11:07:54.675 -> GetMaxLUN 0x00
11:07:54.675 -> MaxLUN 0
11:07:54.807 -> Inquiry 0x24 0x00
11:07:54.807 -> LUN 0 `        USB DISK 3.0    PMAP'
11:07:54.807 -> Qualifier 0 Device type 00 RMB 1 SSCS 0 SCSI version 06
11:07:54.807 -> Device conforms to T10/1731-D (SPC-4) standards.
11:07:54.840 -> CheckLUN...
11:07:54.840 -> 137
11:07:54.840 -> >>>>>>>>>>>>>>>>CAPACITY OK ON LUN 0
11:07:54.840 -> 073887ff00000200
11:07:54.840 -> 
11:07:54.840 -> 137
11:07:54.840 -> Checked LUN...
11:07:54.840 -> Onstart begin
11:07:54.840 -> BS configured
11:07:54.840 -> 
11:07:55.303 -> / mounted.
11:07:55.303 -> Removing '/HeLlO.tXt' file... completed with 4
11:07:55.303 -> 
11:07:55.303 -> Starting Write test...
11:07:55.336 -> File opened OK, fd = 1
11:07:55.336 -> Wrote 19 bytes, File closed result = 0.
11:07:55.336 -> 
11:07:55.336 -> Starting Read test...
11:07:55.336 -> File opened OK, fd = 1, displaying contents...
11:07:55.369 -> ]-[ello \/\/orld!
11:07:55.369 -> 
11:07:55.369 -> Read completed, last read result = -1 (20), file close result = 0.
11:07:55.369 -> Testing rename
11:07:55.369 -> file rename result = 0.
11:07:55.369 -> 
11:07:55.369 -> Removing '/10MB.bin' file... completed with 4
11:07:55.369 -> 10MB write timing test  10240 writes, (0), (0),  61422 ms (61 sec)
11:08:57.295 -> completed with 0
11:08:57.295 -> 10MB read timing test 10240 reads, (20),  7732 ms (8 sec)
11:09:05.528 -> completed with 0
11:09:05.528 -> Directory of '/'
11:09:05.528 -> -rw--a     10485760 2055-04-04 11:06:34 10MB.bin
11:09:05.528 -> -rw--a           19 2055-04-04 11:05:32 newtest.txt
11:09:05.528 -> 61994172416 bytes available on disk.
11:09:05.528 -> 
11:09:05.528 -> Flushing caches...
11:09:05.528 -> Remove and insert media...
```

So everything here works as expected.
Under the  UHS_FS examples try UHS_KINETIS_EHCI demo with a thumb drive formatted as fat32 with no label.
It is very important that it has no label, or the test will not work.
...and yes, the dates are totally off, need to look into that one.

BobHD, 2025-03-05:
So I tried first with an USB thumb drive formatted without label:

```
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/examples/UHS_KINETIS_EHCI/UHS_FS_NEW_DEMO/UHS_FS_NEW_DEMO.ino
```
but it fails with:

```
uhs_fs.h:613: In file included from
uhs_host.h:82: from
GoldCont.ino:115: from
 
FAT.cpp: 31:2: error: #error RTClib NOT LOADED
   31 | #error RTClib NOT LOADED
   |  ^~~~~
 
UHS_KINETIS_EHCI.h:316: In file included from
uhs_host.h:68: from
GoldCont.ino:115: from
UHS_KINETIS_EHCI_INLINE.h: In member function virtual int16_t UHS_KINETIS_EHCI::Init(int16_t)
UHS_KINETIS_EHCI_INLINE.h: 360:15: warning: 'void* memset(void*, int, size_t)' clearing an object of type 'uhs_kehci_qtd_t' {aka 'struct _uhs_kehci_qtd'} with no trivial copy-assignment; use value-initialization instead [-Wclass-memaccess]
   360 |         memset(&qHalt, 0, sizeof (qHalt))
   |         ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
 
uhs_host.h:68: In file included from
GoldCont.ino:115: from
UHS_KINETIS_EHCI.h:162: note  uhs_kehci_qtd_t {aka struct _uhs_kehci_qtd} declared here
   162 | typedef struct _uhs_kehci_qtd {
   |                ^~~~~~~~~~~~~~
 
uhs_fs.h:609: In file included from
uhs_host.h:82: from
GoldCont.ino:115: from
 
ff.c: In function FRESULT f_open(FIL*, const TCHAR*, FBYTE)
ff.c: 2546:34: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   2546 |         res = chk_mounted(&path, &dj.fs, (FBYTE) (mode & ~FA_READ))
   |                                  ^~~~~~
 
ff.c: In function FRESULT f_chdir(const TCHAR*)
ff.c: 3010:34: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   3010 |         res = chk_mounted(&path, &dj.fs, 0)
   |                                  ^~~~~~
 
ff.c: In function FRESULT f_getcwd(TCHAR*, UINT)
ff.c: 3051:50: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   3051 |         res = chk_mounted((const TCHAR**) &buff, &dj.fs, 0); \* Get current volume *
   |                                                  ^~~~~~
 
ff.c: In function FRESULT f_opendir(DIR*, const TCHAR*)
ff.c: 3320:34: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   3320 |         res = chk_mounted(&path, &dj->fs, 0)
   |                                  ^~~~~~~
 
ff.c: In function FRESULT f_stat(const TCHAR*, FILINFO*)
ff.c: 3406:34: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   3406 |         res = chk_mounted(&path, &dj.fs, 0)
   |                                  ^~~~~~
 
ff.c: In function FRESULT f_unlink(const TCHAR*)
ff.c: 3565:34: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   3565 |         res = chk_mounted(&path, &dj.fs, 1)
   |                                  ^~~~~~
 
ff.c: In function FRESULT f_mkdir(const TCHAR*)
ff.c: 3636:34: warning: taking address of packed member of 'DIR' may result in an unaligned pointer value [-Waddress-of-packed-member]
   3636 |         res = chk_mounted(&path, &dj.fs, 1)
   |                                  ^~~~~~
 
ff.c: In function FRESULT f_chmod(const TCHAR*, FBYTE, FBYTE)
[... 56 lines cut ...]
   |   ^~~~~~~~
rtclib.h:149: note    conversion of argument 2 would be ill-formed
GoldCont.ino: 138:32: warning: invalid conversion from 'WORD' {aka 'short unsigned int'} to 'const char*' [-fpermissive]
   138 |                                         if(de->fattrib & AM_DIR) {
   |                                ^
   |                                |
   |   
```
I tested then:
```
https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/examples/UHS_KINETIS_EHCI/IOCTL/IOCTL.ino
```

Same errors:

```
uhs_fs.h:613: In file included from
uhs_host.h:82: from
GoldCont.ino:108: from
 
FAT.cpp: 31:2: error: #error RTClib NOT LOADED
   31 | #error RTClib NOT LOADED
   |  ^~~~~
```
Did I test the files you intended me to?

xxxajk, 2025-03-05:
you need to download as it is stated in the documentation:
```
With a github account:
git clone --recursive git@github.com:felis/UHS30.git
or
Without a github account:
git clone --recursive https://github.com/felis/UHS30.git
```

BobHD, 2025-03-05:
I used git for Windows, and then tested the downloaded file in 
``` 
C:\Users\m1\UHS30\libraries\UHS_FS\examples\UHS_KINETIS_EHCI\IOCTL
```

I get similar errors:

```
uhs_fs.h:613: In file included from
uhs_host.h:82: from
GoldCont.ino:112: from
 
FAT.cpp: 31:2: error: #error RTClib NOT LOADED
   31 | #error RTClib NOT LOADED
```
Same for:
```
C:\Users\m1\UHS30\libraries\UHS_FS\examples\UHS_KINETIS_EHCI\UHS_FS_NEW_DEMO\UHS_FS_NEW_DEMO.ino
```

xxxajk, 2025-03-05:
Is RTClib present in ~Arduino/libraries ?

BobHD, 2025-03-05:
Yes, I have a folder named:

``` C:\Users\m1\Documents\Arduino\libraries\RTClib```

Content is:

```
code-of-conduct.md        5829 15/01/25 04:01
examples                  4096 15/01/25 04:01
keywords.txt              1808 15/01/25 04:01
library.properties         306 15/01/25 04:01
license.txt               1076 15/01/25 04:01
README.md                 4076 15/01/25 04:01
src                       4096 15/01/25 04:01
```

xxxajk, 2025-03-05:
Wrong RTClib :-)
Replace it with the one included in the git clone you made.

BobHD, 2025-03-05:
When I do that, I get:

```
Build Folder: "file:///C:/Users/m1/AppData/Local/Temp/VMBuilds/GoldCont/teensy41/Debug"
Could not find a part of the path c*: Users\m1\Documents\Arduino\libraries\RTClib\src

	An error was encountered during the 'Deep Search' library discovery process.
Debug build failed for project 'GoldCont'
```

xxxajk, 2025-03-05:
Refresh the arduino caches.
To do that, run arduino, open the libraries manager, and then close it.
Then it will be able to discover them correctly.

BobHD, 2025-03-05:
That worked, no more compile errors;

This sketch displays when using the USB thumb drive:

```https://github.com/felis/UHS30/blob/master/libraries/UHS_FS/examples/UHS_KINETIS_EHCI/UHS_FS_NEW_DEMO/UHS_FS_NEW_DEMO.ino```

```
Opening port
Port open
VMDPR_
Start.
SWI_IRQ_NUM 71
USB HOST READY.
No media. Waiting to mount /
```

Same output when connecting the Goldmund.

Then I tried:

```https://github.com/felis/UHS30/blob/master/libraries/UHS_host/UHS_CDC_ACM/examples/UHS_KINETIS_EHCI/acm_terminal/acm_terminal.ino#L10```

And I got with the thumb drive and with the Goldmund:

```
Uploading to I/O board
Opening port
Port open
VMDPR_
Starting CDC-ACM test program...
Waiting for Connection...a
VMDPR_
Starting CDC-ACM test program...
Waiting for Connection...
VMDPR_
```
I'm not sure what sketch to use to try displaying VID:PID.

I'm a bit out of my depth here, I'm just following orders now. :)

xxxajk, 2025-03-05:
Take a picture of the hardware please.

BobHD, 2025-03-05:
Here is the Teensy connected to the thumb drive:

[image]

I don't think you meant the Goldmund, but just for kicks:

[image]

[7 more replies]