felis/USB_Host_Shield_2.0 #847: PS5 info?

sl0sh, 2025-09-05:
Hi,
Just a basic question not an issue. Can USB_Host_Shield_2.0 be used to record the information from a PS5 dualsense controller like e.g. hardware address and then use that information to write an Arduino sketch to automate a PS5 console with an Arduino device. 
Thanks

Lauszus, 2025-09-05:
> Hi,
> Just a basic question not an issue. Can USB_Host_Shield_2.0 be used to record the information from a PS5 dualsense controller like e.g. hardware address and then use that information to write an Arduino sketch to automate a PS5 console with an Arduino device. 
> Thanks

I assumed you meant "PS5 controller" and not "PS5 console"?

Yes it's possible to emulate a PS5 controller. 

What you are looking for is an Arduino that can act as a USB device (look up Teensy). This library is for connecting devices like gamepads, mice etc to an Arduino.

sl0sh, 2025-09-05:
what Teensy library should i use to do that?